moonlight-stream/moonlight-embedded #68: Error Code 522: GeForce Experience 2.1.5.0

PcInfamy, 2015-01-02:
Im getting this error on GeForce Experience 2.1.5.0:

```
root@raspberrypi:/home/pi# java -jar limelight-pi.jar 192.168.1.22
Starting app
com.limelight.nvstream.http.GfeHttpResponseException: GFE error: Streaming session has failed due to a game management error. (Error code: 522)
        at com.limelight.nvstream.http.NvHTTP.verifyResponseStatus(Unknown Source)
        at com.limelight.nvstream.http.NvHTTP.getXmlString(Unknown Source)
        at com.limelight.nvstream.http.NvHTTP.getXmlString(Unknown Source)
        at com.limelight.nvstream.http.PairingManager.getPairState(Unknown Source)
        at com.limelight.nvstream.http.NvHTTP.getPairState(Unknown Source)
        at com.limelight.nvstream.NvConnection.startApp(Unknown Source)
        at com.limelight.nvstream.NvConnection.establishConnection(Unknown Source)
        at com.limelight.nvstream.NvConnection.access$300(Unknown Source)
        at com.limelight.nvstream.NvConnection$1.run(Unknown Source)
        at java.lang.Thread.run(Thread.java:744)
GFE error: Streaming session has failed due to a game management error. (Error code: 522)
Connection Error Starting app failed
```

Any thoughts?

irtimmer, 2015-01-03:
Is a bug in GFE not in Limelight Embedded. You propably need to reinstall GFE/Steam

closed 2015-01-03 by irtimmer

PcInfamy, 2015-01-03:
Already tried that, didn't work.

dodslaser, 2015-01-10:
The code seems to be pretty non-specific. I've been getting the same code when l launch while Kodi is running, or when the system is low on resources. Try killing all unnecessary processes.

PcInfamy, 2015-01-10:
It's a fresh install of Raspian, there isn't anything else running really.

dodslaser, 2015-01-12:
Reboot all equipment between the pi and your computer (including the pi and the computer). Remove *~/client.crt* *~/client.key* and *~/uniqueid.dat* on the pi. Clear the shield devices in GFE and try pairing it again. I know this sounds like a typical dumb IT-support solution, but it might just work.