moonlight-stream/moonlight-embedded #69: Mouse - Low Sensitivity?

danchappers, 2015-01-03:
Hi,

I'm not sure if this is an issue per-se but I the mouse sensitivity seems a bit too low and I can't seem to find anywhere to change it. I don't get any error messages but it does cause a noticeable impact to game play. I don't think I am experiencing lag as the video appears to be coming through just fine.

I'm pretty new to Linux, but if I can get a log or anything to help out let me know

danchappers, 2015-01-05:
Hello,

Just to add to this, after testing using a mouse on the computer streaming the games, this definitely seems to be input lag or low sensitivity on the pi as there was no noticeable lag at all during the furthet testing in the video or mouse movement on the pi.

irtimmer, 2016-01-16:
Mouse sensitivity should be configured in Windows itself

closed 2016-01-16 by irtimmer