moonlight-stream/moonlight-embedded #70: Sound stutter whilst controller plugged in

gforced, 2015-01-04:
Streaming is fine sound only stutters when a button is pressed on the controller 
Currently using Xbox 360 controller via xboxdrv 
Unplugging the controller from the pi resolves the issue. 
I'm on the latest gfe and limelight release

irtimmer, 2015-01-04:
Please test if the last commit resolves (or download limelight.jar from http://www.filedropper.com/limelight ) your issue

gforced, 2015-01-05:
thankyou, is the link from filedropper an alternative to compiling the latest commit?

gforced, 2015-01-05:
ok - downloaded the .jar file from filedropper, ran it on the PI. default settings., controller in the pi. 
sound stutter is MUCH less though still there, I would say reduced by about 70%..

RoelofBerg, 2015-01-05:
Hi, same issue here. When I played with mouse and keyboard, the sound is excellent. After I installed a PS3 controller over bluetooth the shound stutters. limelightpi 3.12.35+

And by the way the joystick-part of the controller works very fluently and reliably. But button-press events are often unreliable and/or delayed. (Maybe there's a connection between sound-delay and button-delay ?)

Here's a protocol of my whole installation (Autoboot limelight on a Raspberry-Pi when powered on, autostart the PC by Wake-On-Lan from the Pi, use a PS3 dual shock controller over bluetooth): http://embedded-software-architecture.com/wk/index.php/Limelight_on_a_Raspberry_Pi#Howto:_50.24_Steambox_with_Limelight_on_a_Raspberry_Pi

gforced, 2015-01-06:
good to see it is not only myself, and yes i concur; i receive the exact symptoms on the xbox360 wireless controller (and that is not bluetooth)

Very nice guide btw, thank you. this helps greatly. 

PS- sound stutter could be avoided by mapping PS3 controller directly to PC (this is not ideal for me since the PI and PC are too far away from each other)

RoelofBerg, 2015-01-08:
I came across several other people facing this issue on http://www.raspberrypi.org/forums/viewtopic.php?f=78&t=65878&p=668055#p668055 .

Is at least one installation known out there that operates Limelight on a Raspberry Pi B+ with a (PI-connected) controller and without sound stuttering ? If yes it would be interesting to know which controller and which way of connection (USB/Wlan/Bluetooth) is used ?

Impulses for workarounds (if this is a limitation/bug in the software):
- Does anyone know about WiFi-Controllers that connect directly to the PC (without the Pi in between) ? That would be a perfect solution. Nvidia uses WiFi, but I had the feeling that the controller does not connect directly to the PC.
- I was wondering whether a common USB-Over-Ethernet appliance (that would make the controller look like being connected directly to the PC) could solve this issue - assuming that the sound-stuttering is caused by the controller and isn't present when no controller driver is operating (or installed).

RoelofBerg, 2015-01-14:
I replaced the PS3 controller by an xbox (USB for PC) controller because I weren't able to get the PS3 controller working properly.

Regarding the sound stuttering:
- I disconnected my bluetooth-dongle for the PS3, left all drivers installed and started limelight with mouse/keyboard only: Stuttering still present !
- I reinstalled everything from scratch without a controller (USB mouse/keyboard only): Video and sound were flawless
- I installed and used xboxdrv: Sound is stuttering and video sometimes too a bit (as I upgraded to limelight V1.2 it is acceptable)
- The next test would be to use mouse/keyboard with the controller disconnected but the drivers installed. (I cannot do this right now without waking up our baby, sorry.)

At the bottomline I wonder wether the existence of the driver allready disturbs the system and the driver - not limelight - needs improvement. Furthermore I wondered wether it could make sence to filter the controller's data (if not allready done) to reduce the data traffic.

I hope I find some time to compare V1.2 with it's predecessor to find out what improved the situation. Maybe I find a solution I can contribute to the driver or limelight. (If anyone can tell me in an abstract fashion what was done in V1.2 and how the controller data is transmitted, that'd help me getting started :)

gforced, 2015-01-15:
nice analysis, i will test the same tonight and let you know if similar.
my xbox controller is wireless however. 
have you tried to over clock and assign GPU more memory also? this helps a little.

RoelofBerg, 2015-01-16:
Thanks for the response. I originally used 256 MB Gfx-Ram and no overclocking.

Today's tests:
- Mouse/keyboard operation while xboxdrv is present and no USB-Controller connected: no stuttering
- USB-Controller operation with no overclocking: sound stutters. When I rotate the right - what is it called ? joystick ? - fast and continuously I even have no sound for a longer period (1 second or so).
- USB-Controller operation with medium overclocking (raspbian): sound still stutters. But when I rotate the right joystick I cannot reproduce the no-sound-event.
- Shaking the controller (producing IRQ-Load of the gyro sensors, I guess) had no impact

Analysis:
So I agree that overclocking has a positive impact on my USB-xboxdrv based system too. Furthermore what I saw on the PS3-Bluetooth system - that the presence of the driver also disturbs mouse/keyboard operation - is no issue with USB-xboxdrv. I wonder if the sensitivity to overclocking leads to the origin, wich might be more at the IRQ- or the network-load - or just the CPU power for data-processing.

RoelofBerg, 2015-01-17:
Addendum: I noticed no difference between the overclock presets 'high' (6V ? This is crazy !) and 'medium'.

RoelofBerg, 2015-01-23:
Unfortunately the software of my TV set is not stable and sometimes looses sound for about 5 minutes that cannot be overcome by a powercycle (hard to believe, I know, but not everything is software in a TV set ...). And the controller-induced sound stuttering of limelight triggers this strange defect of my TV set after a few hours. I never loose sound when watching TV. But I run into this issue when using limelight with a controller.

More experiments:
- I have less stuttering by using V 1.2.1 and the options -Xmn48 -Xmx96m -bitrate 5000
- I tried to use an USB-Over-IP server, but then I saw a terrible stuttering of sound and video when leaving from the steam and game menus to the game itself. Aborted after some minutes. However, as I saw similar effects also without USB-Over-IP - but only during the first seconds - I cannot say yet that an USB-Over-IP server would not solve the issues.

gforced, 2015-01-26:
Has anyone actually been able to narrow the exact reason why this is causing the sound to stutter?
rather than trying to workaround this, we should try to actually identify the root cause of the stuttering..

RoelofBerg, 2015-01-27:
Update: With the latest version and the improved .so file I see no sound stuttering in the game anymore. I also reduced the graphics quality in the game. Possibly the .so file made it, which would then be the root cause asked for. Maybe I didn't install it when trying the predecessor version. (It has to be placed beside the .jar file, then the Java application uses it automatically.)

gforced, 2015-01-28:
great news, where did you get the "improved so file? ?

RoelofBerg, 2015-01-28:
I mean libopus.so (available at https://github.com/irtimmer/limelight-embedded/releases). For V 1.2.1 I just executed
 wget https://github.com/irtimmer/limelight-embedded/releases/download/v1.2.1/libopus.so
while being inside the folder the limelight.jar is located.

Sorry, when updating to V1.2 I just hadn't recognized that there is now a second file to be installed.

gforced, 2015-01-29:
Got it thanks you

---- Roelof Berg wrote ----

I mean libopus.so (available at https://github.com/irtimmer/limelight-embedded/releases). For V 1.2.1 I just executed
 wget https://github.com/irtimmer/limelight-embedded/releases/download/v1.2.1/libopus.so
while being inside the folder the limelight.jar is located.

Sorry, when updating to V1.2 I just hadn't recognized that there is now a second file to be installed.

---
https://github.com/irtimmer/limelight-embedded/issues/70#issuecomment-71938320

closed 2015-07-16 by irtimmer