moonlight-stream/moonlight-embedded #71: Simple GUI with CEC support

mrspidey, 2015-01-06:
I would like to make a suggestion for a future enhancement, whereby you create a simple GUI navigated through CEC using the TV remote where I can start streaming from a pre-saved config (where we can save the server ip, fps and resolution, gamepads maybe too), rather than having to SSH and getting the limelight-embedded to begin streaming. 

What would also be nice is to give users an option to update limelight-embedded when an update is available. Way easier if it is simply done using a few clicks on a remote.

dodslaser, 2015-01-07:
I've managed to write a few scripts that can kill Kodi and launch Limelight from within the Kodi-interface on a raspberry pi. I've tested it myself and it works on Raspbmc and OpenELEC. Since all they do is open the steam big-picture interface there's no CEC though. A Kodi plugin with the options you mention and a games would be fantastic. I was thinking of creating one myself, but I don't really have any experience with writing plugins for Kodi, so it might take a while.

mrspidey, 2015-01-07:
That's quite a nice idea actually. I think my idea is better as I don;t use my PI for Kodi, but it would help people who do use it for Kodi too.

Your idea would mean CEC support thought Kodi, then launching LE thought the plugin. If you want to go back to Kodi just restart the RPI. 

I'd be down for that :)

beakersoft, 2015-01-07:
@dodslaser  i'd be interested in your scripts if you have them somewhere? I'm using limelight on my Raspbmc install, and its a pain having to drop to the command line to start it.

dodslaser, 2015-01-08:
@beakersoft Sure, no problemo!

**limelight.py**

    import subprocess
    subprocess.call( "sudo sh /opt/limelight/limelight.sh &", shell=True )

**limelight.sh**

    sudo initctl stop xbmc &
    sudo sh /opt/limelight/stream.sh &

**stream.sh**

    java -jar /opt/limelight/limelight.jar stream [...]
    sudo initctl start xbmc

I'm not really sure if all the ampersands and sudo:s are actually needed, but this is the only way I could get it to work without Kodi hanging or crashing. From what I can understand Kodi doesn't want to quit before reaching the end of a script. Executing the commands in the background makes sure Kodi can exit without having to wait for limelight to close.

Put the scripts with *limelight.jar* and change */opt/limelight/* to that path. Run *limelight.py* from the Kodi interface. It will close Kodi and open the stream. Once the stream is closed Kodi will restart. If you don't want to use the console at all I recommend adding *snd_bcm2835* to */etc/modules* and adding the path to your java binaries to *PATH* using *~/.profile*. That way everything will load automatically on boot.

You'll also have to fix a bug that prevents initctl from properly killing Kodi. Just open /etc/init/xbmc.conf and replace the line

    test "x" = x`pidof xbmc.bin` || kill `pidof xbmc.bin`

with

    test "x" = x`pidof kodi.bin` || kill `pidof kodi.bin`

HazCod, 2015-03-13:
@dodslaser , any chance you could share your bash history/scripts for the OpenELEC setup?

Dygear, 2015-04-25:
@dodslaser I think you'll need ampersand because the bash process is running a child to the Kodi GUI, and a child can't kill it's parent process. Very much interested in getting a plugin for OpenELEC running for this. The only part that I don't know is how to spawn a process outside of the GUI's influence so you can kill the Kodi GUI, and then on the next line of the script run the GUI again, basically making it so the GUI comes back up once you close Limelight. At that point it will be perfect.

HazCod, 2015-05-13:
FYI, the following script worked for me, even when running from kodi; https://github.com/HazCod/LimeRPi2-kodi/blob/master/README.md

dodslaser, 2015-05-14:
Sorry I haven't been able to reply. I don't think I have the scripts left. A GFE update broke limelight for me, and I haven't really used it since.

The script @HazCod provided looks similar to what I did but more refined. You guys should probably use that! If it works on openELEC it shouldn't be hard to port it to osmc/raspbmc or xbian either. My guess is you only have to change the method for stopping/restarting the kodi service.

dodslaser, 2015-07-18:
I created [moonlight-osmc](https://github.com/dodslaser/moonlight-osmc) for running moonlight from Kodi on OSMC. Since moonlight has CEC support now I might put that into the .deb when I get back home. It's essentially a better version of my scripts (still a hack though, and very much alpha at this point). It should be possible to port it to other dists.

OpenELEC is inherently difficult to port to as it uses a squashfs filesystem. Not sure how one would go about installing all the dependencies without releasing a custom version of OpenELEC altogether.

slawak, 2015-07-18:
I am running moonlight on OpenELEC. The idea is to copy all the needed libraries from raspbian and use LD_LIBRARY_PATH when calling moonlight. This way you dont have to rebuild the whole OpenELEC.
Easiest way is to install both raspian and openelec on one sd card with noobs, build moonlight on raspbian and copy the missing libraries from raspian to openelec data directory. Use ldd to determine which libraries to copy. Then you can use the scripts kodi plugin to start moonlight. When I get some more time I will write down the instructions.
Best would be probably to just put a zip with all the libs somewhere, but I have no idea whether I am allowed to just put some random binary libs from raspbian for download.

dodslaser, 2015-07-18:
@slawak Oh yeah, I remember doing that for libopus. You could probably just fork my scripts and have ```install.sh``` pull the libs from the raspian repo. You would also have to replace the ```dpkg``` line with one that installs moonlight from the tarball, and set the path to install to a writable path in the fs.

slawak, 2015-07-18:
@dodslaser That would involve to find a way to extract the debs on OpenELEC (not even ar exist on it). That would involve kinda to much work for what is still essentially a hack, I think.
New versions of OpenELEC or raspbian might break this anyway, as it is not guaranteed that one gets away with using a few local libs. Even now one has to preload libcurl.so.4 as the version in OpenELEC and the one in raspbian are not compatible with each other.
The proper way would be to build a moonlight kodi module inside of OpenELEC build system. With the emulator integration going forward in kodi somebody might do this.

To give an idea of how much libs you need, here is my script to gather the libs for moonlight
```
cp -av /media/root0/usr/local/lib/arm-linux-gnueabihf/libcec.so.3.0 .
cp -av /media/root0/usr/local/lib/arm-linux-gnueabihf/libcec.so.3.0.0 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libcom_err.so.2 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libcom_err.so.2.1 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libcurl.so.4 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libcurl.so.4.3.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libevdev.so.2 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libevdev.so.2.1.8 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libffi.so.6 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libffi.so.6.0.2 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libgcrypt.so.11 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libgcrypt.so.11.7.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libgnutls.so.26 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libgnutls.so.26.22.4 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libgpg-error.so.0 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libgpg-error.so.0.8.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libgssapi_krb5.so.2 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libgssapi_krb5.so.2.2 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libidn.so.11 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libidn.so.11.6.8 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libk5crypto.so.3 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libk5crypto.so.3.1 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libkeyutils.so.1 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libkeyutils.so.1.5 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libkrb5.so.3 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libkrb5.so.3.3 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libkrb5support.so.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libkrb5support.so.0.1 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/liblber-2.4.so.2 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/liblber-2.4.so.2.8.3 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libldap_r-2.4.so.2 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libldap_r-2.4.so.2.8.3 .
cp -av /media/root0/usr/lib/liblockdev.so.1 .
cp -av /media/root0/usr/lib/liblockdev.1.0.3.so .
cp -av /media/root0/usr/lib/libopus.so.0 .
cp -av /media/root0/usr/lib/libopus.so.0.4.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libp11-kit.so.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libp11-kit.so.0.0.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libsasl2.so.2 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libsasl2.so.2.0.25 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libssh2.so.1 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libssh2.so.1.0.1 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libssl.so.1.0.0 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libtasn1.so.3 .
cp -av /media/root0/usr/lib/arm-linux-gnueabihf/libtasn1.so.3.1.16 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libudev.so.0 .
cp -av /media/root0/lib/arm-linux-gnueabihf/libudev.so.0.13.0 .
```
/media/root0/ is the root of raspbian system

evilsephiroth, 2015-09-27:
testing right now the moonlight module. Should add a PR in 2 days.

posti85, 2017-03-26:
Check out https://github.com/wackerl91/luna. It works fine on OSMC and the doc says it could be easily ported to be use in other kodi based distribution.

dodslaser, 2017-03-26:
@posti85 Luna works fine (last time I checked), and the interface is fine if you want to browse games like any other type of media.

However, the way it interacts with OSMC is bad. I wrote (parts of) the watchdog and launcher script, and it is a hack at best. The correct approach on OSMC is using systemd. This would require a custom package, delivered by the OSMC package manager (app store) to comply with Kodi add-on guidelines.

Scraping box art for games is also problematic. When I used Luna it worked for about 50-60% of my games. There aren't any reliable APIs for this type of thing.

I've spoken with Sam over at OSMC, and I was working on an official solution but that was a while back, and I kinda lost interest since most people seemed to be satisfied with Luna anyway.

Personally I'm using a simple launcher add-on and systemd service to run Steam BP and emulationstation. It's less flashy, but faster and more reliable.

wackerl91, 2017-04-24:
> Scraping box art for games is also problematic. When I used Luna it worked for about 50-60% of my games. There aren't any reliable APIs for this type of thing.

Just leaving this here in case anyone ever wants to build a GUI for moonlight-embedded. As of right now, there are three huge game APIs (at least that I know of): [IGDB](https://www.igdb.com), [TheGamesDB](http://thegamesdb.net) and [GiantBomb](https://www.giantbomb.com). Each of them has over 25k titles across all platforms dating back to the early 80s. Even smaller indie titles are listed, so you'd have to play a lot of pretty niche stuff if you can't find 40-50% of your games on there and if that's the case then GFE wouldn't find it either.

Theoretically, iMDB also lists some games, but they don't offer a proper, public API.

From personal, purely anecdotal experience: both IGDB and TGDB work reliable, though TGDB is usually a tad slow and I have experienced some downtimes, but then again that's exactly why I'd never use a single API but always chain them together - plus in the context of GFE, you can always pull box arts directly from the host (or even NVs servers if you dare) as a baseline and offer additional ones found via the APIs.