moonlight-stream/moonlight-embedded #72: 552 game management error

MarkBadger, 2015-01-08:
Hi 

I am really new to all of this, and I have spent the last two days trying to find a fix, so I would greatly appreciate help.

I can get the pi and the PC to pair, however once i type 

java -jar limelight-pi.jar -720 -30fps 192.168.x.x 
(i am replacing the x's with my ip)
I am getting the 522 error. Can anyone give me some tips on how to fix this? I have reinstalled GFE and I have unpaired and repaired.

Thanks in advance

simonrail, 2015-01-10:
got same problem :(

simonrail, 2015-01-13:
before, i use to get work retropie on other microSD card on my Pi.
i got some problem and the update get the problem off.

and just after my update, i use that : 
sudo apt-get update && sudo apt-get upgrade -y
(it takes at most 20 min to run, ;) be patient )

i dont know if can help debugging limelight too ;)

SiENcE, 2015-01-13:
Why "sudo java -jar limelight-pi.jar stream -720 -30fps" ?

For me it works without being root.

simonrail, 2015-01-13:
this is the command line to start the stream. 
sudo just to be sure but i dont need it too.

and the -720 and -30fps is for debugging, its best to test it at the lowest stream setting.
i use -1080 with no problem now.

( with limelight 1.2, no need to put a " - " before " stream " or " pair " )

SiENcE, 2015-01-13:
I was just curious why 'sudo'. You should't suggest this if not needed.

MarkBadger, 2015-01-14:
Simonrail, you've been a great help. Got it working, i didnt have the latest version of limelight.jar and didnt think to update it. Just waiting for my internet connection to be improved so i can stream without lag. Thanks buddy.

simonrail, 2015-01-15:
do you have any issue with your first try ?

closed 2015-01-16 by MarkBadger