moonlight-stream/moonlight-embedded #73: SocketTimeoutException Error When Attempting to Stream

roymunson2, 2015-01-09:
So I am fairly new to Raspberry Pi and Linux. I have tried compiling the source on my pi without complete success. I always get fatal errors during compilation. I am most likely missing some libraries, but I cannot find documentation showing step-by-step which libraries to install.

Anyways... I was able to get version 1.2 of limelight.jar (released on github) to pair to my PC and use the list command. However, I am getting a time out error when I attempt to stream. Any help would be much appreciated.

```
pi@raspberrypi ~ $ sudo java -jar limelight.jar stream 192.168.1.30
Starting Steam
java.net.SocketTimeoutException: connect timed out
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:345)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at com.squareup.okhttp.internal.Platform.connectSocket(Platform.java:105)
	at com.squareup.okhttp.Connection.connect(Connection.java:150)
	at com.squareup.okhttp.Connection.connectAndSetOwner(Connection.java:169)
	at com.squareup.okhttp.OkHttpClient$1.connectAndSetOwner(OkHttpClient.java:119)
	at com.squareup.okhttp.internal.http.RouteSelector.next(RouteSelector.java:134)
	at com.squareup.okhttp.internal.http.HttpEngine.connect(HttpEngine.java:314)
	at com.squareup.okhttp.internal.http.HttpEngine.sendRequest(HttpEngine.java:237)
	at com.squareup.okhttp.Call.getResponse(Call.java:233)
	at com.squareup.okhttp.Call.execute(Call.java:84)
	at com.limelight.nvstream.http.NvHTTP.openHttpConnection(Unknown Source)
	at com.limelight.nvstream.http.NvHTTP.openHttpConnectionToString(Unknown Source)
	at com.limelight.nvstream.http.NvHTTP.getServerInfo(Unknown Source)
	at com.limelight.nvstream.NvConnection.startApp(Unknown Source)
	at com.limelight.nvstream.NvConnection.establishConnection(Unknown Source)
	at com.limelight.nvstream.NvConnection.access$300(Unknown Source)
	at com.limelight.nvstream.NvConnection$1.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:744)
connect timed out
Connection Error Starting Steam failed
pi@raspberrypi ~ $ 

```

simonrail, 2015-01-11:
try to play with lower resolution.
java -jar limelight.jar stream -720 -30fps

you dont have to sudo at the begining and no IP at the end after you already have pair it with the computer.

roymunson2, 2015-01-12:
simonrail thanks for the quick reply and tips. I have resolved the issue by reflashing the SD card with the latest version of NOOBS (and Rasbian). This allowed my to use the pre-compiled limelight.jar to stream without any errors. The original Rasbian was provided on the SD card from the Raspberry Pi kit I bought on Amazon. The files were dated June 2014. I did perform apt-get update and apt-get upgrade previously, but I still got the error in my original post. I just wanted to summarized for anyone else that might have this problem.

closed 2015-01-12 by roymunson2