moonlight-stream/moonlight-embedded #74: error code during decode -6

simonrail, 2015-01-11:
I was gaming at civilisation 5.  -1080 -30fps
the game freeze and bring me back to the terminal with the error.
[image]

the game is still running on my computer.

is there a way to update something or settings to change to correct this ?
or 
the futur update of limelight pi is the only way to correct this ? ;)

dodslaser, 2015-01-12:
You seem to be be overloading the pi (error -6) and ending up with a segfault, which could be due to several things. Could be a memory thing. Launch java like this to prevent it from grabbing too much memory for the JVM:

    java -Xmn48m -Xmx96m -jar limelight.jar [...]

Another tip is to limit the bitrate:

    [...] limelight.jar [...] -bitrate [bitrate in kbps]

8192 seems to be the magic bitrate for me. I'm using cat.6 ethernet.

simonrail, 2015-01-13:
whats for -Xmn48m ? do i need to put them both ?
do i need to use this code everytime ?

dodslaser, 2015-01-13:
*-Xmn48m* sets the minimum heap size to 48mb. You don't have to set the minimum, but setting a lower limit makes the JVM start faster since the heap doesn't have to grow right away.

simonrail, 2015-01-14:
work like a charm ;)
display do not glitch anymore. 
previous bug get away.
just a little bit laggy ( i need to adjust bitrate surely )

but

i get a hard crash after 30 min of perfect stream play. My screen of my Pi is completely frozen and i am unable to return to the terminal.
hopefully, my computer is okay and i saved my game ;)

do you think this is about JVM ? allow more mb ?

dodslaser, 2015-01-14:
Not sure. Are you using the 256 or 512mb pi? You could also try a slight overclock. **Just make sure to pick up a heatsink kit.** Here are my settings from *config.txt* on my 512 pi:

    gpu_mem_512=256
    arm_freq=950
    core_freq=450
    gpu_freq=350
    sdram_freq=500
    over_voltage=2

simonrail, 2015-01-15:
im using the b+ model 512mb. I overclock it at modest. I do not have any heatsink.
ill seek to buy one ;) and overclock it more.

ty for your config. ill try that for sure.

dodslaser, 2015-01-15:
No problem. I have this config running raspbmc 24/7 for watching TV and movies with no heat/stability issues. I haven't had time to test limelight for longer periods though, so you might have to tweak the bitrate and JVM heap a bit to make it more stable over time.

The bitrate could probably be lowered quite a bit without too much impact on the quality. 8192 was the highest number I could get without audio stutter or visual artifacts, but I only ran my tests for about ten minutes.

As for the JVM heap, with a 512 pi using a 256 split and no other apps running you should be able to get higher than 96m, but for some reason I crashed a lot when I went past 100m. 

Good luck!

dodslaser, 2015-01-15:
By the way, are you using OpenJDK or Oracle Java? I've found Oracle's JDK (8u6) to be both faster and more stable than OpenJDK.

simonrail, 2015-01-15:
im using oracle-java7-jdk
ill try to bitrate 500 kbps and check for the freeze. If not, ill get it higher until the crash.
ill share my number later ;)

simonrail, 2015-01-18:
okay ! 
i played 3h on civ5 with my RPI and no lag, no glitches and no freeze.
My RPI is overclocked at Modest
My code :

java -Xmn48 -Xmx96m -jar limelightpi,jar stream -bitrate 5000 -1080 -30fps

RoelofBerg, 2015-01-24:
Thanks for sharing. I also see the video freezing every about 15 minutes when playing Wolfenstein. However I see no error message on my ssh console. Ctrl+c and restarting limelight helps.

Oracle Java 8 and
java -Xmn48 -Xmx96m -jar limelight.jar -mapping xbox.map -bitrate 5000 -30fps stream ...

The system runs Raspbian, has a heatsink and is slightly overclocked.

simonrail, 2015-01-24:
does the game freeze on your computer ?
if not, try using oracle 7 and the 1.2 version. ;) i dont think that might be the solution but it worth trying.

try to stream at 720 instead of 1080. and lower your game settings.

a new version ( 1.2.1 ) is available today. try it if it solve freeze problem.

RoelofBerg, 2015-01-25:
The PC's process does not freeze. Only the image stream in limelight freezes. I can still use the controller: After the image is frozen I can press the button that would interrupt the game by showing a game menu. Then after restarting limelight I can see the menu - so the controller button had been processed (invisibly).

I used the most recent software with most conservative settings (V1.2.1, 720p, 30 fps, game options far below my gfx-card capabilities). Thanks for the hint to use Java 7. Unfortunately I weren't able to execute V1.2.1 on Java 7. I will downgrade to V1.2 and Java 7 in the next days.

Does limelight have a debug log ? How can the debug information can be turned verbose ?

simonrail, 2015-01-25:
the only way to see error log, is by the Pi terminal. 
But when it freeze ,the screen stay on the game and i cant see the terminal.
Could be cool to have terminal mirror on desktop like "Putty" style. If someone have a soluce for that ;).

RoelofBerg, tell me if using 1.2 limelight and oracle 7 solve your problem ;)

RoelofBerg, 2015-01-26:
You can 'mirror' the terminal by using screen. https://help.ubuntu.com/community/Screen. (Enter 'screen' before starting limelight.jar and enter 'screen -r' in ssh to access the terminal.)

I downgraded to 1.2 and Oracle 7 but it is mandatory to also downgrade GFE. I'm not sure if this is even possible, I will check it out.

closed 2015-05-10 by irtimmer