moonlight-stream/moonlight-embedded #75: 3rd party 360 controller problem

cinphart, 2015-01-12:
I have a 3rd party (PDP) wired 360 controller that I am trying to use with limelight on a raspberry pi. I having issues particularly with the triggers - once I active the right trigger, it keeps firing repeatedly. I see this in both Deus Ex: HR and Portal 2 (haven't tried any other games).

Any ideas for what I can do to debug this issue? I've looked around online, and trying using xboxdrv with and without a mapping file, but nothing seems to change this behavior (except in one case, where the controller was registered but refused to do anything at all).

The only other device I have attached to my pi is a usb bluetooth dongle, that is used for the keyboard/mouse.

simonrail, 2015-01-12:
maybe in this. Check for mapped turbo key :

cd .emulationstation
sudo rm es_input.cfg

or : 

cd RetroPie/emulators/RetroArch/installdir/bin
./retroarch-joyconfig >> ~/RetroPie/configs/all/retroarch.cfg

simonrail, 2015-01-13:
i think is the steam config the problem. 
try to check how to configure controller on bigpicture, not on the pi

:( i cant help you more, i didnt try the controller

7hunderbug, 2015-02-13:
Perhaps you need to use the

    --trigger-as-button

config option for xboxdrv?
This worked for me using MS Xbox 360 wireless controller.

luckyguido, 2015-02-18:
As far i can see from the code it should be a problem of xpad or xboxdrv send  trigger value, this vaule has range from a negative value to a positive value but the game expect trigger value as 0 positive value.
I tried to modify EvdevHandler.java to avoid negative value sent from trigger and games work flowlessy also without --trigger-as-button .
example (i did same thing also for abs_rz): 

if (code==mapping.abs_z)
{
      if(absLT.getByte(value) <0)
            leftTrigger = 0;
      else
            leftTrigger = absLT.getByte(value);
}

A more sofisticated solution should be to find (or let user configure) a calibration number to have a full range of movement from trigger and not just half as my change does.

cinphart, 2015-02-18:
Thanks - I'll take a look at both of these options tonight.

cinphart, 2015-02-19:
I noticed there were some changes related to trigger handling in the latest build, so I built from that. The current changes seem to fix the trigger problem, however, the axes for the thumbsticks now seemed to be tied to always create positive values, so you can only move in one direction in each axis (I think it was right and back, but can't remember now).

closed 2015-07-16 by irtimmer