moonlight-stream/moonlight-embedded #77: Increase sound output and no multiple key presses using USB keyboard

georgcmef, 2015-01-24:
Hello, first thank you for such a awesome port. 
I am having a low volume issue where my PC output is at 100% and my TV volume is at 100% and I can hardly hear the sound from my tv. I is there a way to increase this volume more?

Also, I am unable to have 2 or more buttons pressed on a keyboard at a time. For example, I can not hold shift and press a letter to capitalize it, or shift + a number to use a specific symbol.

scoobyd00, 2015-01-30:
use alsamixer on the pi to increase the volume

georgcmef, 2015-01-31:
awesome thank you that helped, any idea on how to allow multiple key presses or even holding a key down, like the backspace, i have to press and release each time, i cannot hold it down

scoobyd00, 2015-02-01:
Sorry cant help with the multi keypresses, i use a xbox 360 wireless controller on my pi

closed 2015-07-16 by irtimmer