moonlight-stream/moonlight-embedded #79: Video stutter every 20seconds or so

scoobyd00, 2015-01-30:
Ive been using limelight on my PI and have had good performance, since updating to the latest release ive started to get video issues whereby every 20 seconds or so the graphics go from being butterly smooth to lagging, the lagging lasts for about 3 seconds or so and then it becomes butterly smooth again.

Any ideas?

scoobyd00, 2015-01-30:
Tried lowering bitrate to see if that makes any difference but it doesnt.
For information i have a xbox wireless receiver connected to my pc and ive disabled the relevant files in the nvida folders.
When the video stuttering occurs the sound is still fine and doesnt stutter.

closed 2015-01-30 by scoobyd00

Haris1977, 2015-12-06:
Same problem but not every 20 sec. I would say it happens in a random way for me. I think it must be looked.

Have opened a relevant issue - and other ones - here 

https://github.com/irtimmer/moonlight-embedded/issues/252

peterden, 2016-03-28:
I have the same issue, using a samsung galaxy note 3, over a 5ghz wifi network. It stutters randomly, not exactly every 20 seconds, but stutters about 3 seconds, the audio isn't stuttering at all when the video does.