moonlight-stream/moonlight-embedded #80: java - Unsupported major.minor version 52.0

bfenty, 2015-02-04:
I am having trouble when I try to run the java -jar limelight.jar pair IP ADDRESS command. I use my PC's IP address, and I get the error 
```
java - Unsupported major.minor version 52.0.
```

I've updated my java version to 1.8.0 and installed the v8 JDK. Other than installing Limelight, this is the only modification to a fresh install of Raspbian from yesterday. Though I doubt it matters, I'm using Berryboot with this install.

Specifically, I followed the setup guide <a href="http://embedded-software-architecture.com/wk/index.php/Limelight_on_a_Raspberry_Pi">here</a> to get it set up. I've used Limelight on Android with this same PC without trouble, I'm sure if I could solve this Java issue I'd be up and running!

dodslaser, 2015-02-05:
What's your output of *java -version* ?

bfenty, 2015-02-05:
```
pi@raspberrypi ~ $ java -version
java version "1.8.0-ea"
Java(TM) SE Runtime Environment (build 1.8.0-ea-b36e)
Java HotSpot(TM) Client VM (build 25.0-b04, mixed mode)
```

I'm using SSH if that matters as well

BmanDesignsCanada, 2015-02-14:
I am getting this as well on a pi. Java Version 1.7.0_40
Brand new install. Limelight was the only thing I did after installation

bfenty, 2015-02-14:
Bman-I believe you will need to update to the latest Java jdk:

Sudo apt-get install oracle-java8-jdk

cgutman, 2015-02-16:
I think the issue is that 1.6/1.7 needs to be targeted in the javac invocations in build.xml. By default, it's targeting the version of Java that irtimmer built the Jars with.

<javac ... source="1.6" target="1.6" ... > should solve it.

bfenty, 2015-02-16:
Yea that's over my head lol

tcadd, 2015-02-25:
Having the same issue, and with the previous release (1.2) I get virtually the same error except the minor version number is 51.0 - Although my Raspian is loaded up with other programs.

tcadd, 2015-02-26:
And finally: Solved. Incorrect Java version being called - Use:

sudo  update-alternatives --config java

And select the option for:

 /usr/lib/jvm/jdk-8-oracle-arm-vfp-hflt/jre/bin/java
(3 on my system)

Not sure if there would be issues with removing the other packages so I am not doing so at this time.

closed 2015-05-10 by irtimmer