moonlight-stream/moonlight-embedded #81: Raspberry pi 2

scoobyd00, 2015-02-06:
Will limelight need updating to use all 4 cores on the new raspberry pi 2?
We should really a big performance boost hopefully?

7hunderbug, 2015-02-06:
The most recent Limelight embedded (with GFE 2.2.2) already works very 
well indeed on the Raspberry Pi 2 - I have none of the issues I was 
experiencing on my early Raspi model B's.

I've not checked if it is using all cores but will do ASAP.

scoobyd00, 2015-02-06:
Excellent news,  fingers crossed its using all 4 cores.  Streaming should be alot smoother and maybe we will be able to do 1080p!

simon84, 2015-02-08:
I have following problem with the RPi2. The video stream starts correctly, but sound is not working. I'm using a fresh installed raspbian image and tried using the precompiled opus lib and a self compiled one.

simon84, 2015-02-08:
I want to add, that with RPi B+, I had a working audio and video stream.

scoobyd00, 2015-02-08:
Just got my PI 2 up and running with limelight, no sound issues. Have you tried alsamixer ?.
Ive got the same setup, fresh install of Raspbian.

Streaming seems better, no sound issues etc, not sure if its using all 4 cores??. maybe irtimmer will be able to help with this one?.

simon84, 2015-02-08:
Sorry, my fault. After restarting the steam of, everything works fine!

brainstormi, 2015-02-10:
Could anybody confirm if raspberry pi 2 is able to stream at 1080p 60fps smoothly?

Flippage, 2015-02-10:
I would think that the streaming would be struggling at 1080p / 60fps, but not due to the Pi itself, but rather due to the Ethernet. I would say you might need to get a USB->Gigabit Ethernet adapter. Since the Pi is only USB2, this wouldn't be true Gigabit, but it should allow the Ethernet to get up to speeds of around 450Mb/s rather than 100Mb on the built-in Ethernet.

I would still suggest something like, 1080p/30fps, or 720p/60fps. But then again, I haven't tested with the Pi 2 yet, and it's all subjective until irtimmer responds with more information about limelight-embedded on the Pi 2.

brainstormi, 2015-02-11:
I don't think that ethernet bandwidth should be a problem... Initially if you are looking for the lowest latency possible with an acceptable video quality, with a stream of 8-9 mbps should be more than enough. For a good quality stream we are talking around 20mbps... Anything above that, would penalize latency heavily, not improving the image quality at the same rate. If I'm not wrong, limelight by default sets up a 20mbps stream.

Any of the users of rpi2 could check if it's able to sustain a limelight stream at 1080p 60fps properly? With the old B+ 720p/60fps runs flawesly, but 1080/60 is too laggy to be playable. Not sure if it's due to VideoCore GPU processor or to CPU. This is the reason why I would like to know how rpi2 behaves.
Regards,

scoobyd00, 2015-02-11:
Just tried running GRID at 1080/60fps and i get serious input lag on the PI 2, same as the original PI, its unplayable. My old B+ had issues with 720p/60fps on GRID, sound stutter, dropping frames etc but the PI2 runs it flawless. 
Looks like 1080 is a no no until we get some word from irtrimmer if a new version is needed?..

cgutman, 2015-02-11:
The CPU in the Pi 2 is much improved over the original Pi, but the GPU and decoders are the same. VideoCore IV doesn't support 1080p60, so it still won't work on the Pi 2. The extra CPU performance should reduce or eliminate stuttering though (especially related to controller input).

scoobyd00, 2015-02-11:
Thanks for the update cgutman. Limelight is a great piece of software, thanks for bringing it to us.
The PI2 has helped with stuttering and sound issues and mine now plays most things at 720/60 flawlessly.

brainstormi, 2015-02-11:
Long time ago I tested to play 1080p60 with a B+ and it worked well... although it was only a 4/5 mbps stream from a youtube video. It's true I wasn't able to check the latency, something that it isn't  very important when you reproduce a movie (buffering is your friend...), but criticial for a game stream. I think the limitation to 1080p/30fps in Videocore was only for encoding purposes, not decoding.

7hunderbug, 2015-02-11:
I've just tested RPI 2 with the command line options for 1080p@60fps (-1080 -60fps) and played Alien Isolation for 15 minutes. I was using a wireless xbox360 controller (with USB receiver attached to the Pi) using the xboxdrv application, with both my Pi and my gaming PC connected over wired network connections and the performance is absolutely flawless. I also monitored the CPU usage whilst playing by SSH'ing into the Pi from another computer and running top and then htop and it certainly appears that the java process is making use of all 4 CPU cores.

scoobyd00, 2015-02-11:
It could be that my pc gpu isn't up to the task of 1080, I've only got a gtx660.  I'm also running wireless from October router then Etherley into the pi.  I'm on wireless n with rock solid connection though.  
I'll do some more testing tomorrow

7hunderbug, 2015-02-11:
I reckon the GTX660 should cope with 1080p streaming. Mines a GTX670 so not a huge difference AFAIK.

Which controller are you using? I'm almost certain that a lot of the stuttering and lag issues that people have experienced are caused by particular USB devices attached to the Pi. For example, my bluetooth dongle and Wiimote set up caused much audio stuttering for me and as soon as I changed to an Xbox360 wireless for PC controller it made a huge difference. Also, my mini Ralink chipset USB wifi dongle simply kills limelight performance, whereas my larger Atheros based TPLink USB wifi dongle works very well.

brainstormi, 2015-02-11:
GTX 6XX onwards shouldn't have any impact/or very low asociated to encoding performance, in theory the encoding process is done by dedicated hardware encoder in those cards, not using any other GPU resources for that purpose. I'm in line with 7hunderbug, that the issue could be caused for other USB devices for example, or the Opus library/Java versions, Wireless connectivity instead ethernet... In my case with the old Pi, I had to connect XBOX360 wireless receiver directly to the Host PC. Connecting to rPi was causing lag issues for me...
Anyway thank you everybody for your feedback... rPi2 is coming home :)

cgutman, 2015-02-11:
From the Broadcom site: "1080p30 Full HD HP H.264 Video Encode/Decode"

It could be that VideoCore IV supports 1080p60 decode but maybe only at baseline or main profile, not high profile CABAC-encoding. Video decoding hardware is far more intricate than a blanket "1080p30 Full HD" makes people believe.

scoobyd00, 2015-02-11:
I've got my xbox controller plugged directly into my pc,  I'll try connecting it to the Pi tomorrow to see if that makes any difference. I've got no other devices plugged into the Pi USB ports.

SirDigbyCC, 2015-02-12:
I've been testing 1080p60 on the Pi for the last couple of days now, and whilst it is definitely an improvement over the Model B, it's still not quite perfect.

Pi: Running Raspbian, Pi2 overclock selected in config, 256MB allocated to GPU, only keyboard connected, using 360 wireless controller connected to host.

PC: Core i5 2500k @ 4.5GHz, SLI GTX 970

It does actually run at 60fps very well, there is almost no latency whatsoever, but the killer is the quality of the stream, at the default bitrate it is very blocky, upping the bitrate to about 16mb mostly eliminated that but the Pi2 can't quite keep up and starts to drop frames, resulting in a garbled image, or crashing back to Terminal.

Also on my first installation I had issues with sound stuttering, but I was using an older version of Java, hasn't happened with the latest version.

For fairly basic looking games it is perfect, anything with plenty happening on screen though is not so good, games tested:

Resident Evil 1 Remake: Perfect
Deus Ex HR: Perfect
Grid 2: Almost perfect, slightly blocky video but not as noticeable as I'm staring at the road 90% of the time.
FarCry 4: Unplayable, low bitrate looks awful, high bitrate causes frame drops / crashes.
Sniper Elite 3: Again quite blocky, upping the bitrate mostly solves that problem, but quickly zooming in and out of the scope instantly causes frame drops.
Assassins Creed 3: About the same as FarCry 4, too much foliage!

Still quite impressive for a £25 device, maybe there's more to tinker with that I haven't tried yet (packet size?), but just thought I'd let you all know how I got on with it. In the meantime 720p60 works great, and setting the Pi itself to run at 720p60 makes the image look... not as bad.

SirDigbyCC, 2015-02-12:
Given it another test on a fresh install, sound still stutters at 1080p60, even with the bitrate set to 4000, hmm, it's worth noting that it is only the sound that is doing it, there is no noticeable stutter on the video side. Might give Archlinux

jaimie-van-santen, 2015-02-12:
How did you guys manage to get it running? I've added the libopus,so from limelight-embedded and still get: 
```
osmc@osmc:~$ ./limelight.sh
Starting Steam
Starting RTSP handshake
Starting control connection
Starting video stream
Starting audio stream
ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM sysdefault
```
With the rPi1 it was simply a matter of `sudo modprobe snd_bcm2835`.

roseyhead, 2015-02-12:
SirDigby, why set bitrate to 4000? I've been setting mine to 20000-30000 with no noticeable issues.

SirDigbyCC, 2015-02-12:
Just to test whether or not the higher bitrate was causing the sound stuttering.

roseyhead, 2015-02-12:
Ah my bad, I misread. Sorry about that.

fanatictentacle, 2015-02-13:
Performance is better with the rpi 2, but yes input Lag is way too much with 60fps. But i have noticed, that the New rpi seems to overscan in limelight. The boot\config settings are correct, no overscan in tty1 but when the stream starts, i get overscan. Has anyone else this problem? 
How can i change this?

SirDigbyCC, 2015-02-13:
How are you guys configuring your audio, I've tried everything I can think of and I can't get rid of the audio stutter / skipping after a couple minutes of gameplay, even when running at 720p30 at very low bitrate.

My process:

Write latest Raspbian image to SD
Skip installing oracle-java8-jdk as it is included in latest Raspbian build, same for libopus0
Copy libopus.so from releases page to home directory
Copy limelight.jar to home directory
Pair
Stream with command: "java -jar limelight.jar stream [IP] -1080 -60fps -bitrate [Bitrate]"

I've also tried with Xmn48m Xmx96m set, which makes no difference, disabled all non essential running processes on host PC including AV, disabled SLI, tried with the Pi overclocked & at stock, and multiple TVs / amps. Also I've tried using the instructions on the front page regarding fixed point Opus, which leaves me with a 16B libopus.so file (is that normal?)

It's just strange that it is only the audio that is causing me problems atm, I have virtually no input lag, to a degree I can put up with the video artefacts, or going back to 720p60, but the sound is a killer. :(

scoobyd00, 2015-02-13:
Been doing some more testing this afternoon. Ive found that if i plug my xbox wireless controller into the PI then I get better performance then if its plugged directly into the PC (host)
With it plugged into the PC and the necessary files renamed in nvidia i couldnt play grid autosport at 1080 60fps at all due to input lag.
With it plugged directly into the Pi and running xboxdrv it plays fine at 1080. 720p is very smooth whereas before i got very slight jitter with it plugged into the PC.

SirDigbyCC -

Im running exactly the same as you. Latest raspbian (apt-get update etc) . Limelight.jar and libopus.so in the home directly.
PI is connected to my AV amp via hdmi for sound and video.
I just use the command java -jar limelight.jar -720 -60fps -app (name of game) stream (ip address)

This works fine and i dont get any sound issues, been playing Grid Autosport this afternoon and it was fine.
Not sure why you are getting sound issues, normally sound issues are caused by network issues. Are you on ethernet?, im running wireless N from PC to router then into pi via ethernet.

SirDigbyCC, 2015-02-13:
Yep running on 100mbit ethernet, off the same switch as the host in one test.

Even tried it on 720p30 @ 1mbit and the stuttering still occured so I think it's safe to say it's not the choice of bitrate / resolution that's the problem. Very weird.

7hunderbug, 2015-02-13:
I've done some testing today as well. People may want to experiment with different Nvidia driver versions. Previously I was running 340.52 drivers with GFE 2.2.2.0 and could play all games perfectly in 1080p@60fps. I upgraded all NVidia drivers today, including reinstalling GFE 2.2.2.0, (using "Clean install" option) and now I'm running 347.52 drivers and there is about a 1 second lag in the video stream only. Input from the 360 controller connected to RPi is virtually latency free - noticeable by watching the PC screen, however the RPi video lags behind the PC video. 720p@60fps is fine though. There is also virtually no audio stuttering or interference to speak of in either 1080p or 720p with the new gfx driver install on my setup.

@SirDigbyCC - rather than explicitly specifying the bitrate, try not specifying the bitrate at all. Also, try removing all USB devices from the RPi (including the keyboard just after hitting enter to run limelight), then navigate the Steam menu using your PC keyboard/mouse and see if the audio still stutters.

SirDigbyCC, 2015-02-14:
I'll give that a try tomorrow 7hunderbug, I've already tried running with no bitrate defined and still got the issue, but I haven't tried removing the keyboard after launching, worth a shot I guess. Worth pointing out though that the keyboard isn't used at any point after launching Limelight, nor is there any controller connected to the Pi as I use a wireless 360 controller connected to the host.

robharty, 2015-03-12:
@sirdigbycc you might want to check the settings on your router.

I had similar issues to you until this week. I run a rpi2 on raspbian (no overclock) with my GTX 750 Ti PC upstairs connected to the router via homeplugs. Xbox360 controller attached to the rpi which is hooked up to the downstairs TV.

The PC can handle the games I play with ease but I was still getting regular audio and video stuttering after a few minutes of play.

I'm in the UK and have a virgin media superhub 2 router. There were some router features enabled like firewall and ip flood protection that appear were getting in the way of the stream. 

Disabling those have made the streaming absolutely flawless. Although I only run 1080p @30fps. I get terrible input lag @60fps. 

Are those who are able to run @60fps overclocking their rpi2? What amount of memory are you allocating to the GPU?

Also, any plans to support the steam controller when it arrives in Nov?

stsfin, 2015-03-25:
RPI 2 runs 1080p@30FPS flawlessly, but on 60FPS it causes ~½second screenoutputlag, the input is realtime checked from the host computer just the picture lags horribly. 720p@60fps works also fine... This should not be an hardware power issue since the android build can do 1080p/60fps with onpair hardware.

stsfin, 2015-03-26:
Anyone tested to overclock the h264_freq setting? I will try this later today, it could be doubling it to 500mhz could fix the issues.

stsfin, 2015-03-26:
So i tested to over clock the H264 setting and the Pi, that did the trick... Now 60fps 1080p works smoothly:

here are my OC settings

arm_freq=1100
core_freq=500
sdram_freq=500
h264_freq=500
over_voltage=8
force_turbo=1
temp_limit=80

DISCLAIMER: Force_turbo will void your warranty since its keeps the clock on 100% without throttling, so you should use a heatsink with these settings.

roseyhead, 2015-03-26:
stsfin, exciting news! What bitrate/other settings are you using?

stsfin, 2015-03-27:
The default settings... played crysis, sonic generations and a few other
games... they work nicely... it seems that the colors are also more
accurate and there are less compression artifacta when the h264 freq gets
pumped

stsfin, 2015-03-27:
Just to add Im running the Retropie debian build with emulation station, and I start Limelight from emulation station when its running (it does not affect performance)

Here is a small guide I wrote how to install Limelight to Retropie/Emulation station: https://github.com/petrockblog/RetroPie-Setup/wiki/Quickguide-for-configuring-limelight-with-Retropie-and-Emulation-Station

stsfin, 2015-03-27:
Also created a install script for retropie: https://github.com/stsfin/RetropieLimelightInstaller

stsfin, 2015-03-27:
I forked a generic installer and configurator of the Retropie script: available here: https://github.com/stsfin/LimelightPiInstaller

cgutman, 2015-04-05:
Is it possible to run 1080p60 on a Pi by just applying the h264_freq=500 fix? If it's possible to run on a stock Pi (no heatsink), it might be useful to add to the readme.

stsfin, 2015-04-05:
Mayby, havent tried it on a stock device.

cgutman, 2015-04-07:
I did testing on my Pi B+ and Pi 2 with limelight-embedded 1.2.2.

I was able to get playable 1080p60 without voiding the warranty or requiring a heatsink using the following settings:
force_turbo=1
h264_freq=350

Without over_voltage > 0, force_turbo=1 still preserves the warranty as per http://www.raspberrypi.org/forums/viewtopic.php?p=176865#p176865

stsfin, 2015-04-07:
Good to know,

I would still recommend heat sinks with force_turbo since all the cores are
running on full-speed.

Jonne Backhaus
______________
+358-45-1141905
jonne.backhaus@gmail.com
jonne.backhaus@androidsuomi.fi

Google+: http://backhaus.fi/+/
LinkedIn: http://backhaus.fi/linkedin/

scoobyd00, 2015-04-07:
Would be interested to know if heat sinks are required?

closed 2015-05-10 by irtimmer

diagonali, 2015-07-30:
Wondering if allocating more memory to the GPU would reduce slight stuttering now and then? Or is Moonlight CPU bound for the most part? 

I'm getting buttery smooth settings using the overclock from cgutman and these but slightly noticeable stutter seemingly at random:

moonlight stream -mapping xbox.map -localaudio -1080 -60fps -bitrate 30000 192.168.0.90

Sharker, 2015-08-02:
On Moonlight 2.0.2, tuning settings from cgutman not working for 60 fps. On 1.2.2 working very fine but now stream are very bad.