moonlight-stream/moonlight-embedded #82: Steam launches successfully but games won't start

roseyhead, 2015-02-07:
I'm running the latest version on a RPi 2B and it loads the Steam interface just fine but it won't actually launch any games. I get the screen with the big Play button and hit it, then the screen fades out for a moment and fades back in to the same screen.

I don't see any related errors generated by Limelight in the logs on my rpi but on the PC I'm streaming from I found that it generated a folder named NVIDIA.SteamLauncher with a file named Log.txt with the following information:

22:13:11 [0x00000D6C, ERROR] Launcher::Launch threw an exception
Failed to handshake PrepareAppLaunch request

so I went and found the configuration file for NVIDIA.SteamLauncher.exe and found I could change the logging level so I set it from Error to Debug and it gives me the following information:

00:55:05 [0x00001C34,  INFO] Logger successfully initialized at 'D:\Steam\steamapps\common\The Talos Principle\NVIDIA.SteamLauncher\Log.txt'
00:55:05 [0x00001C34,  INFO] Running Version = 1.1
00:55:05 [0x00001C34,  INFO] SteamLauncher input parameters (count = 2) = 
00:55:05 [0x00001C34,  INFO]   0 = C:\Program Files\NVIDIA Corporation\NvStreamSrv\SteamLauncher\NVIDIA.SteamLauncher.exe
00:55:05 [0x00001C34,  INFO]   1 = D:\Steam\steamapps\common\The Talos Principle\Bin\Talos.exe
00:55:05 [0x00001C34,  INFO] Initializing static connection
00:55:05 [0x00001C34,  INFO] Launching in 'Shield' mode
00:55:05 [0x00001C34,  INFO] Entering Launcher::Launch. Args = port=3000, address=127.0.0.1, path="D:\Steam\steamapps\common\The Talos Principle\Bin\Talos.exe"
00:55:05 [0x0000083C,  INFO] Connection thread started
00:55:05 [0x00001C34,  INFO] Requesting to prepare for launch
00:55:05 [0x00001C34,  INFO] Sending message : action: "PrepareAppLaunch"
prepareAppLaunchArgs {
  selfProcessId: 2148
  versionInfo: "v1.0.0.0"
  appId: "257510"
  appPath: "\"D:\\Steam\\steamapps\\common\\The Talos Principle\\Bin\\Talos.exe\""
  appArgs: ""
}
00:55:05 [0x00001C34,  INFO] Request sent successfully
00:55:05 [0x00001C34,  INFO] Waiting for response back to launch the app
00:55:20 [0x00001C34, ERROR] Launcher::Launch threw an exception
Failed to handshake PrepareAppLaunch request
00:55:20 [0x0000083C,  INFO] Connection thread voluntarily exited
00:55:20 [0x00001C34,  INFO] Shutting down logger

I've tried The Talos Principle, Team Fortress 2 and Bioshock Infinite and they all return the same "Failed to handshake PrepareAppLaunch request" error. Bioshock works perfectly from the Limelight app on my Android so I'm sure it's not a game compatibility issue. Is this a known issue? I've been googling around all morning and haven't found a working fix for it.

roseyhead, 2015-02-07:
Apparently this was an environmental issue. Streaming fails if Musicbee is open, possibly my remote access plugin is causing a port conflict or something. Close that and streaming starts up as expected.

closed 2015-02-07 by roseyhead

zer0future, 2015-04-03:
u also can start a game from rpi with the -app argument when starting limelight.
-> java -jar limelight.jar stream -app NAMEOFTHEGAME
The Game name is listed in the GFE setting on your Host Computer. Limelight shows the Games when  you start limelight with:
-> java -jar limelight.jar discover 
And the name's can be changed like u want, by adding games in GFE -> GameStream

praij90, 2016-01-13:
I'm having this issue right now...
Its very strange, when I start the game up using the `-app` modifier and the name of the game it works fine.
When running `moonlight stream my.ip` steam starts up without any problems. However, when attempting to launch any of the games it indicates the game is loading. Shortly after (2-5s) the game fails to load without any indication of an error.

I have tried several games that do work when addressed directly via the `-app` modifier and do not work when started through steam.

Please advise and thanks in advance.