moonlight-stream/moonlight-embedded #83: Very poor audio/stuttering on clean install (no controller)

modusdigital, 2015-02-14:
Hello,

I've just set up Limelight Embedded v1.2.1 on my Raspberry Pi running a clean install of Raspbian, which was updated before installing Limelight. I got Limelight by running the following:

mkdir limelight
cd limelight
wget https://github.com/irtimmer/limelight-embedded/releases/download/v1.2.1/limelight.jar
wget https://github.com/irtimmer/limelight-embedded/releases/download/v1.2.1/libopus.so

So I'm pretty sure I'm up to date with the latest version. I have paired my devices, and the gameplay/video is all running smoothly. However, the audio is awful, with continual stuttering and bubbling sounds. At first I thought this might be a controller issue, so I tried moving the controller to my PC, but that hasn't made any difference. After trying a few different options I eventually wiped the Pi again and started from scratch, but only using KB + mouse instead of the controller. Unfortunately I still get the same problem.

I am running this on a Pi 2, using wifi connections on both my computer and my Pi (I don't really have any option of running a wired connection unfortunately).

Can anyone help?

SirDigbyCC, 2015-02-16:
Same here as I posted in the Pi 2 thread, issue occurs at all resolutions & framerates & with no controller attached to the Pi.

7hunderbug, 2015-02-16:
Does anyone think it would be helpful to gather people's setup details in a table? It's a shame github doesn't have a basic wiki.... 

Something like the following?:

1. Windows OS version (32/64bit)
2. NVidia graphics driver version
3. NVidia Geforce Experience version
4. Raspberry pi version
5. Raspbian version (date from here http://downloads.raspberrypi.org/raspbian/release_notes.txt)
6. Raspbian config.txt (``echo `cat /boot/config.txt | grep -v "#"` ``)
7. Raspberry pi kernel version (`uname -a`)
8. Raspberry pi firmware version (`vgencmd version`)
9. Network setup for PC+RPi (e.g wifi+wifi, ethernet+wifi, wifi+ethernet, ethernet+ethernet)
10. USB devices connected to RPi.
11. RPi PSU output rating.
12. Limelight-embedded experience (e.g perfect, good with issues, bad)

My setup is as follows:

1. Windows 7 Pro 64bit (running as KVM VM with GFX PCI-E passthrough, but that's another story).
2. 347.52
3. 2.2.2.0
4. 2 model B
5. 2015-01-31
6. disable_overscan=1 framebuffer_width=1920 framebuffer_height=1080 framebuffer_depth=16 hdmi_drive=2 hdmi_ignore_edid=0xa5000080 hdmi_force_hotplug=1 config_hdmi_boost=4 disable_overscan=1 start_x=1 gpu_mem=256   (my framebuffer settings were added for testing dfreerdp. They make no difference to limelight)
7. Linux raspberrypi 3.18.7-v7+ #755 SMP PREEMPT Thu Feb 12 17:20:48 GMT 2015 armv7l GNU/Linux
8. Feb 12 2015 17:27:39 
Copyright (c) 2012 Broadcom
version 571f5d2143489e4300d7e2ed5b1a73f33f3b13e5 (clean) (release)
9. ethernet+wifi (PC=ethernet, RPi=wifi)
10. Xbox 360 wireless receiver, keyboard, mouse, USB hub+wifi dongle
11. 5V @ 1.2A
12. 1080p@60fps=good but 1-2s video lag. 720@60fps=near perfect

Perhaps `vcgencmd get_config int` is a better option for 6?
Mine as follows:
arm_freq=900
sdram_freq=450
over_voltage_avs=0x1b774
hdmi_drive=2
hdmi_force_hotplug=1
disable_overscan=1
framebuffer_width=1920
framebuffer_height=1080
framebuffer_depth=16
disable_l2cache=1
program_serial_random=1
config_hdmi_boost=4
disable_commandline_tags=2
emmc_pll_core=1
hdmi_ignore_edid=0xa5000080
hdmi_force_cec_address=65535
ignore_lcd=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
disable_splash=1
arm_control=0xa5800010
temp_limit=85
force_pwm_open=1
pause_burst_frames=1
second_boot=1
avoid_fix_ts=1

irtimmer, 2015-02-16:
Github does have a wiki. It only wasn't enabled for Limelight Embedded. So everybody should now be able to edit the wiki on https://github.com/irtimmer/limelight-embedded/wiki

7hunderbug, 2015-02-16:
So it does! Great, thanks Iwan. I'll get on with making a page then.

modusdigital, 2015-02-18:
Ah, I guess it's reassuring to know I'm not alone. Is this a Pi 2 issue do you think?

cgutman, 2015-02-19:
People that can build from source and have this issue, can you try modifying something for me?

limelight-embedded/libs/common/src/com/limelight/nvstream/av/audio/AudioStream.java

Line 20 should read: 	public static final int RTP_RECV_BUFFER = 12 * 100;

Try changing it to: 	public static final int RTP_RECV_BUFFER = 64 * 1024;

Then rebuild the JAR and see if it changes anything.

Iperpido, 2015-02-24:
@cgutman i tried first compiling the last source cloned from git and then your "fix"... it seems there is no difference at all.
(i was using a lobopus.so compiled from source with fixed point)

And actually... steam works without any problem (even at 1080p 60fps), but the stuttering comes out when i start a game. (in my case, Metal Gear Rising).
And crashes after a while.

Maybe it's game related... but i have no problem running it on my Nexus5.

I'll try another game as soon i have some time.

Iperpido, 2015-02-25:
Ok, i made some more tests.
The problem shouldn't be game related since i tested it with both Metal Gear Rising and GRID.

It seems the main problem was the input device attached, since running limelight without it gives me no crashes at all and it's way faster.

And... seems like that change in AudioStream.java is actually helping.

For this test i just used the default settings. (just a "java -jar limelight.jar stream")

SirDigbyCC, 2015-03-03:
Ok I think I've solved my sound stuttering issue, disabling all audio devices in the Windows control panel seems to have stopped it.

I also tried removing my wireless keyboard/mouse from the Pi, but that didn't seem to help, neither did attaching my wireless 360 controller to the Pi rather than the host.

Perhaps there is something it doesn't like about my sound setup, which is onboard audio Realtek ALC898, connected via optical using DTS Connect to an external receiver. I'll run some more tests in the week but for those of you having problems, maybe try the same and see what happens.

On; another note, Iperpido, would you be able to upload a copy of your Limelight.jar with the modified AudioStream.java so the rest of us can see if it helps at all. :)

Iperpido, 2015-03-04:
Sure. Here: http://www.filedropper.com/limelight (LoL, didn't expected a link like that... i'm the first to upload Limelight there? xD)

xtrem007, 2015-03-07:
Works like a charm now! Thanks Iperpido:)

Iperpido, 2015-03-07:
No need to thank me, the fix is made by cgutman, i've just took the code from git, applied the change and recompiled.

Well, if the fix is really helping... maybe we should consider to committing it to the master branch?

smiley, 2015-03-07:
I had the same issue, and the fix helped somewhat -- but audio still skips & stutters every now and then. Now it's usually when the video changes a lot in a brief period of time.

I thought this might have something to do with audio decompression and the opus resource issue, but even when this happens my Pi reaches 72-73% CPU usage at most.

matty-haughton, 2015-03-29:
Could you please re host the that version of limelight as is is no-longer up and i have the same problem.
thanks

Iperpido, 2015-03-29:
Hope this stime it will stay online longer...

http://www.multiupfile.com/f/a4e5c999

closed 2015-07-16 by irtimmer