moonlight-stream/moonlight-embedded #84: 360 controller triggers ruin controller input

rossrankin, 2015-02-18:
Got a Pi 2 with a clean install of LE1.2.1, GFE is updated to the latest also.

Streaming works fine through the local network, and it recognises a 360 controller plugged in without installing xpad or xdrv stuff. All the buttons and directions work until I tap one of the triggers, then they seem to get stuck in and the buttons swap around. It stays like this until I end stream and restart. It also happens after installing xdrv.

I'm not sure how to configure the -map part of -java -jar -limelight -stream xxx.xxx.xxx.xxx etc.

HellKeeper62, 2015-03-02:
It's due to the trigger writes multiple times the button pressing. Have you try to follow the wiki : https://github.com/irtimmer/limelight-embedded/wiki/Controllers-set-up ? In my side, I tested yesterday and it's ok excepted for the X and Y button that are swapped. Is your case ? May be the mapping.conf file is not OK (but we can remap button in our side).

irtimmer, 2015-05-10:
Duplicate of #75

closed 2015-05-10 by irtimmer