moonlight-stream/moonlight-embedded #88: Error code during decode: -12

rkorn86, 2015-02-25:
Ive build current master 8ecd85ffa3b7e2366d21cf8819b41cdb1512f9d7 on my own on my Wandboard Quad (i.MX6 Freescale).
After successfull pairing ive tried to start via `# java -jar limelight.jar stream -app "FIFA 15"`. Unfortunately `Limelight` crashes immediately with the following response:
```
Discovering GeForce PCs...
 LAPPY 192.168.0.15
Starting FIFA 15
Starting RTSP handshake
Starting control connection
Starting video stream
[INFO]  bitstreamMode 1, chromaInterleave 0, mapType 0, tiled2LinearEnable 0
Starting audio stream
Starting input connection
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -12
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:23 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:24 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:24 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:24 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:24 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:24 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:28:24 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
[... 528 lines cut ...]
        at com.limelight.nvstream.control.ControlStream$1.run(Unknown Source)
java.net.SocketException: Socket closed
        at java.net.PlainDatagramSocketImpl.receive0(Native Method)
        at java.net.AbstractPlainDatagramSocketImpl.receive(AbstractPlainDatagramSocketImpl.java:145)
        at java.net.DatagramSocket.receive(DatagramSocket.java:812)
        at com.limelight.nvstream.av.video.VideoStream$1.run(Unknown Source)
java.net.SocketException: Socket closed
        at java.net.PlainDatagramSocketImpl.receive0(Native Method)
        at java.net.AbstractPlainDatagramSocketImpl.receive(AbstractPlainDatagramSocketImpl.java:145)
        at java.net.DatagramSocket.receive(DatagramSocket.java:812)
        at com.limelight.nvstream.av.audio.AudioStream$2.run(Unknown Source)
Connection Terminated The connection failed unexpectedly
```

```
 # java -version
java version "1.8.0_33"
Java(TM) SE Runtime Environment (build 1.8.0_33-b05)
Java HotSpot(TM) Client VM (build 25.33-b05, mixed mode)
```

Host is running a Windows 8.1 Pro (up to date by current date) with a Geforce 780M, Intel i7-4702, 16GB RAM, Driver Version 347.52

Im a J2EE expert, but dont have a clue on your JNI calls, and didnt find where your result codes are generated.
Is there anything else you need for finding the bug?

rkorn86, 2015-02-25:
Update: Playing around with the Heap Size Params, the error code changed:
```
# java -Xmn48m -Xmx96m -jar -jar limelight.jar stream -app "FIFA 15"
Discovering GeForce PCs...
 LAPPY 192.168.0.15
Starting FIFA 15
Starting RTSP handshake
Starting control connection
Starting video stream
[INFO]  bitstreamMode 1, chromaInterleave 0, mapType 0, tiled2LinearEnable 0
Starting audio stream
Starting input connection
Feb 25, 2015 8:38:14 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -7
Feb 25, 2015 8:38:14 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
Feb 25, 2015 8:38:14 PM com.limelight.LimeLog severe
SCHWERWIEGEND: Error code during decode: -1
...
```

irtimmer, 2015-05-10:
Propably same as #60

closed 2015-05-10 by irtimmer