moonlight-stream/moonlight-embedded #89: Can I disable default Xinput from client to computer?

georgcmef, 2015-02-25:
Hello I use limelight on multiple platforms (Android, IOS and Pi). I connect my controller directly to my PC to play my games via limelight. Some games run into a problem where the emulated Xinput controller from limelight takes priority over every other controller that is connected to my computer, making them not function. 

I fixed one game from having this error by stopping \NVIDIA Corporation\NvStreamSrv\rxinput.dll from executing. This has not fixed the problem for any other of my games. Please help, Thank you

amigagamer, 2015-05-22:
Same problem here (raspberry pi 2).  Most games allow you to choose a controller, however some games default to the first controler which is always the nvidia stream/limelight controller.    
I followed the recommendations on xda developers to rename rxinput.dll to rxinput.dll.old  however this causes a lot of lag/sound problems when running limelight.  (cpu always maxed out on pi)

spacewave, 2015-05-26:
from this page
https://github.com/irtimmer/moonlight-embedded/wiki/Controllers-set-up

Remove xpad kernel module: sudo modprobe -r xpad
Blacklist xpad kernel module: sudo nano /etc/modprobe.d/gamepad.conf. Add the line blacklist xpad, save and exit.

That didn't work for skyrim.  So i'm having the same problem as well.

The 360 wireless receiver connected to the pi didn't work either as some keys don't work right.  Example can't hold down button to open doors in far cry 3, similar issues in gta IV.  Tried using autofire option in xboxdrv with no luck.

Also ran into some streaming glitch issues with my switch.  Once i removed the switch from the equation, then streaming was basically perfect.  Very impressed until i ran into the controller issues.
Was considering trying a playstation controller and other wired controllers.  So so so close to being perfect.  :-(

closed 2016-01-16 by irtimmer