moonlight-stream/moonlight-embedded #90: "Pair illegal character in query at index 40"

qlanth, 2015-02-26:
I'm very new at this so I may have missed sonethi my, somewhere. But, whenever I try to pair with my PC I am getting the following:
"Please enter the following PIN on the target PC: 8402
Pair illegal character in query at index 40: https://193.168.0.2:47904/pair?uniqueid=&devicename=Roth&updatestate=1&phrase=getservercert&salt=...."

It goes on from there for about another 400 characters. From the PC end GFE never asks for the pin. 

I tried pairing to a different IP and got the same result. Tried rebooting all systems involved. As I mentioned I'm very new at this. Not sure where to go from here.

cgutman, 2015-02-26:
What is the contents of your uniqueid.dat file? It should be in the same directory as your jar. It looks like it's empty or corrupt.

qlanth, 2015-02-26:
The file is empty. What should I put into it?

qlanth, 2015-02-26:
figured this out. I renamed the uniqueid.dat to uniqueidold.dat. After running "java -jar limelight.jar pair 192.168.0.2" the uniqueid.dat file was recreated and everything went smoothly! Finally streaming! Thanks for your help!

closed 2015-02-26 by qlanth