moonlight-stream/moonlight-embedded #91: Limelight stopped working

scoobyd00, 2015-02-28:
Ive been running limelight for a while now without any issues.
Then i havent been on it for a few weeks, come to it today and it wont respond.
Its sending video fine but no audio and no controller input.

If i uninstall geforce experience and reinstall it then limelight streams the first time fine with both audio and controller input working.
As soon as i reboot my PC and try streaming again i loose audio and controller.

Ive tried a complete fresh install of windows on my PC and again the same problem. Also done a new raspbian image etc and no change.

Anyone know what could be wrong?, its driving me mad.

scoobyd00, 2015-02-28:
As a last resort ive just gone back to geforce experience 2.1.1 and an older build of limelight and it works fine now!.
Not sure if theres some issue with 2.2.2 and lastest limelight build.
Im running on a PI 2

cgutman, 2015-03-03:
Are you running Limelight 1.2.1 or a build of the current master?

scoobyd00, 2015-03-03:
I was running 1.2.1

scoobyd00, 2015-03-04:
Getting issues with 2.1.1 now, after Ive stopped playing a game with the -app argument limelight doesnt terminate on the pi or PC and i have to stop the nvidastream service, issue is that when i do this it corrupts geforce expereince and i cant play the game again without reinstalling geforce experience.?
If i play via steam then it exists ok, the only problem is that i cant get GRID autosport to play correctly when through steam.

closed 2015-04-04 by scoobyd00