moonlight-stream/moonlight-embedded #93: Not paired

matigev, 2015-03-07:
First of all: Thanks for this great piece of software!
I'm currently running on Raspberry Pi 2 (Raspbian), but I'm experiencing some weird behaviour when running from /etc/rc.local. I get the error "not paired", but when I ssh to my pi no problems at all with the same command. Anyone knows why?
Previously I ran on Pi B+ (older Raspbian) with no errors at all.

gforced, 2015-03-09:
having same issue

matigev, 2015-03-29:
I figured it out finally. First cd to the jar directory! For example in my rc.local:
cd /home/pi
sudo java -jar limelight.jar stream &

matigev, 2015-10-07:
With the new c versions I am getting this error again.. cd'ing to the right dir or setting the PATH won't work. When I put moonlight pair in front of it, it says "Already paired". Is there any way to point moonlight to the right certificate files?

matigev, 2015-10-07:
Never mind, I figured it out myself... you can use 
-keydir /path/to/dir

closed 2015-10-07 by matigev