moonlight-stream/moonlight-embedded #94: Sound still stutters whilst controller plugged in

patmanblu3, 2015-03-09:
Sorry to re-post this but I've been trying to figure this out for the past few days and have not made any progress so I hope I can get some assistance here. Basically this   

"Streaming is fine sound only stutters when a button is pressed on the controller 
Currently using Xbox 360 controller via xboxdrv 
Unplugging the controller from the pi resolves the issue. 
I'm on the latest gfe and limelight release"

everything is updated using limelight embedded 1.2.1. 

I had It fixed for a while but it recently came back. Could it just be my Opus build causing high cpu usage? Any well is greatly appreciated.

HazCod, 2015-03-11:
Try SSHing from a separate machine into your pi and check memory/cpu usage

MikaelVZ, 2015-03-12:
I think I got the same error on my PI B+.
I user -v parameter to debug.
When pushing the left or right stick of my Xbox360 wireless pad sound stutter and I got this error messages :
```
Mar 12, 2015 9:18:25 PM com.limelight.LimeLog warning
WARNING: Alsa error from writei: -32
Mar 12, 2015 9:18:25 PM com.limelight.LimeLog warning
WARNING: Received OOS audio data (expected 29692, got 29718)
Mar 12, 2015 9:18:25 PM com.limelight.LimeLog warning
WARNING: Received OOS audio data (expected 29719, got 29730)
Mar 12, 2015 9:18:25 PM com.limelight.LimeLog warning
WARNING: Received OOS audio data (expected 29740, got 29762)
Mar 12, 2015 9:18:25 PM com.limelight.LimeLog warning
WARNING: Alsa error from writei: -32
Mar 12, 2015 9:18:25 PM com.limelight.LimeLog warning
WARNING: Received OOS audio data (expected 29764, got 29776)
Mar 12, 2015 9:18:25 PM com.limelight.LimeLog warning
WARNING: Received OOS audio data (expected 29792, got 29798)
```

cgutman, 2015-03-13:
This is the issue my change in #83 was designed to fix. The device can't keep up with audio decoding while processing the stream of input from the analog sticks. Increasing the audio stream receive window gives the audio thread time to catch up.

HazCod, 2015-03-13:
@cgutman, Looks like a pretty fix. Is this going to master?

MikaelVZ, 2015-03-16:
I tried the fix, less stuttering but still present, connected my controller to the PC finally solved my problem.
Thanks

patmanblu3, 2015-03-30:
Can confirm sound still stutters with the controller plugged in on version 1.2.2 but it is much much less. I'll try overclocking my pi slightly tomorrow to see if that helps. Great update tho! I can definitely notice the improvements.

mikaelvz, 2015-04-02:
After a quick test I can confirm that there is much less stuttering.
Thanks for this great work

closed 2015-07-16 by irtimmer