moonlight-stream/moonlight-embedded #95: Limelight Setup & GUI

zer0future, 2015-03-19:
Preliminary information This text was translated by Google Translator.

For a long time I want a simple possibility Limelight embedded to install.
In addition Limelight should have a text menu.

1 SETUP

The installation should do the following:

Update & Upgrade of Raspberry
Suggest Recommended overclocking and configuration settings. (CPU clock, clock Ram, Memory Split)
Automatic installation of the required software (Java, XBox 360 or PS3 controller driver, Libopus, Git, Whiptail, WAKEONLAN)
Download Limelight Embedded Controller mapping files
Pairing the PC
Storing the MAC address of the PC for later Wake on Lan in Limelight menu

Tasks for the setup I think I have a pretty good template found. However, I need help with the editing, because I really do not know much about.
Link to the script that should fit graphically and technically:
https://github.com/Mark-Leck/Configurator/blob/master/configurator

Someone would only have to change and adapt.

2 MENU

The menu should have the following options:

Activation and deactivation "Auto Start" for the menu
-Pairing PC
-Save The MAC address of the PC for Wake on Lan option before the start of a game or Steam
- Selection of Transmission Quality (-720 -1080 -30fps -60fps)
- Selection of startup options (Mapping for XBox 360, PS3 controller, Wake on Lan)
-WoL PC
- Query of the installed games on the PC (Discovery)
-Start Of Steam with the predetermined quality
-Start A game from the list in the Discovery predetermined quality and with the selected boot options eg WoL, XBox360 mapping

I think with a whiptail script that is possible, if you have the necessary experience.

#####
German
#####
Vorab eine Information: Dieser Text wurde vom GoogleTranslator übersetzt.

Schon lange wünsche ich mir eine einfache möglichkeit Limelight-Embedded zu Installieren.
Zusätzlich sollte Limelight ein Text Menü haben.

1 SETUP

Die Installation sollte folgendes ausführen:

Update & Upgrade des Raspberry
Empfohlene overclocking und konfigurations einstellungen vorschlagen. (CPU Takt, Ram Takt, Memorysplit)
Automatische Installation der benötigten Software (Java, XBox 360 oder PS3 Controller Treiber, Libopus, Git, Whiptail, WakeonLan)
Download von Limelight-Embedded, Controller Mapping Dateien
Pairing des PC
Speichern der MAC Adresse des PC für späteres Wake on Lan in Limelight Menü

Für die Setup aufgaben glaube ich eine recht gute Vorlage gefunden zu haben. Allerdings brauche ich hilfe bei der Bearbeitung, da ich mich eigentlich nicht auskenne.
Link zum Script das grafisch und technisch passen sollte:
https://github.com/Mark-Leck/Configurator/blob/master/configurator

Jemand müsste es nur abändern und anpassen.

2 MENÜ

Das Menü sollte folgende optionen haben:

-Aktivierung und deaktivierung  "Autostart" für das Menü
-Pairing des PC
-Speichern der MAC Adresse des PC für Wake on Lan option vor dem Start von Steam oder eines Spiels
- Auswahl der Übertragungsqualität (-720 -1080 -30fps -60fps)
- Auswahl der Startoptionen (Mapping for XBox 360, PS3 Controller, Wake on Lan)
-WoL des PC
- Abfrage der Installierten Spiele auf dem PC (Discovery)
-Starten von Steam mit der vorher festgelegten Qualität
-Start eines Spiels aus der Discovery Liste in der vorher festgelegten Qualität und mit den ausgewählten Startoptionen z.b. WoL, XBox360 Mapping

Ich denke mit einem whiptail script ist das möglich, wenn man die nötige erfahrung hat.

stsfin, 2015-03-27:
Hi,

I made a setup script for rasperry pi with a configurator, it generates .sh files for easy run for different qualities and helps with finding the right device from events etc:

https://github.com/stsfin/LimelightPiInstaller

zer0future, 2015-03-29:
Thats pretty cool & thank your for that.
I Think i can use your script to learn a little bit more in debian programming and i think i will upgrade your script with a startup script
maby:
-GUI with a list of Installed games, registerd in Geforce Experience "GameStream" configuration on the Host Computer
(sometimes, some games doesn't startup in steam, 
specialy non steam games (added by klicking "Add a Game" -> "Add a non Steam Game" - in the Steam Client on the Host PC.

With the Limelight startup command: 
java -jar limelight.jar stream -app NAMEOFTHEGAME
all games running without a problem. 

By simply changing the names of the Games in the GfE "GameStream" configuration on the Host Computer, we can grab Game informations from the Host Computer to show a perfekt Checkbox List in the GUI of the Raspberry, with all the Installed Games and no need to start Steam.

The GUI shoud have:
Checkboxes for options to start a game or steam like 
 - 60 fps
- 1080p
- input options 
- ...simply all options from the limeliight-embedded script

The List of the Installed Games grabed from gefroce Experience "GameStream" game list,

stsfin, 2015-03-29:
Yeah, I'm thinking of adding a autostart and oc-setting to the script...
the app selection with text input would be easy to do

stsfin, 2015-03-29:
@zer0future Here is the "primary" version of the script made for RetroPie and Emulation Station: it automatically creates a menu item in Emulation station for the different resolution scripts and the reconfigurator: https://github.com/stsfin/RetropieLimelightInstaller/

Here's how it looks when its running: https://www.youtube.com/watch?v=CT0Zhmh9e9o

wkewlin, 2015-04-17:
Here is my take on this
https://github.com/wkewlin/limelight-openelec

The new thing here is a a deamon script listening for a exit key on cec to kill limelight and go back to kodi.  
One can resume the same session with the same game running by resume script as killing limelight does not end the session, the game still runs on pc.  
This also solves the problem when you are thrown to desktop but game is still running. Just kill limelight and resume session.  
I use steam for game selection inside a session so I live without a list of games.  
Some should probably pack this in a openelec/kodi addon so that you can install it without going to ssh and a better integration instead of the commands addon would also be nice but as I am good with how it works now I probably wont do it.  
Main reason for openelec is that it supports cec as far as I know Emulationstation from retropie does not support it yet. As I do not use other emulators my solution works better for me and maybe someone finds it also usefull

irtimmer, 2015-05-10:
Duplicate of #71

closed 2015-05-10 by irtimmer