moonlight-stream/moonlight-embedded #96: Limelight-embedded Raspian install script

stsfin, 2015-03-27:
Hi,

I made a configurator and install script for rasbian based systems:

https://github.com/stsfin/LimelightPiInstaller

This should help with issues (finding the right event etc) for first timers to get limelight working... This could be forked to work with other systems as well

stsfin, 2015-03-27:
@irtimmer hope you can fork it further.. My knowledge of other (read rpm based) systems is limited.

RoelofBerg, 2015-04-04:
Cool, thanks for figuring out how to 1080p@60Hz :) I might add this to my setup tutorial at http://embedded-software-architecture.com/?page_id=140

stsfin, 2015-04-05:
Naturally yes!

barryownt, 2015-06-21:
Thanks bro i think this would be helpful for a lot of people !

closed 2015-07-16 by irtimmer