moonlight-stream/moonlight-embedded #98: Supply libopus.so.0

HazCod, 2015-03-31:
So I'm getting the following error on OpenELEC, would it be possible to also supply libopus.so.0?
At the moment, I have a symlink to libopus.so but this does not seem to work.

`
HTPC-Niels:~ # systemctl stop kodi && /storage/jdk/bin/java -jar /storage/limeli
ght/limelight.jar stream 192.168.0.150
Starting Steam
Starting RTSP handshake
Starting control connection
Starting video stream
Starting audio stream
Exception in thread "Thread-2" java.lang.UnsatisfiedLinkError: /tmp/nv_opus_dec-689515303325140630.so: libopus.so.0: cannot open shared object file: No such file or directory
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1937)
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1835)
        at java.lang.Runtime.loadLibrary0(Runtime.java:870)
        at java.lang.System.loadLibrary(System.java:1119)
        at com.limelight.binding.audio.OpusDecoder.<clinit>(Unknown Source)
        at com.limelight.binding.audio.AlsaAudioDecoderRenderer.streamInitialize(Unknown Source)
        at com.limelight.nvstream.av.audio.AudioStream.setupAudio(Unknown Source)
        at com.limelight.nvstream.av.audio.AudioStream.startAudioStream(Unknown Source)
        at com.limelight.nvstream.NvConnection.startAudioStream(Unknown Source)
        at com.limelight.nvstream.NvConnection.establishConnection(Unknown Source)
        at com.limelight.nvstream.NvConnection.access$200(Unknown Source)
        at com.limelight.nvstream.NvConnection$1.run(Unknown Source)
        at java.lang.Thread.run(Thread.java:745)
java.net.SocketException: Broken pipe
        at java.net.SocketOutputStream.socketWrite0(Native Method)
        at java.net.SocketOutputStream.socketWrite(SocketOutputStream.java:109)
        at java.net.SocketOutputStream.write(SocketOutputStream.java:153)
        at com.limelight.nvstream.control.ControlStream$NvCtlPacket.write(Unknown Source)
        at com.limelight.nvstream.control.ControlStream.sendPacket(Unknown Source)
        at com.limelight.nvstream.control.ControlStream.sendLossStats(Unknown Source)
        at com.limelight.nvstream.control.ControlStream.access$100(Unknown Source)
        at com.limelight.nvstream.control.ControlStream$1.run(Unknown Source)
java.net.SocketException: Socket closed
        at java.net.PlainDatagramSocketImpl.receive0(Native Method)
        at java.net.AbstractPlainDatagramSocketImpl.receive(AbstractPlainDatagramSocketImpl.java:145)
        at java.net.DatagramSocket.receive(DatagramSocket.java:812)
        at com.limelight.nvstream.av.video.VideoStream$1.run(Unknown Source)
Connection Terminated The connection failed unexpectedly
`

stsfin, 2015-03-31:
Have you tried building it from source?

stsfin, 2015-03-31:
Since as far as I know the prebuild lipopus is for Debian systems, and since OpenElec is a Linux For Scratch build its better to build opus from source directly on the Elec system

stsfin, 2015-03-31:
Have you tried to copy the opus-codec to the same folder as limelight and renaming it?

HazCod, 2015-03-31:
Sadly, I don't have access to a build environment at the moment.

And yes, I copied the libopus.so to the same folder and setup a symlink from libopus.so.0 to libopus.so

stsfin, 2015-03-31:
Ok, tell you what, since you helped me with the install script I could in return wipe my Emulation Station from the SD-card and install openelec, and try to get this running. It would anyhow fun to add openelec install to the script as a user choice... The more platforms it supports the better ^^

HazCod, 2015-03-31:
Thanks mate, would be wonderful!

stsfin, 2015-03-31:
Im anyhow thinking of merging the Emulation station version with the "basic" version, so why not also add openelec to the mix... Would anyhow be easier to keep the versions updated.

HazCod, 2015-03-31:
FYI; default SSH credentials for openelec machines are root & password 'openelec'.
Files must be placed in /storage

stsfin, 2015-03-31:
Just a question, is there a special reason why you want to run Openelec? (Retropie has also integrated Kodi, a few builds ago and you can install it from the experimental packages menu)

HazCod, 2015-03-31:
Mostly because it's VERY lightweight and sports an automatic updater + backup.

stsfin, 2015-03-31:
Did the same thing: It's the same file... they are just symlinks that the build enviroment creates

lrwxrwxrwx  1 pi pi      16 Mar 31 23:32 libopus.so -> libopus.so.0.5.0
lrwxrwxrwx  1 pi pi      16 Mar 31 23:32 libopus.so.0 -> libopus.so.0.5.0
-rwxr-xr-x  1 pi pi 1267928 Mar 31 23:32 libopus.so.0.5.0

stsfin, 2015-03-31:
limelight does work on openelec: http://openelec.tv/forum/90-miscellaneous/68987-limelight-nvidia-shield-game-streaming since people have been succesfully getting it to run on it... No one just shares how :D

HazCod, 2015-03-31:
Well it worked, but without sound. And when I switched from Steam to my putty window, I saw the stacktrace mentioned above.

HazCod, 2015-04-01:
Update: Tonight I will try with LD_LIBRARY_PATH modified, /storage/limelight should be prepended and not appended;

```
export LD_LIBRARY_PATH=/storage/limelight:$LD_LIBRARY_PATH
```

kaasszje, 2015-04-01:
I'm using this on openelec as well.
I'll try latest version tonight and let you know my findings.
Previous version was working fine.

Only thing I had to do, was a modprobe for sound:
modprobe snd_bcm2835

HazCod, 2015-04-01:
Thank you for responding @kaasszje , I will include instructions on the README when I get this working. I didn't know about the modprobe.

kaasszje, 2015-04-01:
I'm starting limelight from the limelight folder where limelight.jar and
libopus.so are located.
As posted earlier I can't check right now, but I'll check it tonight.
This is my script:
#!/bin/sh
systemctl stop kodi
modprobe snd_bcm2835
cd /storage/limelight
/storage/tools/jre/bin/java -jar limelight.jar stream 192.168.178.10
systemctl start kodi

kaasszje, 2015-04-01:
Sounds works super for me in kodi as well.
However when using the sound out of kodi, it first needs to be known by the system.
Therefor you use the modprobe else the system doesn't load any drivers.
(I only use this over hdmi as well, no idea about the 3.5mm jack)

HazCod, 2015-04-01:
Ahh, good news! Thanks for the heads up, will try it this evening.

HazCod, 2015-04-01:
Works with the modprobe, thank you @kaasszje !

closed 2015-04-01 by HazCod

HazCod, 2015-04-01:
Instructions made available [at my repo.](https://github.com/HazCod/LimeRPi2-kodi/blob/master/README.md)