moonlight-stream/moonlight-embedded #100: Keyboard issues

wkewlin, 2015-04-17:
I managed to get limelight running in openelec 5.
Here is a list of scripts I am using and a short guide how to set them up.
https://github.com/wkewlin/limelight-openelec

Here are my issues

1. First issue is that keyboard and mouse are only working if limelight is attached to a console. I noticed that when I saw that they worked if i started my scripts from a ssh session but did not when i run the same python script from kodi. Adding `screen` to fake a console solved the issue. I do not have any experience with linux input event handling, but it seems like a permission issue. Why it is working with a ssh console or a detached sceen console I dont know.

2. A bigger issue is that certain keys are not passed correctly. Most important for me `ctrl` and `shift` are not working not working a modificators. Also keys are not repeated by holding them. I am not sure if it is a issue with OpenELEC or with limelight. In Kodi the keys are working and also they do repeat. They also work if a connect by using limelight-pc on a linux system.

3. Keyboard mapping is something like `qwerty` even so my pi and my windows machine are on `qwertz`. I am saying something like because z and y are reversed but ö gives ü,  ä and ü are not working at all as are some other keys. On limelight-pc neither ö or ü or ä are working but z and y are in right position.

For 1. I found a hack and I can live with 3. but getting 2. solved so that `ctrl` and `shift` would be nice.

kingrastamon, 2015-04-21:
Issue 2 affects me as well.  I am unable to send any modifier keys to my Windows PC.  The code that should detect modifiers was updated in Feb (2b35932) but it seems to have no effect.

My setup:
GFE 2.2.2
Raspberry Pi 2
Raspbian
Custom Compiled Kernel: 3.18.9-v7+ (so I could get my Xbox DVD kit working)
Java 1.8.0
Logitech K400 Wireless Keyboard/Trackpad
Limelight-Embedded 1.2.2 running from shell (I hope to start it from Kodi eventually).

closed 2015-07-16 by irtimmer

maxellnormalbias, 2017-04-13:
Why was this issue closed? I'm still having #2, even after upgrading GFE and moonlight-embedded to the latest version.