moonlight-stream/moonlight-embedded #104: Cant build new master - cmake error

scoobyd00, 2015-05-11:
Im trying to build the new master which by the looks of it has been converted to c.
When im using cmake its giving me an error saying cmake is the incorrect version, it requires 3.2 but Raspbian only has 2.8.9 ??

7hunderbug, 2015-05-11:
I've not tried this myself yet so no guarantee it will work but did you try changing CMakeLists.txt so:

    cmake_minimum_required(VERSION 3.2)

becomes:

    cmake_minimum_required(VERSION 2.8.9)

?

scoobyd00, 2015-05-11:
Yes i tried changing to 2.8.9 but didnt work.
Just compiled 3.2 from source and it gets abit further now but still errors missing CURL_LIBRARY

Hopefully a new release will be posted soon to download.

7hunderbug, 2015-05-11:
Is libcurl4-openssl-dev installed?

scoobyd00, 2015-05-11:
I didnt, ive just installed it and tried again and it gets further again! . lol, but then more errors expat and all sorts.
Its beyond my unix skills (complete novice!).

Hopefully irtimmer will be able to help out wtih a complied version.

7hunderbug, 2015-05-11:
I'm having a go at compiling this now too. It looks like there are lots of dependencies still required to be installed to Raspbian... apart from cmake and libopus-dev, so far we've identified:

libcurl4-openssl-dev
libexpat1-dev

I'm now stuck on a dependency on libevdev, but Raspbian doesn't have a development package by that name.... the search continues

scoobyd00, 2015-05-11:
Ok cheers, good luck. Let me know how you get on.

7hunderbug, 2015-05-11:
OK, the story so far.....
I've got the cmake command to complete. Unfortunately, in Raspbian "Wheezy" (which is the stable version as is available from raspberrypi.org via the standalone and NOOBs installer) there is no libevdev-dev package. Thankfully, this is available from the Raspbian "Jessie" repos here:

http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb
http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-tools_1.4.2%2bdfsg-1_armhf.deb
http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb

So, the instructions so far:

1. Change CMakeLists.txt so that cmake_minimum_required(VERSION 2.8.9)
2. Install cmake and libopus-dev. Install libcurl-openssl-dev and libexpat1-dev. Install libjs-jquery which is a dependency of the libevdev packages we're about to install manually.

        sudo apt-get install cmake libopus-dev libcurl-openssl-dev libexpat1-dev libjs-jquery

3. Get the libevdev packages:

        wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb
        wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb

4. Manually install the evdev packages.

        sudo dpkg -i libevdev2_1.4.2+dfsg-1_armhf.deb libevdev-dev_1.4.2+dfsg-1_armhf.deb

5. Start the build as instructed. 

        mkdir build
        cd build/
        cmake ../
        make

Unfortunately, compiling fails. It was failing due to mapping.h being missing from the new C source, but irtimmer has just committed a fix for that. So after a git pull, trying to compile again results in this error:

        /home/pi/sources/moonlight-embedded/src/input.c: In function ‘input_handle_event’:
        /home/pi/sources/moonlight-embedded/src/input.c:167:9: error: ‘for’ loop initial declarations are only allowed in C99 mode
        /home/pi/sources/moonlight-embedded/src/input.c:167:9: note: use option -std=c99 or -std=gnu99 to compile your code
        /home/pi/sources/moonlight-embedded/src/input.c: In function ‘input_poll’:
        /home/pi/sources/moonlight-embedded/src/input.c:341:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
        CMakeFiles/moonlight.dir/build.make:77: recipe for target 'CMakeFiles/moonlight.dir/src/input.c.o' failed
        make[2]: *** [CMakeFiles/moonlight.dir/src/input.c.o] Error 1
        CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/moonlight.dir/all' failed
        make[1]: *** [CMakeFiles/moonlight.dir/all] Error 2
        Makefile:72: recipe for target 'all' failed
        make: *** [all] Error 2

scoobyd00, 2015-05-11:
Excellent work, hopefully irtimmer will be along to let us know what the problem is.
Looking forward to seeing how this new build performs

7hunderbug, 2015-05-11:
adding

        SET(CMAKE_C_FLAGS "-std=gnu99")

below

    SET(CMAKE_MODULE_PATH "${PROJECT_SOURCE_DIR}/cmake")

in CMakeLists.txt helps to progress compiling the code. (setting std=gnu99 is more forgiving than std=c99 I believe). However, now there is another error showing another missing header file in the source:

    [ 52%] Building C object CMakeFiles/moonlight.dir/common/limelight-common/VideoDepacketizer.c.o
    In file included from /home/pi/sources/moonlight-embedded/common/limelight-common/VideoDepacketizer.c:3:0:
    /home/pi/sources/moonlight-embedded/common/limelight-common/LinkedBlockingQueue.h:3:22: fatal error: platform.h: No such file or directory compilation terminated.
    CMakeFiles/moonlight.dir/build.make:330: recipe for target 'CMakeFiles/moonlight.dir/common/limelight-common/VideoDepacketizer.c.o' failed
    make[2]: *** [CMakeFiles/moonlight.dir/common/limelight-common/VideoDepacketizer.c.o] Error 1
    CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/moonlight.dir/all' failed
    make[1]: *** [CMakeFiles/moonlight.dir/all] Error 2
    Makefile:72: recipe for target 'all' failed
    make: *** [all] Error 2

I'm guessing the code is currently a work in progress for irtimmer.....

7hunderbug, 2015-05-11:
Ah, no, that error is purely a bug in the moonlight-common-c tree. LinkedBlockingQueue.h includes platform.h but the file is capitalised and is called Platform.h, so editing ../common/limelight-common/LinkedBlockingQueue.h to correct this issue allows compiling to progress even further (but not quite to completion yet! :)

7hunderbug, 2015-05-11:
I'm currently stuck at the same point as in issue #106

7hunderbug, 2015-05-11:
Thanks @irtimmer for another quick fix to #106. However, now when compiling your code on Raspbian, the linker throws this error:

    [100%] Building C object CMakeFiles/moonlight.dir/src/video/fake.c.o
    Linking C executable moonlight
    /usr/bin/ld: cannot find -lPUBLIC
    collect2: ld returned 1 exit status
    CMakeFiles/moonlight.dir/build.make:691: recipe for target 'moonlight' failed
    make[2]: *** [moonlight] Error 1
    CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/moonlight.dir/all' failed
    make[1]: *** [CMakeFiles/moonlight.dir/all] Error 2
    Makefile:72: recipe for target 'all' failed
    make: *** [all] Error 2

The fix appears to be removing "PUBLIC" from the following line in CMakeLists.txt.

    target_link_libraries (moonlight PUBLIC ${CMAKE_THREAD_LIBS_INIT} ${CURL_LIBRARIES} ${OPENSSL_LIBRARIES} ${EXPAT_LIBRARIES} ${EVDEV_LIBRARIES} ${ALSA_LIBRARY} ${OPUS_LIBRARY})

And with that, the compiling completes.

scoobyd00, 2015-05-11:
Have you got it up and running ok now?

7hunderbug, 2015-05-11:
No, it doesn't work. Pairing works, streaming does not. moonlight is able to start Nvidia Stream and Steam on the Windows Box but then there's this error in the Raspberry Pi console:

    Initializing platform...connection_stage_starting 1
    connection_stage_complete 1
    done
    Starting RTSP handshake...connection_stage_starting 2
    RTSP OPTIONS request failed
    failed -1
    connection_stage_failed 2

More changes have been committed since earlier today and with the new changes it doesn't compile again. It seems to be a work in progress so I think we'll have to wait patiently for irtimmer to finish his new C project. Keep up the good work @irtimmer!

scoobyd00, 2015-05-12:
Thanks for the update 7hunderbug.

I await patiently for the new 'c' compile :) been having alot of issues with the java version lately with it not streaming video for some reason. Keep on getting 'broken pipe' error message in the terminal.
So hopefully the new 'c' compile will sort this out.

Any ETA @irtimmer ?

irtimmer, 2015-05-12:
Almost all the funcionality of the java version has been implemented and the C version is currently working on the Raspberry Pi using ArchLinux (with GFE 2.4.3.22). But 'broken pipe' and 'RTSP OPTIONS request failed' sounds more like connection problems, firewall issues or GFE problems and therefore propably won't be fixed by a new version.

scoobyd00, 2015-05-12:
Thanks for the update, excellent work being done there.
I'll await the new release and try and sort out any issues.  I'm currently running a raspbian setup so I'll wait for you to post a compiled version on the releases page.
Keep up the good work, its great being able to stream to my 47inch TV :)

7hunderbug, 2015-05-12:
I don't think "RTSP OPTIONS request failed" in my case is related to connection problems, firewalls of GFE....
Today I reinstalled GFE, Nvidia drivers and Steam and cleared all device pairings.
I'm using Raspbian. With the old java based limelight.jar project I am able to pair, list apps and stream perfectly.
With the new C based moonlight project (admittedly compiled on Raspbian as detailed above) I am able to pair, list apps, but I'm not able to stream.
I'll try compiling on Arch over the coming days to see how that goes.

scoobyd00, 2015-05-12:
@7hunderbug could you tell me how you clear the device pairings? I've paired numerous times when testing different builds and this could be why I get the broken pipe problem sometimes

7hunderbug, 2015-05-12:
I recompiled moonlight on Raspbain using the most recent code git pulled today as well. There are two more dependencies for compiling:  libudev-dev and libavahi-client-dev.

7hunderbug, 2015-05-12:
@scoobyd00 - delete any certificate files located in the same directory as limelight.jar, e.g. client.key, client.crt, client.pem on the Raspberry Pi and on your Windows box, in the GFE settings under Preferences->GameStream, click the button at the bottom that says "Forget Devices"

scoobyd00, 2015-05-12:
Thanks 7hunderbug, that's a great help, I'll get it cleaned up.

7hunderbug, 2015-05-12:
Great Odin's Raven! It works! Sooo.. rather than specifying the Windows box by hostname as I was doing, specifying by IP address works. However it is now complaining about the ALSA sound card on the Raspberry Pi,.... diagnosing that one now.

cgutman, 2015-05-12:
Sounds like he's not resolving the hostname before passing it to moonlight-common-c.

7hunderbug, 2015-05-16:
I've forked irtimmer's repo, made the necessary changes to enable compiling on Raspbian in a new moonlight-raspbian branch and have published the resultant binary for anyone who needs it.

https://github.com/7hunderbug/moonlight-embedded/tree/moonlight-raspbian
https://github.com/7hunderbug/moonlight-embedded/releases

scoobyd00, 2015-05-16:
Thanks 7hunderbug.
Is the issue with alsa resolved?
Do you find any performance gains?

7hunderbug, 2015-05-16:
The alsa issue was nothing really, my mistake. Performance seems excellent. Streaming starts much quicker and everything feels snappier and more responsive. I'm trying to get my Xbox 360 wireless controller working properly now...

scoobyd00, 2015-05-16:
That's great news.
I got my Xbox wireless controller setup yesterday after having issues with the dpad not working correctly. I posted what worked for me on the raspberry pi forums under the limelight subject.

scoobyd00, 2015-05-16:
Downloaded the binary but cant get it to run.
just says 'command not found'

7hunderbug, 2015-05-16:
Maybe try this to make thet binary exectuable:
```chmod +x ./moonlight```

Try this to run it (from within the folder you have the file):
``` ./moonlight ```

7hunderbug, 2015-05-17:
I've updated the release and gzipped it. It might be possible the original upload was corrupted.

scoobyd00, 2015-05-17:
@7hunderbug  thanks for the new update, I'll try it out tomorrow, have you got the Xbox wireless controller working correctly now?

scoobyd00, 2015-05-29:
The build now compiles but it needs some fixes for Raspbian ....

in CMakeLists.txt change.

-cmake_minimum_required(VERSION 3.1)
+cmake_minimum_required(VERSION 2.8.9)

+SET(CMAKE_C_FLAGS "-std=gnu99")

and on target_link_libraries you need to remove PUBLIC

these changes were made by @7hunderbug on his fork.

When these changes are made I can confirm it completes the compile successfully
on Raspbian

RikJm, 2015-05-30:
Hi guys! I've been following all your progress and steps. Thank you!

I managed to compile successfully on my Rpi2. The quality of audio streaming is great, something I wasn't able to achieve with released on irtimer's github. 

However I'm facing an xbox controller issue, I can only use my left stick to go right or left (no up down) and I can't use any other buttons... I've tried different map files but nothing seems to fix it. Are you experiencing these issues too?

scoobyd00, 2015-05-30:
@RikJm I havent had any issues with my Xbox wireless controller. Have you mapped it in moonlight to get your own mapping file created?. Also trying using method 3 in this link, 

https://github.com/petrockblog/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller

I found if i used method 1 then I got all sorts of issues with the controller, using method 3, the init script makes my controller work perfectly.

RikJm, 2015-05-30:
@scoobyd00 I did map it in moonlight and I even copied that file in /usr/local/share/mappings/default.conf as well as passing the -mapping mapfile.map 

What's strange is that the controller works great on any other app, I have a retropie installation and I haven't had any problems with it, even the released java version of limelight has no controller problems. 

The thing is, I decided to compile moonlight myself, to get rid of some audio issues on the one irtimer's release. Now, those audio issues are gone, however the one I compiled can't handle the controller where the one from github can. I also tried the build 7hunderbug posted on his github, same controller issue.

Thanks for the suggestion, I'm going to try and use method 3 on the controller setup guide, I'll let you know if that solves anything.

unchartedxx, 2015-06-11:
You don't need to make changes to compile on Raspbian. You need to download the source of cmake 3.2 (latest )and then compile install it. 

sudo apt-get install build-essential
wget http://www.cmake.org/files/v3.2/cmake-3.2.3.tar.gz
tar xzf cmake-3.2.3.tar.gz
cd cmake-3.2.3
./configure
sudo make install

Then you apt-get all the packet required to compile moonlight (libcurl4-openssl-dev,...), however one doens't exist so you will need to download,compile,install it separately. You find it here: http://www.freedesktop.org/software/libevdev/libevdev-1.4.2.tar.xz

When you compile moonlight I suggest to force optimization:
mkdir build
cd build/
cmake -D CMAKE_BUILD_TYPE=Release ../
make
make install

I don't know if it's only my case but the executable when launched doesn't find a library I don't remember the name. You can copy in the same directory as the executable. It is found in usr/local/lib or something like that.

This version made in C is much better than the Java one, it starts much faster and with overclock I can stream 1080p 60fps with a raspberry 1.

RikJm, 2015-06-13:
@unchartedxx 
I compiled following your steps ang my xbox controller is still not working, what type of controller are you using? If it's xbox, is it wired or wireless?
Thanks :)

7hunderbug, 2015-06-13:
@RikJm please post your moonlight mapping file here so we can troubleshoot. Could you also post the following:
1. the command line you use to create your own moonlight mapping file
2. the command line you use to run moonlight so that it uses your mapping file.
Thanks.

RikJm, 2015-06-13:
Fixed!! I just posted the update here: https://github.com/petrockblog/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller

amigagamer, 2015-06-17:
@7hunderbug thanks for your work/guides on this. Ive been using your fork without problems on raspbian.  Im curious about your moonlight-raspbian-opus branch. By the folder names it looks like your building against opus-1.1 ?  Would performance not be better linked with opus-1.1.1-beta due to the arm optimizations?

7hunderbug, 2015-06-18:
Thanks for your feedback and the heads up @amigagamer, I'll certainly 
give that a go! My Opus branch is just an experiment to see if compiling 
Opus in statically and adding compiler (gcc 4.8) switches to optimise 
for the Arm7 core and FPU in the Raspberry Pi 2 would improve 
performance much (not that Moonlight-C needed it at all!) and to be 
honest, it doesn't seem to make a lot of difference so far. I will try 
with opus 1.1.1 beta and report back.

irtimmer, 2015-06-21:
Packages for Raspbian are available

closed 2015-06-21 by irtimmer