moonlight-stream/moonlight-embedded #106: Can't compile fake.c

captainju, 2015-05-11:
I guess it has something to do with h264bitstream not included in source list when Broadcom is not found.

```
CBi➜  build git:(master) ✗ make 
[  4%] Building C object CMakeFiles/moonlight.dir/src/video/fake.c.o
/home/cbi/temp/moonlight-embedded/src/video/fake.c:26:8: error: unknown type name ‘h264_stream_t’
 static h264_stream_t* stream;
        ^
/home/cbi/temp/moonlight-embedded/src/video/fake.c: In function ‘decoder_renderer_setup’:
/home/cbi/temp/moonlight-embedded/src/video/fake.c:32:3: warning: implicit declaration of function ‘h264_new’ [-Wimplicit-function-declaration]
   stream = h264_new();
   ^
/home/cbi/temp/moonlight-embedded/src/video/fake.c:32:10: warning: assignment makes pointer from integer without a cast
   stream = h264_new();
          ^
/home/cbi/temp/moonlight-embedded/src/video/fake.c:35:5: warning: implicit declaration of function ‘exit’ [-Wimplicit-function-declaration]
     exit(EXIT_FAILURE);
     ^
/home/cbi/temp/moonlight-embedded/src/video/fake.c:35:5: warning: incompatible implicit declaration of built-in function ‘exit’
/home/cbi/temp/moonlight-embedded/src/video/fake.c:35:10: error: ‘EXIT_FAILURE’ undeclared (first use in this function)
     exit(EXIT_FAILURE);
          ^
/home/cbi/temp/moonlight-embedded/src/video/fake.c:35:10: note: each undeclared identifier is reported only once for each function it appears in
/home/cbi/temp/moonlight-embedded/src/video/fake.c: In function ‘decoder_renderer_submit_decode_unit’:
/home/cbi/temp/moonlight-embedded/src/video/fake.c:53:10: error: ‘entry’ undeclared (first use in this function)
   while (entry != NULL) {
          ^
CMakeFiles/moonlight.dir/build.make:606: recipe for target 'CMakeFiles/moonlight.dir/src/video/fake.c.o' failed
make[2]: *** [CMakeFiles/moonlight.dir/src/video/fake.c.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/moonlight.dir/all' failed
make[1]: *** [CMakeFiles/moonlight.dir/all] Error 2
Makefile:75: recipe for target 'all' failed
make: *** [all] Error 2
```

irtimmer, 2015-05-11:
Fixed by 567c341a2b78a8214607fba4b950e37c7be17f5b

closed 2015-05-11 by irtimmer

captainju, 2015-05-11:
Great work !
it compiles very well on cubox-i (except the typo problem with LinkedBlockingQueue.h / platform.h from common)
:+1:  

I'll try running it this evening