moonlight-stream/moonlight-embedded #111: Xbox 360 wireless gamepad "Y" button not passing through to game stream

7hunderbug, 2015-05-17:
Xbox 360 wireless controller for PC with xboxdrv.
"Y" button doesn't register as input to the game stream.
Old java limelight doesn't have this issue.

Only difference between java common code and C common code I can see is
```#define Y_FLAG 0x8000``` is in https://github.com/irtimmer/moonlight-common-c/blob/master/limelight-common/Limelight.h
and ```public static final short Y_FLAG = (short)0x8000;``` in https://github.com/moonlight-stream/moonlight-common/blob/master/src/com/limelight/nvstream/input/ControllerPacket.java

I've been trying to find a way to debug this but don't have the skill or expertise to progress further.

I've confirmed "Y" doesn't register by streaming the AntiMicro gamepad software instead of Steam or a Game. https://github.com/Ryochan7/antimicro. All other input works as expected.

Here is my set up:

I set xboxdrv running on the Pi with "dpad as buttons" because Moonlight C API (Limelight.h) doesn't accommodate abs_dpad (as far as I can tell, only LeftStick, RightStick and buttons). Also, if I don't set "dpad as buttons", it causes erratic behaviour in Steam where as soon as you press the dpad once, there is a continuous left and up input:

```sudo xboxdrv --dpad-as-button --silent &```

evtest output as follows:

```
~$evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      Xbox Gamepad (userspace driver)
Select the device event number [0-0]: 0
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "Xbox Gamepad (userspace driver)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 294 (BTN_BASE)
    Event code 295 (BTN_BASE2)
    Event code 296 (BTN_BASE3)
    Event code 297 (BTN_BASE4)
    Event code 304 (BTN_A)
    Event code 305 (BTN_B)
    Event code 307 (BTN_X)
    Event code 308 (BTN_Y)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min   -32768
      Max    32767
    Event code 1 (ABS_Y)
      Value      0
      Min   -32768
      Max    32767
    Event code 3 (ABS_RX)
      Value      0
      Min   -32768
      Max    32767
    Event code 4 (ABS_RY)
      Value      0
      Min   -32768
      Max    32767
    Event code 9 (ABS_GAS)
      Value      0
      Min        0
      Max      255
    Event code 10 (ABS_BRAKE)
[... 377 lines cut ...]
reverse_rx = false
reverse_ry = false
abs_deadzone = 0
abs_dpad_x = -1
abs_dpad_y = -1
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 312
btn_tr = 313
btn_tl2 = 310
btn_tr2 = 311
btn_dpad_up = 294
btn_dpad_down = 295
btn_dpad_left = 296
btn_dpad_right = 297
```

7hunderbug, 2015-05-17:
I've added pull request https://github.com/irtimmer/moonlight-embedded/pull/112 to fix the trigger and shoulder button mapping issue.

cgutman, 2015-05-17:
Moonlight's C API supports D-pads the same way the Java one did. NVIDIA sends them as buttons so they are exposed as button flags. Conversion from abs_dpad to button flags is the responsibility of the client.

I also confirm non-functional Y button and reversed Y-axis on analog sticks. I'll take a look at those and see if I can find anything.

I had a mapping issue where pressing the left trigger would leave some events in the queue which get picked up when it goes to map the right trigger. I fixed it by draining the input queue after a delay when mapping. I'll submit a PR soon.

cgutman, 2015-05-17:
@7hunderbug I believe your d-pad issue is on input.c lines 399 and 411. The "else" and the "else if" cases are swapped. That means when d-pad is at 0,0 we get constant left and up buttons down.

Try swapping those and see if that fixes the issue.

closed 2015-05-18 by irtimmer

7hunderbug, 2015-05-20:
Thanks for those fixes @cgutman and @irtimmer.