moonlight-stream/moonlight-embedded #114: Moonlight Updater?

firestorm185, 2015-05-18:
I have a version of moonlight-embedded installed on my Raspi 2, but I came back and saw that this was updated a few days ago. Does anyone know what to do to update to the new version? Is there a java thing I have to type into the terminal, or is it manual? Thx!

scoobyd00, 2015-05-18:
The new version is written in 'C' and is still work in progress, a finally compiled release hasnt been released yet but its nearly there so i shouldnt think it will be much longer before irtimmer releases a new compiled release.
The new 'c' code doesnt use java and is much 'snappier' as ive tried an early build.

closed 2015-05-18 by irtimmer

firestorm185, 2015-05-18:
Thanks ScoobyD00! So would I just have to reinstall that release manually? I used the install script that guy made to install it initially.

scoobyd00, 2015-05-19:
Yes you will have to install manually.
Its just a simple case of copying over the new release to your SD card, install the dependencies which are now listed on the readme doc, pair up the new release and away you go.

HazCod, 2015-05-19:
No java? Awesome! Another dependency we don't need on OpenELEC. :-)

firestorm185, 2015-05-19:
Thanks! That helps.

firestorm185, 2015-05-19:
Agreed!

WilliamStam, 2015-07-27:
still no way to get the new C version installed on openelec?

HazCod, 2015-07-27:
@WilliamStam , not yet. You will need to compile it on raspbian (or ask someone for the binaries) and copy all required libraries.