moonlight-stream/moonlight-embedded #116: Gamepad Axes

Ilos, 2015-05-20:
Hey, so i have problem with gamepad mapping and i cant find a solution for this. Tried to fiddle around a bit but with no luck.
So the problem in short:
- L-Stick-y-axis is affected by "reverse_dpad_y"
- R-Stick-y-axis is affected by "reverse_ry"
- Dpad-y-axis is affected by nothing, it is reversed for me no matter what
- "reverse_y" does nothing, it has no effect on any axis.
(I didnt test the X-axes since they seem to work like they should.)
I tested it with 2 Controller (DS4 connected via Bluetooth, Xbox-One via USB)

And another question:
In src/input.c line:435, shouldn't        
```
dev->buttonFlags &= ~DOWN_FLAG;
dev->buttonFlags |= UP_FLAG;
```
be like
```
dev->buttonFlags &= ~UP_FLAG;
dev->buttonFlags |= DOWN_FLAG;
```
?

7hunderbug, 2015-05-20:
What driver are you using? xboxdrv or the xpad kernel module or something else?
What is the output for evtest for each controller?
The evtest output could help fine-tune the input handling logic in src/input.c to allow moonlight to better handle different controllers.
I've created a pull request to fix the D-pad Y-axis input handling https://github.com/irtimmer/moonlight-embedded/pull/117/commits

Ilos, 2015-05-20:
The driver for Sony is ds4drv (v 0.5.0), gamepad connectected via Bluetooth.
I didn't install any driver for the Xbox-gamepad, so i think i use the one delivered by the kernel.

evtest gives me
```
~ $ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      CHICONY USB Keyboard
/dev/input/event1:      CHICONY USB Keyboard
/dev/input/event2:      Microsoft X-Box One pad
/dev/input/event3:      Sony Computer Entertainment Wireless Controller
Select the device event number [0-3]: 3
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x54c product 0x5c4 version 0x111
Input device name: "Sony Computer Entertainment Wireless Controller"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_A)
    Event code 305 (BTN_B)
    Event code 306 (BTN_C)
    Event code 307 (BTN_X)
    Event code 308 (BTN_Y)
    Event code 309 (BTN_Z)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    127
      Min        0
      Max      255
      Flat       5
    Event code 1 (ABS_Y)
      Value    129
      Min        0
      Max      255
      Flat       5
    Event code 2 (ABS_Z)
      Value    132
      Min        0
      Max      255
      Flat       5
    Event code 3 (ABS_RX)
      Value      0
      Min        0
      Max      255
      Flat       5
    Event code 4 (ABS_RY)
      Value      0
      Min        0
      Max      255
      Flat       5
    Event code 5 (ABS_RZ)
      Value    124
      Min        0
      Max      255
      Flat       5
    Event code 6 (ABS_THROTTLE)
      Value    159
      Min   -16385
      Max    16384
    Event code 7 (ABS_RUDDER)
[... 123 lines cut ...]
      Value      0
      Min       -1
      Max        1
Properties:
Testing ... (interrupt to exit)
Event: time 1432107273.970684, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 1
Event: time 1432107273.970684, -------------- SYN_REPORT ------------
Event: time 1432107274.114692, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 0
Event: time 1432107274.114692, -------------- SYN_REPORT ------------
Event: time 1432107274.670773, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 1
Event: time 1432107274.670773, -------------- SYN_REPORT ------------
Event: time 1432107274.802777, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 0
Event: time 1432107274.802777, -------------- SYN_REPORT ------------
Event: time 1432107275.690890, type 3 (EV_ABS), code 16 (ABS_HAT0X), value -1
Event: time 1432107275.690890, -------------- SYN_REPORT ------------
Event: time 1432107275.810908, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 0
Event: time 1432107275.810908, -------------- SYN_REPORT ------------
Event: time 1432107276.787041, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value -1
Event: time 1432107276.787041, -------------- SYN_REPORT ------------
Event: time 1432107276.903090, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 0
Event: time 1432107276.903090, -------------- SYN_REPORT ------------
```

With your button-flag update in input.c the dpad-y-axis gives me now correct directions. 
But the new problem is, that if i press down, the "down-command" stuck in this direction. Only pressing up releases this command. So tap down on gamepad is like holding the button, till i tap the up button. (it happens with both gamepads, evtest shows no issues, only on moonlight).

7hunderbug, 2015-05-20:
Could you post your moonlight mapping files please? Did you check that your mapping files have the correct evdev event codes for the respective controller actions?

Ilos, 2015-05-20:
my Dualshock-file

```
~/Moonlight $ cat Sony.map
abs_x = 0
abs_y = 1
abs_z = 3
reverse_x = false
reverse_y = false
abs_rx = 2
abs_ry = 5
abs_rz = 4
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = 16
abs_dpad_y = 17
reverse_dpad_x = false
reverse_dpad_y = true
btn_north = 307
btn_east = 306
btn_south = 305
btn_west = 304
btn_select = 312
btn_start = 313
btn_mode = 316
btn_thumbl = 314
btn_thumbr = 315
btn_tl = 308
btn_tr = 309
btn_tl2 = -1
btn_tr2 = -1
btn_dpad_up = -1
btn_dpad_down = -1
btn_dpad_left = -1
btn_dpad_right = -1
```
and my Xbox-file
```
~/Moonlight $ cat Xbox.map
abs_x = 0
abs_y = 1
abs_z = 2
reverse_x = false
reverse_y = false
abs_rx = 3
abs_ry = 4
abs_rz = 5
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = 16
abs_dpad_y = 17
reverse_dpad_x = false
reverse_dpad_y = true
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = -1
btn_tr2 = -1
btn_dpad_up = -1
btn_dpad_down = -1
btn_dpad_left = -1
btn_dpad_right = -1
```

The Xbox-pad is essentially just their to test, if it is a global issue or has something to do with just the Dualshock. I will only use the Sony-pad since this is a pad i can use wireless.
The problem still persists if i unplug the xbox pad and only configure everything for the Sony-pad.
I checked the mapping-files against the evdev-events at least 3 times. I cant find any missmatches in the mapping-files.

Like is said, "reverse_y" is set to false, since this does no changes at all. 
abs_y is affected by reverse_dpad_y. 
Strange thing is, abs_ry is affected by reverse_ry.

7hunderbug, 2015-05-20:
Comparing your evtest output with your mapping file, it looks like your Sony mapping file is all wrong.
You need to recreate the mapping file each time the moonlight code is updated or recompiled because changes in src/input.c could affect the outcome of the mapping procedure.

I think your sony.map should look like this:
```
abs_x = 0
abs_y = 1
abs_z = 2
reverse_x = false
reverse_y = false
abs_rx = 3
abs_ry = 4
abs_rz = 5
reverse_rx = false
reverse_ry = false
abs_deadzone = 0
abs_dpad_x = 16
abs_dpad_y = 17
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 
btn_tl = 310
btn_tr = 311
btn_tl2 = -1
btn_tr2 = -1
btn_dpad_up = -1
btn_dpad_down = -1
btn_dpad_left = -1
btn_dpad_right = -1
```

I'm not sure if the trigger buttons on your Sony controller are ABS_Z and ABS_RZ or ABS_THROTTLE and ABS_RUDDER. If the trigger buttons don't work the following might be more correct:
```
abs_x = 0
abs_y = 1
abs_z = 7
reverse_x = false
reverse_y = false
abs_rx = 3
abs_ry = 4
abs_rz = 6
reverse_rx = false
reverse_ry = false
abs_deadzone = 0
abs_dpad_x = 16
abs_dpad_y = 17
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 
btn_tl = 310
btn_tr = 311
btn_tl2 = -1
btn_tr2 = -1
btn_dpad_up = -1
btn_dpad_down = -1
btn_dpad_left = -1
btn_dpad_right = -1
```

Ilos, 2015-05-20:
huh, thats interessing.
According to the evtest my L-Stick has the 0&1-axis, R-Stick has 3,4-axis and triggers 2&5.
But creating a mapping file gives me the one above (L->0,1 | R->2,5 | Trigger -> 3,4). And it also works with this setup. I only have problems with the reversion of some axis. If i put the evtest-values in the mapping-file, then the controlls are all messed up.
The THROTTLE-,RUDDER-,WHEEL-,DISTANCE-,TILT_X-,TILT_Y-ABS are for the sixaxis. The Dualshock 4 has a gyroscope build in and i could use it, if i want to (which i won't).

The second problem i have is, that the Triggers are kinda not working as axes. If i use the settings above, then this happens:
If i press the triggers ~3/4 the way down, then i get a value of 97%.
If i press the triggers fully down, i got a value of ~4%. 
I cant reach 100%. So i cant really use them as axis, and have to use them as buttons ("abs_z = -1 | abs_rz = -1 | btn_tr2 = 310 | btn_tl2 = 311")

The Dualshock worked with the java-limelight build, but i dont want to use it, since i got some video-perfomance issues and the c-version is much snappier and all around better.

7hunderbug, 2015-05-20:
The input.c gamepad logic is definitely messed up. I've been trying to make sense of it myself because I still have a few issues too. The problem is, as we've seen, that certain controllers and/or certain gamepad drivers don't provide consistent input event names or even consistent minimum and maximum values for each input to libevdev. I imagine that this makes it difficult to come up with one set of logic to cope with all gamepad/driver combinations.

Could you run ds4drv using the ```--emulate-xboxdrv``` switch and provide the initial evtest output and a new mapping file created again please?

Do you use a conf file for ds4drv? If so, can you post that also please?

I'm guessing that ds4drv is not supplying expected results to evdev. For example, if your triggers really are ABS_Z and ABS_RZ, then they seems to have a non-zero value at rest, between the minimum and maximum values, which doesn't seem right. It looks like ABS_RX and ABS_RY could be your triggers instead, which again, doesn't seem right.

Ilos, 2015-05-20:
evtest with emulated xboxdrv
```
~/Moonlight $ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      CHICONY USB Keyboard
/dev/input/event1:      CHICONY USB Keyboard
/dev/input/event2:      Xbox Gamepad (userspace driver)
Select the device event number [0-2]: 2
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "Xbox Gamepad (userspace driver)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_A)
    Event code 305 (BTN_B)
    Event code 307 (BTN_X)
    Event code 308 (BTN_Y)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    124
      Min        0
      Max      255
      Flat       5
    Event code 1 (ABS_Y)
      Value    126
      Min        0
      Max      255
      Flat       5
    Event code 3 (ABS_RX)
      Value    130
      Min        0
      Max      255
      Flat       5
    Event code 4 (ABS_RY)
      Value    126
      Min        0
      Max      255
      Flat       5
    Event code 9 (ABS_GAS)
      Value      0
      Min        0
      Max      255
      Flat       5
    Event code 10 (ABS_BRAKE)
      Value      0
      Min        0
      Max      255
      Flat       5
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
Properties:
Testing ... (interrupt to exit)
^C
```
And the assosciated mappingfile
```
abs_x = 0
abs_y = 1
abs_z = 10
reverse_x = false
reverse_y = true
abs_rx = 3
abs_ry = 4
abs_rz = 9
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = 16
abs_dpad_y = 17
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = -1
btn_tr2 = -1
btn_dpad_up = -1
btn_dpad_down = -1
btn_dpad_left = -1
btn_dpad_right = -1
```
I'm not using any custom ds4drv-config. I use the original one.
[ds4drv.conf](http://pastebin.com/rMmXS2E4)

With the emulated xboxdrv i got the exact same issues. nothing changed.

scoobyd00, 2015-05-20:
@7hunderbug 
Ive complied your latest master release and its working great, only issue is the controller.
Im using xboxdrv with a wireless xbox360 controller. The controller works fine except for the x and y axsis are switched on the left thumb stick. Pressing up goes down, and down goes up!.

Ive tried altering the mapping file but nothing seems to have any effect.
Do you have this problem?.

7hunderbug, 2015-05-20:
@scoobyd00, yes I'm having the same problem but it's just the y-axis with an issue on both thumb sticks; x-axis (left and right) seems to be fine. I'm also having the same issue as you and @Ilos in that changing the reverse_y or reverse_ry mapping in the mapping file has no effect. I'm trying to make sense of the logic in the code but I'm not skilled enough to rationalise it! So I'm hacking away to try and come up with a solution.

scoobyd00, 2015-05-20:
Excellent work @Ilos that commit does indeed fix the issue.

Ilos, 2015-05-20:
Thanks, now the issues with the axes should be fixed.
Still i got the problem with the "stuck" dpad_down_btn. Try to find something thats fixes this.
Do you have this issue, @scoobyd00 ?

scoobyd00, 2015-05-20:
@Ilos  No i dont seem to have an issue with the dpad buttons at all, they are all working fine.
I am using a xbox360 wireless controller though.

7hunderbug, 2015-05-20:
Good find @Ilos! I've found out why the default mappings for your Sony Dual Shock seem wrong. ds4drv has them mapped in an unusual way and I can't think of a reason why it has been done this way. See here: https://github.com/chrippa/ds4drv/blob/master/ds4drv/uinput.py
This is the code:
```
create_mapping(
    "ds4", "Sony Computer Entertainment Wireless Controller",
    # Bus type,     vendor, product, version
    ecodes.BUS_USB, 1356,   1476,    273,
    # Axes
    {
        "ABS_X":        "left_analog_x",
        "ABS_Y":        "left_analog_y",
        "ABS_Z":        "right_analog_x",
        "ABS_RZ":       "right_analog_y",
        "ABS_RX":       "l2_analog",
        "ABS_RY":       "r2_analog",
        "ABS_THROTTLE": "orientation_roll",
        "ABS_RUDDER":   "orientation_pitch",
        "ABS_WHEEL":    "orientation_yaw",
        "ABS_DISTANCE": "motion_z",
        "ABS_TILT_X":   "motion_x",
        "ABS_TILT_Y":   "motion_y",
    },
```

It might be an idea to emulate xboxdrv behaviour for your Dual Shock, and use native xboxdrv for your Xbox controller too (```modprobe -r xpad``` first). I think the xpad kernel driver works ok for USB Xbox controllers but has slightly different input event names than xboxdrv and has issues with flashing LEDs on some controllers (mostly wireless ones I think).
With regards to your D-pad - if you check in evtest that your D-pad only creates a value "1" for right or down and a value "-1" for left and up then it should work the same as mine and @scoobyd00 gamepads.... hopefully.

Ilos, 2015-05-20:
Found out, why my triggers had this issues with the wrong values. Was hard to find this small mistake in the code :P

7hunderbug, 2015-05-20:
My only issue is having to move thumbsticks up instead of down so that Y-axis is mapped properly - it seems inconsistent with the code but it works and I can live with it :) Good job all!

scoobyd00, 2015-05-20:
@Ilos  Ive just recompiled with your latest commit and although my triggers were working correctly before your commit still makes them work fine now.

Ilos, 2015-05-20:
@scoobyd00, my last commit shouldn't effect your controls, since you use your triggers as buttons, not as axes.

@7hunderbug, i will look into the ds4drv stuff, this looks really messed up. 
But for now i'm happy, that the controls work for me like intended. Thanks for all!

7hunderbug, 2015-05-22:
@Ilos I'm using the triggers as analogue inputs, not buttons, and your patch doesn't adversely affect my control mappings either so all good.
@scoobyd00 your mapping file has the same entries for reverse_y = true and reverse_ry=true as mine, which are required for our Xbox controllers however, after this commit https://github.com/irtimmer/moonlight-embedded/commit/7c639ab04bdac27f9e4bdd845de417f695fdb4ea, you are promted to push up instead of down to ensure the mapping logic of "detect if movement is reversed or not" works correctly and the resultant mapping file has reverse set to true.

closed 2015-05-25 by Ilos