moonlight-stream/moonlight-embedded #119: Static Noise During Certain Cutscenes

vololo0, 2015-05-20:
Environment:
Raspberry Pi 2
All Prerequisites installed
OS: Retropie
Game: Mortal Kombat X via Steam.
GFE Version:  2.4.3 (Latest)

Issue:
Hi i have an issue with static noises during certain cut scenes or game play of all the games being streamed at 1080p 30fps via HDMI. It runs like a charm the only problem is just the static noise encountered during certain parts of the game.

HazCod, 2015-05-28:
Try a different cable, different input on your TV or leaving the optical out unplugged.

vololo0, 2015-05-29:
HI i have already tried, problem still persist. Thank you for the suggestion. I have also tried a freshly installed Retropie and the problem is still there. I just hear white noise during cut scenes. But when during the actual gameplay the audio is perfect.

scoobyd00, 2015-05-29:
If its only doing it on a couple of games then it looks like its an issue with GeForce experience.

vololo0, 2015-05-30:
I tries streaming to my phone and the i had the same issue. @scoobyd00 you are right it the problem with the gfe. Thannk you

cgutman, 2015-05-30:
Since a few releases ago, GFE stops sending audio packets at all if the program isn't outputting any audio (rather than sending silent samples as they used to). This causes a buffer underrun on the client when the audio stream stops mid-stream. Depending on the client, hardware, OS, etc this can be fine or cause noticeable clicks, pops, etc.

vololo0, 2015-05-30:
@cgutman hi , Mine is more of a white noise just like this https://www.youtube.com/watch?v=t0I4mTEdAf8. and its not random. Do you know of a fix or workaround for this issue? I have look at many posts and forums but i cant find any problem similiar to mine.

Any my pc specs:

GRAPHICS: GTX 780
CPU : I7 4970k
RAM: 32GB

vololo0, 2015-06-03:
Found out the issue!! The issue was unique to my setup, i have an apple display with built in speakers and i had many audio devices active. I deactivated the display audio and used NVIDIA High Definition Audio instead to output my pc audio. No more static problem! issue resolved!. Thank you All!!!

closed 2015-06-03 by vololo0