moonlight-stream/moonlight-embedded #123: RTSP OPTIONS request failed + nvstreamer crash

pbourachot, 2015-05-22:
Hi all,

I have issue with the new C version receiving some RTSP OPTIONS request failed.

As i saw in a previous thread, i was using Ip adress without any success  :  moonlight stream 192.168.2.110.

But on the server side, i saw launching moonlight kill the nvstreamer process.

It was working fine with java version.
Geforce driver is 2.4.3.31.

Thanks

Pascal

scoobyd00, 2015-05-22:
What version of Linux are you running on?
I don't have any issues with raspbian.
Have you made sure you've got all the dependencies installed?

pbourachot, 2015-05-22:
I've tried it on a fresh raspbian install  using and following https://github.com/7hunderbug/moonlight-embedded/releases.

I suspect more something wrong on server side.
I've tried to launch nvstreamer on my server in dos command, and once moonlight try to connect, it crashes.

Something strange, the date of the error log of nvstreamer is 2 hours before of the standard logs.
But server and raspberry are at the same hour.

I will try to add more logs details in the next days.

Java version is working fine.

Thanks.

Pascal

7hunderbug, 2015-05-22:
I would recommend a clean install of all Nvidia related software. Reinstall all Nvidia drivers and Geforce Experience and when the installer prompts you, choose "Advanced" and then "Clean Install" so that the installer wipes any old driver configuration and files before installing the new ones.
I would also reset the Game Stream devices associations - GFE->Preferences->Game Stream->Shield Devices->Forget Devices and then pair your Raspberry Pi to your PC again.
Any issues I've had in the past with game streaming crashing have been solved by doing this.

pbourachot, 2015-05-25:
Complete re-installation of NVidia vomponent fixed it.
Thank a lot.

closed 2015-05-25 by pbourachot