moonlight-stream/moonlight-embedded #126: Audio stuttering (HDMI)

r3dfaction, 2015-05-30:
2 days ago i have bought raspberry pi b+, i have installed RASPBIAN with preinstalled noobs setup and then the moonlight-embedded. With HDMI cable it was working perfectly with keyboard or any mouse i have connected. Yesterday i bought xbox 360 usb controller and i thought it might be cool to update all packages and install new firmware, but after this entire upgrade, audio started to stutter when i stream steam application or any running game with or without any device connected.

Tried almost all suggestions posted all over the web.

I installed RASPBIAN from scratch, but the version i have downloaded was already higher than the preinstalled one.
My GeForce Experience is up to date: 2.4.5.28
Firmware version: raspberrypi 3.18.14+
I don't have PulseAudio installed.

Configured my config to support HDMI as far as possible:
hdmi_force_hotplug=1
hdmi_force_edid_audio=1
hdmi_drive=2

used ´amixer cset numid=3 2´ and ´raspi-config´ to set my hdmi audio output, but it was unsuccessfully as everything else i tried, anyone had and solved this problem?

scoobyd00, 2015-05-31:
I've had a problem like this but it was due to bitdefender anti virus causing an issue. I've also found teamviwer causes problems. It could be some software that you're recently installed on your PC?
I'm running the latest raspbian release although I'm on a pi 2 as it gives much better performance, some people have found that going from a b+ to a pi 2 has sorted out their sound stutter.

Can you go back to your original noobs release and see if this resolves the issue . I'd also try reinstalling geforce exerience.
A fresh install of raspbian, no system upgrade, fresh install of geforce experience should be fine. You can then try attaching the Xbox controller once you have a working setup.

r3dfaction, 2015-05-31:
thank you for your answer, i haven't installed anything new and i tried many new things:
- to use different cables
- to reinstall GeForce Experience
- to deactivate running software
- to use both latest limelight versions
- to reinstall Raspbian using NOOBS installer 1.4.1 (i don't think this is the version, which was installed when i bought raspberry pi, i will try 1.3.x later)

stuttering and even sound disappearing problem still persists, i am confused what else could it be

ChrisOwen101, 2015-05-31:
Hi, I'm experiencing the same audio stuttering also. I got a brand new RPi2 yesterday and installed the newest version of NOOBS and Limelight on it just now. Video seems to be working fine but audio and stutters alot.

I have the most up to date GeForce Experience and Limelight installed. Do you have any more idea on how to fix this bug?

scoobyd00, 2015-05-31:
Im not sure if its something to do with the noobs install of Raspbian but im using the standalone Raspbian image from the 5/5/2015 with raspbian version number  3.18.7-v7+
This image along with the latest Geforce experience and the latest compile 'c' of moonlight is working fine on my pi 2.

Im not sure why the audio stuttering is occuring, try the latest raspbian release without using noobs and dont do a system update as i found this on the pi forums, not sure if its related to your issues but working looking into...

This was from a post dated 15/5/15.
A warning: yesterday I did a sudo apt-get upgrade to update my raspbian system. However, it broke my streaming setup (GFE 2.1.3, limelight 1.2.2)! 
Problems occuring after the update: Sound was stuttering extremely (unplayable) and I got multiple messages "SEVERE: error code during decode -6" during playing. While the video stream seemed to be ok, this has NEVER happened before. 

I managed to find the offending packages (excerpt from my apt-get history):
CODE: SELECT ALL
Start-Date: 2015-05-14  16:59:42
Commandline: apt-get upgrade
Upgrade: libraspberrypi0:armhf (1.20150214-2, 1.20150421-1), raspberrypi-bootloader:armhf (1.20150214-2, 1.20150421-1), libraspberrypi-bin:armhf (1.20150214-2, 1.20150421-1), libraspberrypi-dev:armhf (1.20150214-2, 1.20150421-1), libraspberrypi-doc:armhf (1.20150214-2, 1.20150421-1)
End-Date: 2015-05-14  17:02:17

Downgrading them to the old versions manually seems to fix the problem.

r3dfaction, 2015-05-31:
@scoobyd00 :thumbsup: yes, this fixed also my issue too!

ChrisOwen101, 2015-06-01:
What is the simplest way of downgrading these packages? I'd rather not do a full reinstall of the OS.

r3dfaction, 2015-06-01:
@ChrisOwen101 just remove installed packages by typing:
apt-get remove raspberrypi-bootloader
and then install the old version:
apt-get install raspberrypi-bootloader=1.20150214-2
same for other packages, this should work

AgentMcMuffin, 2015-06-06:
@r3dfaction hey Im having the audio stutter issues as well with moonlight on my RPi 2. I tried using 'sudo apt-get install raspberrypi-bootloader=1.20150214-2' but all I get is that it couldn't find that version of raspberrypi-bootloader. What am I doing wrong?

nanotechnician, 2015-06-06:
I did not have any of this till I got a kernel panic (testing other things, nothing related to this) and flashed new raspbian img... suddenly moonlight audio stuttering all over the place, came here to check... nothing can help... but then... decided to try last raspbian img (the one that was on the SD I was using) and all went back to normal, and its working like it should be, FLAWLESSLY EXCELLENT.

RASPBIAN IMAGE USED IS 2015-02-16-raspbian-wheezy

hope this can help.

r3dfaction, 2015-06-06:
@AgentMcMuffin
have you deleted the old version? have you internet connection?
try `apt-get update` before installation

@nanotechnician
the problem are those updated packages, old image are using old packages, that's why it is working flawlessly. This can be a bug created by raspberry developers or audio improvement, which should be implemented in limelight, can't tell exactly, but this is a bug for sure.

H3llsing, 2015-06-07:
Hey, i got the same problem using osmc on rpi2 not overclocked. Video smooth and sound stuttering as hell. Anybody knows which osmc version to use or which packages i should downgrade?

AgentMcMuffin, 2015-06-08:
thanks man

r3dfaction, 2015-06-08:
@H3llsing have you tried to follow the post, which scoobyd00 found at raspberry forums? and do experience stuttering only in osmc?

H3llsing, 2015-06-08:
Im new to linux and pi, but i'll try theway with rasbian today. Right now im doing a backup of my existing OSMC. Then i will download the rasbian version and install limelight later. I will report you the results.

In OSMC using Kodi, everything runs perfect. 
With limelight in the osmc enviroment the video is perfect, the sound is stuttering terrible over HDMI. With and without connected devices. Im using the newest GEF and my OSMC was updated to the latest version via menu update.
i also edited the boot.cfg with:

hdmi_force_hotplug=1
hdmi_force_edid_audio=1
hdmi_drive=2
amixer cset numid=3 2

Also tried to raise the GPU_mem to 512MB because i read somewhere that low gpu mem will cause this sound stuttering, but 256MB should be enough.

Im also verified that pulseAudio is not installed.

PS: Thx for the great support and project!

AgentMcMuffin, 2015-06-10:
@r3dfaction still not working. It replies with "Version '1.20150214-2' for 'raspberrypi-bootloader' was not found'

Don't understand what I'm doing wrong?

diagonali, 2016-01-04:
Can someone please let us know how to downgrade the packages or otherwise follow the instructions/info given by @scoobyd00? I'm running OSMC and am having terrible trouble with audio stuttering when the video is otherwise fine. Would really appreciate step by step commands to enter via SSH or else is there anything else can be done to help with the audio stuttering? Thanks.

cedarrapidsboy, 2017-01-07:
I'm running a Pi 3 and have tried moonlight-embedded with stock Raspbian and the pre-made Retropie image. In both, streaming from Windows 10 (Steam Big Picture) presents audio breaks. Short (.5-1s) instances of muted sound. Not predictable. The sound is not out of sync, just goes away for split seconds occasionally. 

Is this the same problem? A year since the last update here... downgrading packages seems a little risky.

To reduce the variables, I installed a lite Raspbian on a new SD card and only installed moonlight (2.2.2). The problem presented itself when playing a game via Steam big picture (Rogue Legacy, specifically).

I do have a SteamLink in the house without this problem. And, moonlight on my chrome book also does not have this problem. The Pi 3 is ethernet wired with a Canakit 2.5 amp PSU.

Any suggestions or elaboration on the packages that need to be changed.

closed 2017-03-18 by irtimmer