moonlight-stream/moonlight-embedded #127: Please add VDPAU support

pomkac, 2015-05-30:
Hi all! I have Cubietruck, moonlight not working on it. Can you add VDPAU support for SUNXI? 
I'm not a software programmer in C, so it is difficult to add it to moonlight. Please, help. For C developer this is one day. Link for SUNXI-decoder: https://github.com/linux-sunxi/libvdpau-sunxi/blob/master/decoder.c

pomkac, 2015-07-12:
I wrote the code to support VDPAU. How can I add it to the repository?

irtimmer, 2015-07-12:
You could fork the repository, commit your changes and fill a pull request. https://help.github.com/articles/fork-a-repo/
Or you could just dump the code somewhere and hope someone else pushes it to the repository.

irtimmer, 2015-08-08:
I you have troubles committing the code please let me know. I really like to add VDPAU support.

pomkac, 2015-08-09:
I create commit jn next week. It is necessary to improve the functionality, but I did not have enough time. A lot of work has accumulated. (

aPollO2k, 2015-09-29:
I'm waiting for this very excited. I own a banana pi and it would be great to run moonlight on it.

pomkac, 2015-10-12:
http://www.filedropper.com/repo
This is patch to VDPAU support. I'm still going to optimize it, but it is work.

irtimmer, 2015-10-12:
After making a few changes because the code was based on an older version of Moonlight, I have committed it to the branch [vdpau2](https://github.com/irtimmer/moonlight-embedded/tree/vdpau2).

closed 2016-01-16 by irtimmer