moonlight-stream/moonlight-embedded #130: DualShock 3 Analog sensitivity

Serg86, 2015-06-06:
Hey, I've decided to give moonlight a try yesterday and I love it!

After dozens of attempts at mapping the DualShock 3 buttons, I finally managed to get it right:

    reverse_y=false
    reverse_x=false
    abs_ry=5
    btn_dpad_right=293
    abs_rx=2
    btn_start=291
    btn_tl2=296
    btn_select=288
    btn_tr2=297
    btn_west=303
    btn_east=301
    btn_tr=299
    btn_south=302
    btn_dpad_left=295
    btn_mode=704
    btn_tl=298
    btn_thumbr=290
    btn_dpad_up=292
    btn_north=300
    btn_dpad_down=294
    abs_y=1
    abs_x=0
    btn_thumbl=289
    reverse_ry=false
    reverse_rx=false

I had to manually switch Square/Circle (X/B on XBox 360 pads), L1/L2 (LB/LT) and R1/R2 but it works now. However, the analog stick sensitivity is too low. In Witcher 3 for example, if you tilt it forward all the way, Geralt (Main Character) walks forward, he is supposed to run at this point. He starts running when the analog stick is in a diagonal position. I have no trouble playing games that do not require precise analog control.

Is there any way to adjust the sensitivity?

I'm Limelight 1.22 with OpenELEC 5.08 on Raspberry Pi 2.

Thank you.

theharleyquin, 2015-06-08:
I can second this. I tried this with a PS3 and a PS4 controller and had the same issue in Skyrim. Apperars to be na issue in games were you can tilt the left analog stick slightly to have a character walk vs fully tilting it to run. Bastion was no issue because I think left analog is "all or nothing" with a light tilt or fully tilt.

Attached are my PS3 and PS4 mappings:

#Gamepad PS4
#Sat Jun 06 18:00:13 EDT 2015
reverse_y=false
reverse_x=false
abs_ry=5
abs_rx=2
btn_start=313
btn_tl2=310
btn_select=312
btn_tr2=311
btn_west=304
btn_east=306
btn_tr=309
btn_south=305
btn_mode=316
btn_tl=308
btn_thumbr=315
btn_north=307
abs_y=1
abs_x=0
btn_thumbl=314
reverse_ry=false
reverse_dpad_y=false
reverse_rx=false
reverse_dpad_x=false
abs_dpad_y=17
abs_dpad_x=16

#Gamepad PS3
#Mon Jun 08 19:40:37 EDT 2015
reverse_y=false
reverse_x=false
abs_ry=5
btn_dpad_right=293
abs_rx=2
btn_start=291
btn_tl2=298
btn_select=288
btn_tr2=299
btn_west=303
btn_east=301
btn_tr=297
btn_south=302
btn_dpad_left=295
btn_mode=704
btn_tl=296
btn_thumbr=290
btn_dpad_up=292
btn_north=300
btn_dpad_down=294
abs_y=1
abs_x=0
btn_thumbl=289
reverse_ry=false
reverse_rx=false

Serg86, 2015-06-11:
I managed to create a workaround. I downloaded the "xboxdrv" executable from the Raspbian repo, scrambled together the dependencies manually and it works for me.

Even bluetooth.

Serg86, 2015-06-11:
I managed to create a workaround using xboxdrv.

Using xboxdrv everything works just fine. On OpenELEC there is no package manager so I had to manually download it and it's dependencies.

Even bluetooth works.

theharleyquin, 2015-06-12:
@Serg86 Can you specify what steps to use after the xboxdrv install? Installed it via: 

sudo apt-get install xboxdrv

But using an already made mapping the sensitivity is still low. Did you have to redo your mapping?

theharleyquin, 2015-06-13:
Since I've had it with PS# controllers I found a USB gamepad (Logitech DualAction). After creating a mapping file the same analog in-sensitivity occurs. So it seems that sensitivity isn't isolated to only PS# controllers.

Serg86, 2015-06-13:
@theharleyquin Sorry was pretty busy the last few day. If you plug it in with a USB cable, when you fire up xboxdrv it should tell you it found a DualShock 3 controller. You can then simply make a new mapping file for it with moonlight.

If you are using bluetooth, you'll probably have to tell it the device manually, and the key bindings as explained here: https://github.com/xboxdrv/xboxdrv/issues/54

I just took the whole line provided in the second comment, changed event number to 0 and ran with it. There were a few issues like inverted Y axis on both analog sticks, but I ironed them all out with this mapping file:

    reverse_z=false
    reverse_y=true
    reverse_x=false
    abs_rz=9
    abs_ry=4
    abs_rx=3
    btn_start=315
    btn_tl2=310
    btn_select=314
    btn_tr2=311
    btn_west=307
    btn_east=305
    btn_south=304
    btn_mode=316
    btn_thumbr=318
    btn_north=308
    abs_z=10
    abs_y=1
    abs_x=0
    reverse_rz=false
    btn_thumbl=317
    reverse_ry=true
    reverse_dpad_y=false
    reverse_rx=false
    reverse_dpad_x=false
    abs_dpad_y=17
    abs_dpad_x=16

If you use xboxdrv in a script, you have to end the line with an "&", this will make sure that the script continues after it starts up xboxdrv. If you don't end it with an & it will simply stop and wait for xboxdrv to close before it continues. I then simply do a "killall xboxdrv" to close it again.

I think it is possible to run xboxdrv as a service, which is perhaps the better option, though as I use OpenELEC and had to manually download it and it's dependencies, I don't know if I would be able to achieve that or how I would even go about it honestly.

Right now everything works perfectly for me, I can start up moonlight from within Kodi, the controller works like a charm and I've been enjoying playing the Witcher 3 lately.

7hunderbug, 2015-06-13:
I have the same problem using a Nintendo Wii Classic controller with wminput. The left hand analogue thumbstick has a much lower "limit" then the right hand one and so any games where the left hand stick is used for acceleration, driving or walking, you can never reach full speed.

7hunderbug, 2015-06-14:
The output of ```evtest``` for Nintendo Wii Classic Controller, connected to ```wminput```, with measured values of the limits of the analogue inputs shown at the bottom, is as follows:
```
$ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Xbox Gamepad (userspace driver)
/dev/input/event1:	Nintendo Wiimote
Select the device event number [0-1]: 1
Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x1 product 0x1 version 0x3
Input device name: "Nintendo Wiimote"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_A)
    Event code 305 (BTN_B)
    Event code 307 (BTN_X)
    Event code 308 (BTN_Y)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min       -1
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min       -1
      Max        1
    Event code 16 (ABS_HAT0X)
      Value     31
      Min        0
      Max       63
    Event code 17 (ABS_HAT0Y)
      Value     29
      Min        0
      Max       63
    Event code 18 (ABS_HAT1X)
      Value     16
      Min        0
      Max       31
    Event code 19 (ABS_HAT1Y)
      Value     16
      Min        0
      Max       31
Properties:
Testing ... (interrupt to exit)
...
Event: time 1434286457.876534, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 55
...
Event: time 1434286488.853186, type 3 (EV_ABS), code 17 (ABS_HAT0Y), value 5
...
Event: time 1434286508.669508, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 56
...
Event: time 1434286541.726803, type 3 (EV_ABS), code 16 (ABS_HAT0X), value 6
...
Event: time 1434286568.583548, type 3 (EV_ABS), code 19 (ABS_HAT1Y), value 28
...
Event: time 1434286585.033798, type 3 (EV_ABS), code 19 (ABS_HAT1Y), value 3
...
Event: time 1434286610.197719, type 3 (EV_ABS), code 18 (ABS_HAT1X), value 29
...
Event: time 1434286627.489083, type 3 (EV_ABS), code 18 (ABS_HAT1X), value 4
...
```

This shows two things:

1. The evtest reported maximum values for the ABS_HAT* inputs can not actually be reached in practice (theoretical 0-63 compared to actual 5-55 on ABS_HAT0Y for example).
2. Probably most relevant to this issue is the fact that the resolution of the left stick is twice that of the right stick. ABS_HAT0X/Y has resolution of 50 (55-5 or 56-6) and ABS_HAT1X/Y has a resolution of 25 (28-3 or 29-4).

It would be interesting to see the output of ```evtest``` for your PS3 and PS4 controllers and your Logitech DualAction to see if the resolution is different between each analogue stick on those controllers also.

jonwingrove, 2015-06-22:
I found this with a generic USB controller - my left stick Y axis seems to not give the full range.

Running evtest - I get values going from 0 to 255 (127 at neutral) on both the X and Y axes. - it looks like I'm getting the full range in evtest.

But, when I connect through Limelight (1.2.2 - haven't tried the new Moonlight, will do tonight) - I load up "Gamepad Test" (google it - just shows gamepad input in a browser window) in the Steam browser. My X axis has full range, but the Y axis only goes about 80% in either direction.

irtimmer, 2015-06-22:
Not getting the full range for some gamepads is already fixed in master :P c7547bf771f02beba6d9bc44b3608f29b96d0c45

theharleyquin, 2015-06-22:
Excited to test it out

jonwingrove, 2015-06-22:
Awesome :)

closed 2015-06-23 by Serg86