moonlight-stream/moonlight-embedded #131: No Audio Over HDMI

Krazypoloc, 2015-06-08:
I'm also not getting any audio over HDMI. I can add some variety to the mix. Details of my configuration are below.

Rasplex 0.6.0 (Openelec)
java version "1.8.0_33"
Java(TM) SE Runtime Environment (build 1.8.0_33-b05)
Java HotSpot(TM) Client VM (build 25.33-b05, mixed mode)
limelight 1.2.2 with matching libopus release

This is my command to launch limelight
#java -jar limelight.jar stream limelight -input /dev/input/event0 -720

This is a snippet of the error I'm getting.
[code]com.limelight.nvstream.av.audio.AudioStream$1.run(Unknown Source)[/code]

Here's the entire thing.
http://pastie.org/private/9t8dgqmnxpalaflljgyucw

Here is my confit.txt
http://pastie.org/private/wsjxupkljojm2bezkxsyw

I've tried GTAV, Project Cars, and Broforce all through Steam with no audio outputting in the Steam menus or in-game.

I'll try and launch one of them with the -app command and report back with my findings.

Its also worth noting that this is the first time I've tried this so I'm also going to attempt to downgrade limelight to a previous version.

7hunderbug, 2015-06-08:
This line in your config.txt: 
```amixer cset numid=3 2```
does not belong there. You need to run that command from the bash shell so that audio is routed via HDMI. See here http://elinux.org/R-Pi_Troubleshooting#Sound_does_not_work_at_all.2C_or_in_some_applications

7hunderbug, 2015-06-08:
Also, what is the output of ```aplay -l``` when run from the command line?

Krazypoloc, 2015-06-09:
Ok I remove the line from my config.txt.

Apparently Openelec doesn't use amixer, from all I can find it uses OMXplayer.
http://elinux.org/Omxplayer

This is confirmed by this.
aplay -l
aplay: device_list:268: no soundcards found...

7hunderbug, 2015-06-09:
amixer is a utility for Alsa http://www.alsa-project.org/main/index.php/Main_Page , the Linux kernel sound driver, and is used to manipulate the audio mixer, whereas omxplayer is software for playing multimedia files, audio and video, h264 encoded mp4 files and streams etc. I'd be very surprised if Openelec doesn't have amixer if it has aplay because they are both included in the alsa-utils package.
What your output of ```aplay -l``` shows is that your Raspberry Pi's sound-processor driver may not be loaded. Could you confirm this by running ```sudo aplay -l``` which will run the command as root (running as root is required for some distributions and not others due to the logged in user not having privileges to access the sound device)?

What does the output of ```lsmod``` show you? It should list the currently running kernel drivers and the one you are looking for is snd_bcm2835. If snd_bcm2835 isn't listed, run ```sudo modprobe snd_bcm2835``` to load the sound driver into the kernel, then confirm the sound processor has been detected by running ```sudo aplay -l``` again.

The problem might be that the user you are logged in as does not have enough privileges to access the sound device (which might explain why ```aplay -l``` would report no soundcards found). Try running limelight using sudo or as root:
```sudo java -jar limelight.jar stream -input /dev/input/event0 -720 PC-hostname/IPaddress```
where PC-hostname/IPaddress is the hostname or IP address of your gaming PC.

You might still need to use ```amixer cset numid=3 2``` to force audio through HDMI rather than the 3.5mm jack.

Here is the output from the above commands on my Raspberry Pi:
```
$ sudo aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
$ lsmod
Module                  Size  Used by
btusb                  20210  0 
bluetooth             277038  2 btusb
uio_pdrv_genirq         3281  0 
uio                     8710  1 uio_pdrv_genirq
sch_fq_codel            7252  2 
snd_bcm2835            20385  0 
snd_pcm                84615  1 snd_bcm2835
snd_timer              19608  1 snd_pcm
snd                    60092  3 snd_bcm2835,snd_timer,snd_pcm
bcm2708_rng             1143  0 
rng_core                5666  1 bcm2708_rng
ipv6                  343408  24
```

Krazypoloc, 2015-06-10:
Loading the sound module into the kernel resolved my issue. Thanks a lot!

modprobe snd_bcm2835

7hunderbug, 2015-06-10:
I'm glad to hear you got it sorted! I would go ahead and close this 
issue now that it is fixed.

closed 2015-06-10 by irtimmer