moonlight-stream/moonlight-embedded #132: Gamepad not mapping correctly with moonlight-c

schulzad, 2015-06-12:
I pulled and build the latest moonlight and I've notice my Gamepad no longer maps correctly.

Only left and right d_pad work and all other button are unresponsive. (I'm using a wireless xbox 360 controller)

I use xboxdrv with a config file that works just find with the java implementation. I've never had to muck with the moonlight mapping config file before.

RikJm, 2015-06-13:
Same here! I've been stuck with this too :(

scoobyd00, 2015-06-13:
Use the mapping option and create your own mapping file and then use this when running moonlight to see if it fixes the issue

7hunderbug, 2015-06-13:
@scoobyd00 is correct; a mapping file wasn't required for the java version but it is a requirement of the new C version.

RikJm, 2015-06-13:
@7hunderbug Thanks a lot for your answer. I've seen you've also replied to issue 104.

I had a look at my commands and I've found the issue, I'll post it here since @schulzad  seems to be having the same problem.

To create my map file I used: 
>moonlight map xbox.map  

I compared this to the default.conf file on usr/local/share/mappings and it was quite different. I replaced that file with my new contents and also copied the xbox.map file to my moonlight folder /usr/local/bin/xbox.map <-- This is the one I was calling using the -mapping flag

To launch moonlight in emulation station I was using an .sh file that contained: 

#!/bin/bash
cd /usr/local/bin/ && sudo ./moonlight stream -720 -30fps 192.168.0.3 -app Steam -input /dev/input/event0 -mapping xbox.map

This is where the problem was!!! I had a look at what were the streaming options in moonlight and it seems that -mapping needs to be used before -input. 

#!/bin/bash
cd /usr/local/bin/ && sudo ./moonlight stream -720 -30fps 192.168.0.3 -app Steam -mapping xbox.map -input /dev/input/event0 

Once I changed it everything worked like a charm!!

Thanks a lot for your help :)

schulzad, 2015-06-14:
@RikJm Do you still use the xboxdrv config you used before?

Also any chance you have post your xbox.map for posterity? 

Thanks for all the help on this!

sudozulu, 2015-06-14:
I'm still having the same issue, even after trying out Rikjm's suggestion. If I use the java version, everything works without issue, but if I use the exact same settings with 2.0, only the left analog stick works.

If I run xboxdrv, I can see all the buttons recognized.

sudozulu, 2015-06-14:
schulzad: wget https://raw.githubusercontent.com/RoelofBerg/limelightpisteambox/master/xbox.map

Here's the one I used with the Java version and it works well.

7hunderbug, 2015-06-14:
@schulzad @sudozulu If it's any help, here's my Xbox 360 wireless controller mapping for moonlight. This mapping file works when xboxdrv is run without any switches, i.e. so the default settings where both trigger-button and d-pad act as analogue inputs:
```
abs_x = 0
abs_y = 1
abs_z = 10
reverse_x = false
reverse_y = true
abs_rx = 3
abs_ry = 4
abs_rz = 9
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = 16
abs_dpad_y = 17
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = -1
btn_tr2 = -1
btn_dpad_up = -1
btn_dpad_down = -1
btn_dpad_left = -1
btn_dpad_right = -1
```
If you run xboxdrv with the --trigger-as-button and --dpad-as-button switches then your mapping file will probably look like this (from @scoobyd00's comment in https://github.com/irtimmer/moonlight-embedded/issues/116):
```
abs_x = 0
abs_y = 1
abs_z = -1
reverse_x = false
reverse_y = true
abs_rx = 3
abs_ry = 4
abs_rz = -1
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = -1
abs_dpad_y = -1
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = 312
btn_tr2 = 313
btn_dpad_up = 294
btn_dpad_down = 295
btn_dpad_left = 296
btn_dpad_right = 297
```
@schulzad are you specifying the input for moonlight? I've found that you have to specify the gamepad input device file as a switch to moonlight otherwise the Xbox controller won't work. As @RikJm has discovered, the order in which you put the arguments to moonlight matters, so mapping file switch then input device switch (I always put the PC IP address at the end of all arguments). My command line is as follows:
```./moonlight stream -mapping xbox360wirelessPC.map -input /dev/input/event0 192.168.1.100```

RikJm, 2015-06-14:
@schulzad  I have been trying different xboxdrv configurations, the 1st method here: https://github.com/petrockblog/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller seems to work without issues.

Right now I'm trying to be more especific on how I configure my xbox controller using a config file when I launch xboxdrv, adding this at the end of each line:

>xboxdrv --trigger-as-button --wid 0 --led 2 --deadzone 4000 --silent -c /home/YOUR_USERNAME/default.xboxdrv &
sleep 1

I took the config file from https://github.com/xboxdrv/xboxdrv/blob/stable/examples/default.xboxdrv

Everything seems to work fine, I'm now going to atttempt to setup multiple controllers :)

sudozulu, 2015-06-15:
Here's how I run moonlight (as part of a larger script): sudo moonlight -mapping mapfile.map -input /dev/input/event0 -1080 -30fps stream 192.168.1.xxx -app Steam

Even putting it at the beginning of the program arguments, I still end up with the same result (I also copied your map file. No joy): The left analog stick will let me move right and left and that's it. No buttons, bumpers, or other analog stick is registered. SOOOOO close.

jamaalsawyerd, 2015-06-15:
I'm having the exact same problem as sudozulu but with a paired PS3 controller on my RetroPie.  I mapped it and can only perform Left/Right with the Left analog stick and no buttons.

7hunderbug, 2015-06-15:
Everyone who has controller problems should run ```evtest`` and post the output here to assist in troubleshooting. Please also post the contents of your moonlight mapping file so it can be compared with the values displayed by evtest.

schulzad, 2015-06-16:
>Everyone who has controller problems should run evtest and post the output here to assist in troubleshooting.

Running `evtest` just explained everything to me. My keyboard was getting two event devices "event0" and "event1". My gamepad was actually on "event2"

I had not been using and `-input` switch in version 1 which I assumes means it just used /dev/input/js0.

Anyway once I: 
1. added -input /dev/input/event2
2. create a map file via `moonlight map xbox.map`
3. made sure I placed the `-mapping` switch before the `-input` switch as @RikJm suggested

It works!

sudozulu, 2015-06-16:
Hrmm, it seems that OSMC doesn't have fuser as a built-in command. apt-get doesn't have it. What could I install to get the same result?

sudozulu, 2015-06-16:
evtest on /dev/input/event0 and event1 yield the same result (but I only have one controller plugged in, which is weird...)

Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "Xbox Gamepad (userspace driver)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 294 (BTN_BASE)
    Event code 295 (BTN_BASE2)
    Event code 296 (BTN_BASE3)
    Event code 297 (BTN_BASE4)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min   -32768
      Max    32767
    Event code 1 (ABS_Y)
      Value      0
      Min   -32768
      Max    32767
    Event code 3 (ABS_RX)
      Value      0
      Min   -32768
      Max    32767
    Event code 4 (ABS_RY)
      Value      0
      Min   -32768
      Max    32767
Properties:
Testing ... (interrupt to exit)
***********************************************
  This device is grabbed by another process.
  No events are available to evtest while the
  other grab is active.
  In most cases, this is caused by an X driver,
  try VT-switching and re-run evtest again.
  Run the following command to see processes with
  an open fd on this device
 "fuser -v /dev/input/event0"
***********************************************

7hunderbug, 2015-06-17:
@sudozulu 
Try stopping kodi and running evtest again. Also, make sure of the following:

1. No other controllers are plugged in to your Raspberry Pi, except for the one you want to use with moonlight. This means mouse and keyboard too, at least until you get your controller working.

2. No other controllers are connected to your gaming PC.

3. You don't have ```xpad``` kernel module loaded on the Raspberry Pi - check with ```lsmod``` and if ```xpad``` is listed, remove it with ```sudo modprobe -r xpad```.

4. You don't have a desktop session running - stop ```X``` or ```Xorg``` if so.

5. You don't have any other software running that might interfere - e.g. stop "kodi" from running.

6. Make sure only one instance of your game controller driver is running. If you use ```xboxdrv```, first stop all instances of it with ```sudo killall xboxdrv``` or similar. Then run ```sudo xboxdrv --silent &``` from the command line on your Raspberry Pi. Then run moonlight to make a new mapping file and then run moonlight a second time to test game streaming.

jamaalsawyerd, 2015-06-18:
I apologize in advance for the verbose post:

Running evtest with nothing pulled in except PS3 controller:
Input driver version is 1.0.1
Input device ID: bus 0x6 vendor 0x54c product 0x268 version 0x100
Input device name: "PLAYSTATION(R)3 Controller (38:C0:96:8B:F0:AA)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 288 (BTN_TRIGGER)
    Event code 289 (BTN_THUMB)
    Event code 290 (BTN_THUMB2)
    Event code 291 (BTN_TOP)
    Event code 292 (BTN_TOP2)
    Event code 293 (BTN_PINKIE)
    Event code 294 (BTN_BASE)
    Event code 295 (BTN_BASE2)
    Event code 296 (BTN_BASE3)
    Event code 297 (BTN_BASE4)
    Event code 298 (BTN_BASE5)
    Event code 299 (BTN_BASE6)
    Event code 300 (?)
    Event code 301 (?)
    Event code 302 (?)
    Event code 303 (BTN_DEAD)
    Event code 304 (BTN_A)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min     -127
      Max      127
    Event code 1 (ABS_Y)
      Value      0
      Min     -127
      Max      127
    Event code 2 (ABS_Z)
      Value      0
      Min     -127
      Max      127
    Event code 3 (ABS_RX)
      Value      0
      Min     -127
      Max      127
    Event code 4 (ABS_RY)
      Value      0
      Min     -622
      Max     -402
    Event code 5 (ABS_RZ)
      Value      0
      Min      402
      Max      622
    Event code 6 (ABS_THROTTLE)
      Value      0
      Min      402
      Max      622
    Event code 7 (ABS_RUDDER)
      Value      0
      Min     -127
      Max      127
    Event code 8 (ABS_WHEEL)
      Value      0
      Min     -255
      Max      255
    Event code 9 (ABS_GAS)
      Value      0
      Min     -255
      Max      255
    Event code 10 (ABS_BRAKE)
      Value      0
      Min     -255
      Max      255
    Event code 11 (?)
      Value      0
      Min     -255
      Max      255
    Event code 12 (?)
      Value      0
      Min     -255
      Max      255
    Event code 13 (?)
      Value      0
      Min     -255
      Max      255
    Event code 14 (?)
      Value      0
      Min     -255
      Max      255
    Event code 15 (?)
      Value      0
      Min     -255
      Max      255
    Event code 26 (ABS_TILT_X)
      Value      0
      Min     -255
      Max      255
    Event code 27 (ABS_TILT_Y)
      Value      0
      Min     -255
      Max      255
    Event code 28 (ABS_TOOL_WIDTH)
      Value      0
      Min     -255
      Max      255
    Event code 29 (?)
      Value      0
      Min     -255
      Max      255
    Event code 30 (?)
      Value      0
      Min    -1250
      Max     1250
    Event code 31 (?)
      Value      0
      Min    -1250
      Max     1250
    Event code 32 (ABS_VOLUME)
      Value      0
      Min    -1250
      Max     1250
    Event code 33 (?)
      Value      0
      Min    -1250
      Max     1250
    Event code 34 (?)
      Value      0
      Min    -1250
      Max     1250
    Event code 35 (?)
      Value      0
      Min    -1250
      Max     1250
    Event code 36 (?)
      Value      0
      Min    -1250
      Max     1250
    Event code 37 (?)
      Value      0
      Min    -1250
      Max     1250
    Event code 38 (?)
      Value      0
      Min    -1250
      Max     1250
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)

The moonlight generated .map file (which is actually identical to the one that I used previously with the java version):

abs_x = 0
abs_y = 1
abs_z = -1
reverse_x = false
reverse_y = true
abs_rx = 2
abs_ry = 3
abs_rz = -1
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = -1
abs_dpad_y = -1
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 300
btn_east = 301
btn_south = 302
btn_west = 303
btn_select = 288
btn_start = 291
btn_mode = 304
btn_thumbl = 289
btn_thumbr = 290
btn_tl = 296
btn_tr = 297
btn_tl2 = 298
btn_tr2 = 299
btn_dpad_up = 292
btn_dpad_down = 294
btn_dpad_left = 295
btn_dpad_right = 293

7hunderbug, 2015-06-21:
@jamaalsawyerd which driver are you using for your PS3 controller? If it is ds4drv https://github.com/chrippa/ds4drv then you might want to try xbox controller emulation.

maverickjesus, 2015-06-21:
I have the same issue - evtest output:

evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      Xbox Gamepad (userspace driver)
Select the device event number [0-0]: 0
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "Xbox Gamepad (userspace driver)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 294 (BTN_BASE)
    Event code 295 (BTN_BASE2)
    Event code 296 (BTN_BASE3)
    Event code 297 (BTN_BASE4)
    Event code 304 (BTN_A)
    Event code 305 (BTN_B)
    Event code 307 (BTN_X)
    Event code 308 (BTN_Y)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min   -32768
      Max    32767
    Event code 1 (ABS_Y)
      Value      0
      Min   -32768
      Max    32767
    Event code 3 (ABS_RX)
      Value      0
      Min   -32768
      Max    32767
    Event code 4 (ABS_RY)
      Value      0
      Min   -32768
      Max    32767

Mapping:

abs_x = 0
abs_y = 1
abs_z = -1
reverse_x = false
reverse_y = true
abs_rx = 3
abs_ry = 4
abs_rz = -1
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = -1
abs_dpad_y = -1
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = 312
btn_tr2 = 313
btn_dpad_up = 294
btn_dpad_down = 295
btn_dpad_left = 296
btn_dpad_right = 297

Not running X, xpad is not loaded, only one XboxDRV process running, basically all of the suggestions so far have come back clear - but I still only have left/right on the left analog stick. Tried as many tweaks as I can think of and no dice.

7hunderbug, 2015-06-21:
@maverickjesus what OS are you using? Did you compile moonlight yourself or did you download a pre-compiled release from somewhere? What does the output of ```ldd moonlight``` show (you'll need to run this in the same directory that the moonlight executable is located)?

maverickjesus, 2015-06-22:
@7hunderbug Raspian on the Pi2, Windows 8.1 w/ GTX670 on the host. Moonlight is loaded from the apt repo.

LDD output:

pi@raspberrypi /usr/bin $ ldd moonlight
        /usr/lib/arm-linux-gnueabihf/libcofi_rpi.so (0x76f3b000)
        libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76f0e000)
        libopenmaxil.so => /opt/vc/lib/libopenmaxil.so (0x76f00000)
        libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76ef2000)
        libvcos.so => /opt/vc/lib/libvcos.so (0x76ee0000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ec1000)
        libcurl.so.4 => /usr/lib/arm-linux-gnueabihf/libcurl.so.4 (0x76e5f000)
        libssl.so.1.0.0 => /usr/lib/arm-linux-gnueabihf/libssl.so.1.0.0 (0x76e0e000)
        libcrypto.so.1.0.0 => /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0 (0x76caa000)
        libexpat.so.1 => /lib/arm-linux-gnueabihf/libexpat.so.1 (0x76c81000)
        libevdev.so.2 => /usr/lib/arm-linux-gnueabihf/libevdev.so.2 (0x76c61000)
        libasound.so.2 => /usr/lib/arm-linux-gnueabihf/libasound.so.2 (0x76b9d000)
        libopus.so.0 => /usr/lib/libopus.so.0 (0x76b58000)
        libavahi-common.so.3 => /usr/lib/arm-linux-gnueabihf/libavahi-common.so.3 (0x76b44000)
        libavahi-client.so.3 => /usr/lib/arm-linux-gnueabihf/libavahi-client.so.3 (0x76b2d000)
        libudev.so.0 => /lib/arm-linux-gnueabihf/libudev.so.0 (0x76b18000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76b0d000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x769dc000)
        /lib/ld-linux-armhf.so.3 (0x76f48000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x769cd000)
        libGLESv2.so => /opt/vc/lib/libGLESv2.so (0x769b0000)
        libEGL.so => /opt/vc/lib/libEGL.so (0x76987000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76916000)
        libidn.so.11 => /usr/lib/arm-linux-gnueabihf/libidn.so.11 (0x768dc000)
        libssh2.so.1 => /usr/lib/arm-linux-gnueabihf/libssh2.so.1 (0x768ba000)
        liblber-2.4.so.2 => /usr/lib/arm-linux-gnueabihf/liblber-2.4.so.2 (0x768a6000)
        libldap_r-2.4.so.2 => /usr/lib/arm-linux-gnueabihf/libldap_r-2.4.so.2 (0x7685c000)
        libgssapi_krb5.so.2 => /usr/lib/arm-linux-gnueabihf/libgssapi_krb5.so.2 (0x76824000)
        librtmp.so.0 => /usr/lib/arm-linux-gnueabihf/librtmp.so.0 (0x76805000)
        libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x767e7000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x767bf000)
        libdbus-1.so.3 => /lib/arm-linux-gnueabihf/libdbus-1.so.3 (0x7677d000)
        libgcrypt.so.11 => /lib/arm-linux-gnueabihf/libgcrypt.so.11 (0x766fc000)
        libresolv.so.2 => /lib/arm-linux-gnueabihf/libresolv.so.2 (0x766e8000)
        libsasl2.so.2 => /usr/lib/arm-linux-gnueabihf/libsasl2.so.2 (0x766c9000)
        libgnutls.so.26 => /usr/lib/arm-linux-gnueabihf/libgnutls.so.26 (0x7660e000)
        libkrb5.so.3 => /usr/lib/arm-linux-gnueabihf/libkrb5.so.3 (0x76560000)
        libk5crypto.so.3 => /usr/lib/arm-linux-gnueabihf/libk5crypto.so.3 (0x7652f000)
        libcom_err.so.2 => /lib/arm-linux-gnueabihf/libcom_err.so.2 (0x76524000)
        libkrb5support.so.0 => /usr/lib/arm-linux-gnueabihf/libkrb5support.so.0 (0x76515000)
        libkeyutils.so.1 => /lib/arm-linux-gnueabihf/libkeyutils.so.1 (0x7650a000)
        libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0x764ff000)
        libtasn1.so.3 => /usr/lib/arm-linux-gnueabihf/libtasn1.so.3 (0x764e8000)
        libp11-kit.so.0 => /usr/lib/arm-linux-gnueabihf/libp11-kit.so.0 (0x764d1000)

7hunderbug, 2015-06-22:
@maverickjesus are you running moonlight with something like this: ```moonlight stream 192.168.1.23 -mapping xbox.map -input /dev/input/event0```? You'll probably need the ```-input``` switch and the ```-mapping``` switch for the controller to be picked up properly. /dev/input/event# should be the number of your controller and the mapping switch needs to come before the input switch.

maverickjesus, 2015-06-23:
I've already tried reordering input & mapping and it makes no difference, and I know its the right input device as I've debugged it with evtest - it recognises all of the inputs correctly.

Coincidentally I upgraded to the Windows 10 preview today and the issue still persists.

jamaalsawyerd, 2015-06-23:
@7hunderbug I'm using sixpair and the sixad driver.

mallocint, 2015-06-27:
my xbox360 controller buttons are not recognized, i can navigate in steam big picture mode using the left stick, but i cant for example select the shop by pressing the A button.

My system specs and configs:

   
host pc is running windows 8.1 
            gtx 750ti driver version 353.30
            gfe  version 2.4.5.57

client raspberry pi2
            Arch Linux ARM 4.0.6-2-ARCH #1 SMP PREEMPT Fri Jun 26 23:05:30 MDT 2015 armv7l GNU/Linux

 ldd /usr/local/bin/moonlight 
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76f41000)
	libopenmaxil.so => /opt/vc/lib/libopenmaxil.so (0x76f91000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76f33000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76f21000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x76ef8000)
	libcurl.so.4 => /usr/lib/libcurl.so.4 (0x76e8a000)
	libssl.so.1.0.0 => /usr/lib/libssl.so.1.0.0 (0x76e1e000)
	libcrypto.so.1.0.0 => /usr/lib/libcrypto.so.1.0.0 (0x76c91000)
	libexpat.so.1 => /usr/lib/libexpat.so.1 (0x76c68000)
	libevdev.so.2 => /usr/lib/libevdev.so.2 (0x76c47000)
	libasound.so.2 => /usr/lib/libasound.so.2 (0x76b83000)
	libopus.so.0 => /usr/lib/libopus.so.0 (0x76b3a000)
	libavahi-common.so.3 => /usr/lib/libavahi-common.so.3 (0x76b1f000)
	libavahi-client.so.3 => /usr/lib/libavahi-client.so.3 (0x76b00000)
	libdbus-1.so.3 => /usr/lib/libdbus-1.so.3 (0x76ab6000)
	libudev.so.1 => /usr/lib/libudev.so.1 (0x76950000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x7693b000)
	libc.so.6 => /usr/lib/libc.so.6 (0x767fc000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x767cf000)
	librt.so.1 => /usr/lib/librt.so.1 (0x767b8000)
	libGLESv2.so => /opt/vc/lib/libGLESv2.so (0x7679a000)
	libEGL.so => /opt/vc/lib/libEGL.so (0x76771000)
	libm.so.6 => /usr/lib/libm.so.6 (0x766f3000)
	/lib/ld-linux-armhf.so.3 (0x76f71000)
	libidn.so.11 => /usr/lib/libidn.so.11 (0x766b2000)
	libssh2.so.1 => /usr/lib/libssh2.so.1 (0x76680000)
	libgssapi_krb5.so.2 => /usr/lib/libgssapi_krb5.so.2 (0x76632000)
	libkrb5.so.3 => /usr/lib/libkrb5.so.3 (0x7657a000)
	libk5crypto.so.3 => /usr/lib/libk5crypto.so.3 (0x7653c000)
	libcom_err.so.2 => /usr/lib/libcom_err.so.2 (0x76529000)
	libz.so.1 => /usr/lib/libz.so.1 (0x76506000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x764de000)
	libcap.so.2 => /usr/lib/libcap.so.2 (0x764d3000)
	libkrb5support.so.0 => /usr/lib/libkrb5support.so.0 (0x764b8000)
	libkeyutils.so.1 => /usr/lib/libkeyutils.so.1 (0x764ad000)
	libattr.so.1 => /usr/lib/libattr.so.1 (0x764a0000)

/usr/lib/systemd/system/xboxdrv.service 
[Unit]
Description = Gamepad management daemon

[Service]
User = root
Type = simple
ExecStart = /usr/bin/xboxdrv -D -c /etc/default/xboxdrv --detach-kernel-driver

[Install]
WantedBy = multi-user.target

xbox.map 

abs_x = 0
abs_y = 1
abs_z = 10
reverse_x = false
reverse_y = true
abs_rx = 3
abs_ry = 4
abs_rz = 9
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = 16
abs_dpad_y = 16
reverse_dpad_x = false
reverse_dpad_y = true
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = -1
btn_tr2 = -1
btn_dpad_up = -1
btn_dpad_down = -1
btn_dpad_left = -1
btn_dpad_right = -1

evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Xbox Gamepad (userspace driver)

moonlight -mapping /root/xbox.map -input /dev/input/event0 stream hostname

mallocint, 2015-06-27:
Found out that if i pass just the mapping file without the relative path controller works in big picture mode.

im launching moonlight like this:
moonlight -mapping xbox.map -input /dev/input/event0 -app Steam stream goliath

maverickjesus, 2015-06-27:
Weird - I just tried removing my absolute path on mapping and it still didn't work, but I did notice my args were all after the 'stream' command. I moved that all the way to the end and now it works?

Is there just some weird/undocumented logic for the cmd line arguments?

Now working version: moonlight -720 -mapping xbox.map -input /dev/input/event0 -app Steam stream

cgutman, 2015-06-27:
I've got a branch that should fix the issues I've seen in this bug.

git clone https://github.com/cgutman/moonlight-embedded.git
git checkout mappingfixes

Fixes:
- Warn if your mapping file couldn't be loaded
- Print out unmapped keys
- Fix absolute paths to mapping files
- Include a default mapping if none could be loaded

mallocint, 2015-06-28:
tried cgutman's fork its fixed, thanks

maverickjesus, 2015-06-28:
Yep, same here. Great job.

closed 2015-06-28 by irtimmer