moonlight-stream/moonlight-embedded #134: moonlight compiled, but no video on TV

pivichkrsk, 2015-06-13:
Moonlight is started, Stream in Big Picture mode also started, i can select menu in Steam via controller BUT there is no video on my Raspberry Pi.
Lastest Java Limelight works fine.
I've reinstalled all libs and make/gcc+/cmake utils twice, but no still luck. 

$cmake --version
cmake version 3.2.3

$git clone https://github.com/irtimmer/moonlight-embedded
$cd moonlight-embedded
$mkdir build
$cd build
$cmake ../
cmake ../
-- The C compiler identification is GNU 4.7.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.38.0")
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.0.1k")
-- Found EXPAT: /usr/lib/arm-linux-gnueabihf/libexpat.so (found version "2.1.0")
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.0.28")
-- Found Opus: /usr/lib/arm-linux-gnueabihf/libopus.so
-- Could NOT find Broadcom (missing:  BROADCOM_INCLUDE_DIR)
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
-- checking for module 'libevdev'
--   found libevdev, version 1.3
-- checking for module 'avahi-client'
--   found avahi-client, version 0.6.31
-- checking for module 'libudev'
--   found libudev, version 215
-- Configuring done
-- Generating done
-- Build files have been written to: /home/osmc/2/moonlight-embedded/build

$make
make
Scanning dependencies of target moonlight
[  3%] Building C object CMakeFiles/moonlight.dir/src/mapping.c.o
[  6%] Building C object CMakeFiles/moonlight.dir/src/global.c.o
[  9%] Building C object CMakeFiles/moonlight.dir/src/mkcert.c.o
[ 12%] Building C object CMakeFiles/moonlight.dir/src/main.c.o
[ 16%] Building C object CMakeFiles/moonlight.dir/src/input.c.o
[ 19%] Building C object CMakeFiles/moonlight.dir/src/xml.c.o
[ 22%] Building C object CMakeFiles/moonlight.dir/src/discover.c.o
[ 25%] Building C object CMakeFiles/moonlight.dir/src/client.c.o
[ 29%] Building C object CMakeFiles/moonlight.dir/src/http.c.o
[ 32%] Building C object CMakeFiles/moonlight.dir/src/video.c.o
[ 35%] Building C object CMakeFiles/moonlight.dir/src/audio.c.o
[ 38%] Building C object CMakeFiles/moonlight.dir/src/connection.c.o
[ 41%] Building C object CMakeFiles/moonlight.dir/src/video/fake.c.o
[ 45%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/ControlStream.c.o
[ 48%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/RtspConnection.c.o
[ 51%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/PlatformSockets.c.o
[ 54%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/RtpReorderQueue.c.o
[ 58%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/VideoDepacketizer.c.o
[ 61%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/InputStream.c.o
[ 64%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/Connection.c.o
[ 67%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/FakeCallbacks.c.o
[ 70%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/RtspParser.c.o
[ 74%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/AudioStream.c.o
[ 77%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/SdpGenerator.c.o
[ 80%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/Platform.c.o
[ 83%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/ByteBuffer.c.o
[ 87%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/VideoStream.c.o
[ 90%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/LinkedBlockingQueue.c.o
[ 93%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/Misc.c.o
[ 96%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/OpenAES/oaes_lib.c.o
[100%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/OpenAES/oaes_base64.c.o
Linking C executable moonlight
[100%] Built target moonlight

$make install
 make install
[100%] Built target moonlight
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/bin/moonlight
-- Up-to-date: /usr/local/share/mappings
-- Up-to-date: /usr/local/share/mappings/default.conf

$moonlight pair 192.168.1.5

$moonlight pair 192.168.1.5
Generating certificate...done
Please enter the following PIN on the target PC: xxxx
Paired

$moonlight -720 -60fps -mapping /opt/limelight/PS2.map stream 192.168.1.5
Initializing platform...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Loss stats thread terminating #1
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done

$uname -a
uname -a
Linux osmc 3.18.13-1-osmc #1 SMP PREEMPT Sun May 17 18:50:26 UTC 2015 armv7l GNU/Linux

$cat /boot/config.txt
gpu_mem_1024=512
disable_overscan=1
over_voltage=0
force_turbo=0
over_voltage_sdram=0
core_freq=450
initial_turbo=0
start_x=1
hdmi_drive=2
gpu_mem_512=256
hdmi_force_hotplug=1
arm_freq=900
sdram_freq=450
hdmi_force_edid_audio=1
overscan_scale=1
gpu_mem_256=128
hdmi_ignore_cec_init=1
sdtv_aspect=1

raspberry pi2, with 2 connected PS3 controllers (they working fine in retropie / limelight java), TV connected via HDMI.

cgutman, 2015-06-13:
"Could NOT find Broadcom (missing: BROADCOM_INCLUDE_DIR)"

Looks like it didn't find the video decoder headers so it didn't build in support for the Pi video decoder.

I wonder if @irtimmer could provide binaries for at least Raspbian-based distros. I suspect build issues will crop up often (missing includes or user has no idea how to compile this).

7hunderbug, 2015-06-13:
Does OSMC have the raspberrypi userland libraries and packages installed? i.e the ```/opt/vc/include``` directory from the libraspberrypi-dev package?

sudozulu, 2015-06-14:
I was able to get video output by compiling the userland libraries from here: https://github.com/raspberrypi/userland

Once i compiled them, I rebuilt the moonlight binary and got video on my TV. Wired 360 controller input is the only thing not working for me now.

pivichkrsk, 2015-06-14:
Wow, thanks sudozulu advices worked !

closed 2015-06-21 by irtimmer