moonlight-stream/moonlight-embedded #135: Installing on Raspberry Pi

Breadcrust, 2015-06-15:
Hi, no matter how many guides I have read. No matter how many attempts I have not gotten Limelight Jar to run with proper controller support working (All the mappings are wrong even though I have tried literally everything even re-mapping and installing different controller drivers with different controllers) Nor have I got Moonlight-C to install! Can someone please explain all this cmake compiling thing to me. I can't even install past V2.8.3 of cmake on my Raspberry Pi. Can someone please give me a guide on how to get this to run.

sudozulu, 2015-06-15:
Same issue as here: https://github.com/irtimmer/moonlight-embedded/issues/132

Breadcrust, 2015-06-15:
I want to install the C version but I don't know how! The guide makes no sense to me. I don't understand this whole cmake thing.

Breadcrust, 2015-06-15:
Limelight Jar no longer pairs and so I want to try C

SkewPL, 2015-06-15:
@unchartedxx send instructions how to compile new version of cmake here: #104 , that worked for me and i compiled moonlight without problems.

Download, compile and install cmake 3.2:

sudo apt-get install build-essential
wget http://www.cmake.org/files/v3.2/cmake-3.2.3.tar.gz
tar xzf cmake-3.2.3.tar.gz
cd cmake-3.2.3
./configure
sudo make install

Compile and install limelight:

cd moonlight-embedded-2.0
mkdir build
cd build/
cmake -D CMAKE_BUILD_TYPE=Release ../
make
sudo make install

dansl, 2015-06-19:
@SkewPL worked for me, thanks for posting! The only other thing I had to do was get the libevdev packages installed. Basically did this from #104 :

sudo apt-get install libjs-jquery
wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb
wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb
sudo dpkg -i libevdev2_1.4.2+dfsg-1_armhf.deb libevdev-dev_1.4.2+dfsg-1_armhf.deb

melzk, 2015-06-20:
i tried to compile the newest version but i got this error message, i followed SKewPL's step by step

pi@raspberrypi ~/moonlight-embedded-2.0/build $ sudo cmake -D CMAKE_BUILD_TYPE=Release ../
CMake Error at /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find CURL (missing: CURL_LIBRARY CURL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.2/Modules/FindCURL.cmake:61 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:17 (find_package)

Rassell, 2015-06-20:
I have the same problem as @melzk , tried everything to change the issue with the curl libraries. Any tips with this issue?

Serg86, 2015-06-20:
Could anyone please provide the already compiled binary for Moonlight 2.0.1? I am on a Raspberry Pi 2 with OpenElec 5.0.8. OpenElec doesn't have any dev stuff on it, there is not even a package manager. I am able to run Moonlight but not compile it.

H3llsing, 2015-06-20:
im not an expert, but i solved it with these libs:

# this is the missing curl lib  found it with ->  apt-cache search "curl"
sudo apt-get install libcurl4-openssl-dev 

# from another thread i found that these dependency libs should be installed... 
sudo apt-get install libopus-dev libexpat1-dev libssl-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libjs-jquery

wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb

wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-tools_1.4.2%2bdfsg-1_armhf.deb

wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb

sudo dpkg -i libevdev*

# now try again!

melzk, 2015-06-20:
no luck :( io got the same message
 cmake -D CMAKE_BUILD_TYPE=Release ../
CMake Error at /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find CURL (missing: CURL_LIBRARY CURL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.2/Modules/FindCURL.cmake:61 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/moonlight-embedded-2.0/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/moonlight-embedded-2.0/build/CMakeFiles/CMakeError.log".
CMake Error: Unable to open check cache file for write. /home/pi/moonlight-embedded-2.0/build/CMakeFiles/cmake.check_cache

if it to any help i using rasberry pi 2 and cmake 3.2.3

Rassell, 2015-06-20:
@H3llsing thanks but already tested it and nothing change...

H3llsing, 2015-06-20:
Ok i got another error:
is this correct, that moonlight tries to include a limelight-common/Limelight.h ?

[  5%] Building C object CMakeFiles/moonlight.dir/src/connection.c.o
In file included from /home/pi/moonlight-embedded/src/connection.c:20:0:
/home/pi/moonlight-embedded/src/connection.h:20:40: fatal error: limelight-common/Limelight.h: No such file or directory
compilation terminated.
CMakeFiles/moonlight.dir/build.make:146: recipe for target 'CMakeFiles/moonlight.dir/src/connection.c.o' failed
make[2]: *** [CMakeFiles/moonlight.dir/src/connection.c.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/moonlight.dir/all' failed
make[1]: *** [CMakeFiles/moonlight.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

melzk, 2015-06-20:
Uppgrade of a fresh new installed raspian
 
	sudo apt-get dist-upgrade
	sudo apt-get update
	sudo apt-get upgrade	

download and installed all recomended libs in this trehead

	sudo apt-get install libjs-jquery
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb
	sudo dpkg -i libevdev2_1.4.2+dfsg-1_armhf.deb libevdev-dev_1.4.2+dfsg-1_armhf.deb
	sudo apt-get install libcurl4-openssl-dev
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-tools_1.4.2%2bdfsg-1_armhf.deb
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb
	sudo dpkg -i libevdev*

Download och unpack moonlight

	wget https://github.com/irtimmer/moonlight-embedded/releases/download/v2.0.1/moonlight-embedded-2.0.1.tar.xz
	tar xf moonlight-embedded-2.0.1.tar.xz

download and install cmake 3.2.3
	
	sudo apt-get install build-essential
	wget http://www.cmake.org/files/v3.2/cmake-3.2.3.tar.gz
	tar xzf cmake-3.2.3.tar.gz
	cd cmake-3.2.3
	./configure
	sudo make install

now its time to get into the moonlight folder
	cd ..
	cd moonlight-embedded-2.0.1
	mkdir build
	cd build/
	pi@raspberrypi ~/moonlight-embedded-2.0.1/build $ cmake -D CMAKE_BUILD_TYPE=Release ../
-- The C compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.26.0") 
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.0.1e") 
CMake Error at /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find EXPAT (missing: EXPAT_LIBRARY EXPAT_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.2/Modules/FindEXPAT.cmake:56 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:19 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/moonlight-embedded-2.0.1/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/moonlight-embedded-2.0.1/build/CMakeFiles/CMakeError.log".

and as usal error........... gah it drives me crazy here

melzk, 2015-06-20:
also tried to compile the moonlight-embedded-master.zip  and got this error:
pi@raspberrypi ~/moonlight-embedded-master/build $ cmake -D CMAKE_BUILD_TYPE=Release ../
-- The C compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.26.0") 
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.0.1e") 
CMake Error at /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find EXPAT (missing: EXPAT_LIBRARY EXPAT_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.2/Modules/FindEXPAT.cmake:56 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:19 (find_package)
 

-- Configuring incomplete, errors occurred!
See also "/home/pi/moonlight-embedded-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/moonlight-embedded-master/build/CMakeFiles/CMakeError.log".

H3llsing, 2015-06-20:
@melzk u are missing the Expat library

sudo apt-get update && sudo apt-get install libexpat1-dev

Rassell, 2015-06-20:
still the same issue:

CMake Error at /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find CURL (missing: CURL_LIBRARY CURL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.2/Modules/FindCURL.cmake:61 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/moonlight-embedded-2.0.1/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/moonlight-embedded-2.0.1/build/CMakeFiles/CMakeError.log".

melzk, 2015-06-20:
i installed the libexpat1-dev and got this when i tried to compile:

pi@raspberrypi ~/moonlight-embedded-master/build $ cmake -D CMAKE_BUILD_TYPE=Release ../
-- The C compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.26.0") 
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.0.1e") 
-- Found EXPAT: /usr/lib/arm-linux-gnueabihf/libexpat.so (found version "2.1.0") 
CMake Error at /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find ALSA (missing: ALSA_LIBRARY ALSA_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.2/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.2/Modules/FindALSA.cmake:56 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:20 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/moonlight-embedded-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/moonlight-embedded-master/build/CMakeFiles/CMakeError.log".

scoobyd00, 2015-06-20:
Follow this guide..

It compiles perfectly for me..

http://www.instructables.com/id/Stream-games-to-your-Raspberry-Pi-2/

I just had to use

Chmod +x moonlight to get it to run

H3llsing, 2015-06-20:
@melzk 
now ur missing ALSA (sound)

try to run this command, it will install all the missing and necessary libs

sudo apt-get install libopus-dev libexpat1-dev libssl-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libjs-jquery

@Rassell 
try to locate a curl package 

sudo apt-get update && apt-cache search "curl"

you should look after something like
 libcurl........
usually you should install the latest the  libcurl4... and openssl but you could try to install another libcurl.

install with: 
sudo apt-get install  libcurl4.....   (complete package name
)

Rassell, 2015-06-20:
@H3llsing perfect! now i'm at the same point as melzk :D

dansl, 2015-06-20:
Did you guys install all the other Dependencies?

sudo apt-get install libopus-dev libexpat1-dev libssl-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev

Here's a complete rundown of the process with fresh install of Raspian. Doing this from memory... so hopefully it's accurate...
```
sudo apt-get update
sudo apt-get upgrade 
sudo apt-get dist-upgrade

sudo apt-get install libopus-dev libexpat1-dev libssl-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libjs-jquery build-essential

wget http://www.cmake.org/files/v3.2/cmake-3.2.3.tar.gz
tar xzf cmake-3.2.3.tar.gz
cd cmake-3.2.3
./configure
sudo make install

cd ~
wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb
wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb
sudo dpkg -i libevdev2_1.4.2+dfsg-1_armhf.deb libevdev-dev_1.4.2+dfsg-1_armhf.deb

wget https://github.com/irtimmer/moonlight-embedded/releases/download/v2.0.1/moonlight-embedded-2.0.1.tar.xz
tar xzf moonlight-embedded-2.0.1.tar.xz
cd moonlight-embedded-2.0.1
mkdir build
cd build/
cmake -D CMAKE_BUILD_TYPE=Release ../
make
sudo make install
```
From there you cd into the Moonlight/build folder, and you can do the following to see the list of available commands
```
cd ~/moonlight-embedded-2.0.1/build
./moonlight -help
```
If you want to setup a new controller
```
./moonlight -map controller.map
```
If you want to pair your PC
```
./moonlight pair
```
Running Moonlight, XXX.XXX.XXX.XXX is your PC's IP address, and this will run Steam at 720 30fps
```
./moonlight stream XXX.XXX.XXX.XXX -720 -30fps -app Steam -mapping controller.map
```

Rassell, 2015-06-20:
@dansl thanks! now im on:

pi@raspberrypi ~/moonlight-embedded-2.0.1/build $ make
Scanning dependencies of target moonlight
[  2%] Building C object CMakeFiles/moonlight.dir/src/audio.c.o
[  5%] Building C object CMakeFiles/moonlight.dir/src/mapping.c.o
[  8%] Building C object CMakeFiles/moonlight.dir/src/connection.c.o
[ 10%] Building C object CMakeFiles/moonlight.dir/src/xml.c.o
[ 13%] Building C object CMakeFiles/moonlight.dir/src/video.c.o
[ 16%] Building C object CMakeFiles/moonlight.dir/src/client.c.o
[ 18%] Building C object CMakeFiles/moonlight.dir/src/input.c.o
[ 21%] Building C object CMakeFiles/moonlight.dir/src/global.c.o
[ 24%] Building C object CMakeFiles/moonlight.dir/src/discover.c.o
[ 27%] Building C object CMakeFiles/moonlight.dir/src/mkcert.c.o
[ 29%] Building C object CMakeFiles/moonlight.dir/src/http.c.o
[ 32%] Building C object CMakeFiles/moonlight.dir/src/main.c.o
[ 35%] Building C object CMakeFiles/moonlight.dir/src/video/fake.c.o
[ 37%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/FakeCallbacks.c.o
[ 40%] Building C object CMakeFiles/moonlight.dir/third_party/moonlight-common-c/limelight-common/VideoStream.c.o

I hope it goes correctly :D

melzk, 2015-06-20:
here is all thing i did to get it to work :)
if someone look for it

Uppgrade of a fresh new installed raspian
 
	sudo apt-get dist-upgrade
	sudo apt-get update
	sudo apt-get upgrade	

download and installed all recomended libs in this trehead

	sudo apt-get install libjs-jquery
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb
	sudo dpkg -i libevdev2_1.4.2+dfsg-1_armhf.deb libevdev-dev_1.4.2+dfsg-1_armhf.deb
	sudo apt-get install libcurl4-openssl-dev
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-dev_1.4.2%2bdfsg-1_armhf.deb
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev-tools_1.4.2%2bdfsg-1_armhf.deb
	wget http://archive.raspbian.org/raspbian/pool/main/libe/libevdev/libevdev2_1.4.2%2bdfsg-1_armhf.deb
	sudo dpkg -i libevdev*
	sudo apt-get install libexpat1-dev
	sudo apt-get install libopus-dev libexpat1-dev libssl-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libjs-jquery

Download och unpack moonlight

	wget https://github.com/irtimmer/moonlight-embedded/releases/download/v2.0.1/moonlight-embedded-2.0.1.tar.xz
	tar xf moonlight-embedded-2.0.1.tar.xz

download and install cmake 3.2.3
	
	sudo apt-get install build-essential
	wget http://www.cmake.org/files/v3.2/cmake-3.2.3.tar.gz
	tar xzf cmake-3.2.3.tar.gz
	cd cmake-3.2.3
	./configure
	sudo make install

now its time to get into the moonlight folder
	cd ..
	cd moonlight-embedded-2.0.1
	mkdir build
	cd build/
	cmake -D CMAKE_BUILD_TYPE=Release ../
	make
	sudo make install

Rassell, 2015-06-20:
thanks to all, now is working fine... Only reamins to map correctly the xbox one controller and go :+1:

irtimmer, 2015-06-21:
Packages for moonlight-embedded are available for Raspbian.

Add this line to your */etc/apt/sources.list*
```
deb http://archive.itimmer.nl/raspbian/moonlight wheezy main
```

Install the package using
```
apt-get update
apt-get install moonlight-embedded
```

closed 2015-06-21 by irtimmer

Rutherberg, 2015-11-18:
Hi,

I installed the package for Raspbian following the wiki, but when I try to launch a moonlight command, it returns "Illegal instruction"
I'm running it on a RPi B with the lastest wheezy image.

I had to install libevdev2 manually because it was unable to locate it with apt-get (wheezy-backports : https://packages.debian.org/wheezy-backports/libevdev2 for armhf )

Do I have to compile it myself or is there another solution ?

Thanks !

Rutherberg, 2015-11-18:
Nevermind, I installed the wrong version of libevdev2, sorry !

georgcmef, 2015-11-21:
Can you please elaborate what is the rite file, please