moonlight-stream/moonlight-embedded #137: Freescale is missing, Segmentation fault

jbltx, 2015-06-19:
I tried to build moonlight-embedded-2.0 but when I try to use a ``pair`` or ``stream`` command, I have this error: 
```
Segmentation fault
```
Here is my system: 
```
Linux osmc 3.18.13-1-osmc #1 SMP PREEMPT Sun May 17 18:50:26 UTC 2015 armv7l GNU/Linux - Raspberry Pi 2
```
I had 2 errors during cmake:
```
-- Could NOT find Broadcom (missing: BROADCOM_INCLUDE_DIR)
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
```
So I built [userland](https://github.com/raspberrypi/userland) and now Broadcom is well found, but not Freescale. (like [this issue](https://github.com/irtimmer/moonlight-embedded/issues/134)). I am not sure if the segmentation error comes from this missing library.

closed 2015-06-19 by jbltx

Flip5, 2015-08-07:
I have the same issue, how did you fix it?

dezmaeth, 2015-09-17:
how do i fix this? got same issue!