moonlight-stream/moonlight-embedded #139: Failed to create client: Daemon not running

willag, 2015-06-20:
I installed Moonlight successfully but when I try to do "moonlight list", it tells me "Failed to create client: Daemon not running" and "Can't find server: Resource temporarily unavailable". Anyone else having these problems or have any idea how to fix it? I can game stream to my phone just fine.

7hunderbug, 2015-06-20:
You need to put your PC hostname or IP address at the end, so 
```moonlight list 192.168.1.43``` or similar.

irtimmer, 2015-06-21:
You need to run the avahi daemon to be able to use automatic discovery.

closed 2015-06-21 by irtimmer

willag, 2015-06-21:
Thank you so much 7hunderbug! That worked! The tutorial I was using didn't mention anything about that. I figured it was something obvious.