moonlight-stream/moonlight-embedded #143: PC won't pair with Raspberry Pi running Moonlight

habbgun, 2015-06-22:
My PC isn't pairing with my RaspberryPi that is running Moonlight. I found that my router C1000Z isn't compatiable with GameStream services so I went out and got a test router (Linksys WRT1200AC). I ran the pair command again with my PC's new ip address and it is still not connecting. The console says: Connection failed: HTTP response code said error. I was wondering what I can do to fix this? Or can you direct me to a place to figure out how to fix this? I am new to Raspberry Pi's and coding in general so you may have to spell it out for me. Thanks!

jonwingrove, 2015-06-22:
I'm having the same problem. I could pair fine with the Java version.
I just get "You must pair with the PC first", even when I have paired...

cgutman, 2015-06-22:
What version of GeForce Experience are you running? How did you build and install Moonlight?

jonwingrove, 2015-06-23:
GFE 2.4.5.44.
I got & built from git. with whatever was the latest revision about 6 hours ago... :) not going to be able to look at this for a few days now though...

habbgun, 2015-06-23:
I am using GFE version 2.4.5.44. And I used all the instructions in this tutorial: http://www.instructables.com/id/Stream-games-to-your-Raspberry-Pi-2/.
I also went through this: https://forums.geforce.com/default/topic/795004/gamestream/gamestream-basics-faq/1/. And made sure all the requirements were okay.

habbgun, 2015-06-25:
I made a paste bin of my commands and the results: http://pastebin.com/NC6drZSK.
I also have an image showing what ports are forwarded on my router: http://imgur.com/u5kt3r2
GeForce Version: GFE 2.4.5.57

cgutman, 2015-06-25:
The Freescale stuff is normal on a Pi. Moonlight embedded 2.0 isn't compatible with GFE 2.4.5.57. Download the 2.0.1 source and build that.

habbgun, 2015-06-25:
Wow, I cant believe that was it! Well I just got it to pair.
$ moonlight pair 192.168.1.101
Please enter the following PIN on the target PC: 9999
Paired

Now I am trying to list the games:
$ moonlight list
Failed to create client: Daemon not running
Can't find server: Resource temporarily unavailable

I get this result? Is this something I need to setup with GameStream?

cgutman, 2015-06-25:
That happens if Avahi isn't running because it tries to autodiscover your machine. Just put your IP address after like you did when pairing

habbgun, 2015-06-25:
That also worked! I now have one more issue. (hoping I dont have issues setting up my controller) 
$ moonlight stream 192.168.1.101 -1080 -30fps -app Steam
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...RTSP OPTIONS request failed
failed: -1
Cleaning up platform...done

Thank you so much for all the help!

habbgun, 2015-06-25:
I actually just reset my PC and my Pi and got it to work! Thanks for all the help. If I have anymore issues I'll will post.

cgutman, 2015-06-25:
Good. I really need to get around to making the RTSP failures print useful error codes instead of -1...

cgutman, 2015-06-26:
This issue should be closed since #149 kids merged

closed 2015-06-26 by habbgun