moonlight-stream/moonlight-embedded #144: Input Question Regarding Mouse and Gamepad

Dreadlow, 2015-06-22:
I was able to get my XBox 360 Wireless controller to work with the new C implementation of Moonlight, however upon switching inputs I no longer have the ability to use my mouse or keyboard if ever needed.

This necessity comes from some games having a launchpad pop-up prior to loading the game, or when input is necessary. An example is with Elite: Dangerous and trying to load in, as well as occasional times on the galaxy map and being able to search for specific planets.

Is there perhaps a way that I missed in order to get both the controller and keyboard/mouse to work together? I assumed that not specifying an input would pass all devices through, which it did in fact do to a point. However I noticed two things happen:

1.) The controller would only go left or right; no other input would work.
2.) If the keyboard is used, it registers all input into the console as well as the stream, thus sometimes causing me to start the stream again as soon as I stop streaming due to pushing up and enter.

Locking the program to only using the controller is fine for most cases, but it's just a question I had whether or not I should try to find another approach or if it was coming down the pipe.

Thanks!

jonwingrove, 2015-06-22:
you can pass in multiple inputs, e.g.
-input /dev/input/event1 -input /dev/input/event2
this seems to work for me.

Dreadlow, 2015-06-22:
Of course the one thing I don't try.

Haha, well I definitely appreciate the answer!

closed 2015-06-22 by Dreadlow

cgutman, 2015-06-22:
I don't think this issue should be closed yet. I was under the impression that passing no input devices should be equivalent to passing all input devices. If that's the case, this bug is still valid.