moonlight-stream/moonlight-embedded #145: Sound stutter - Moonlight-c (v 2.0.1) / Limelight (v1.2.2)

Dirrelito071, 2015-06-22:
Have tried Limelight v1.2.2 as well as Moonlight-C

Setup:
OS : Raspbian Wheezy (Split mem to 256)
HW : RPI 2 (not overclocked)
Note : Installed xboxdrv to remove flashing LED
(https://github.com/irtimmer/moonlight-embedded/wiki/Controllers-set-up)

Problem description :
Sounds starts to stutter (break up into silent fragments) 0.5-1 sec bursts with an approx intervall of 20 sec.

Testing :
Limelight v1.2.2 (java version) did not provide any prints but has the same sound stuttering.
Running Moonlight did however show the following prints : 

pi@raspberrypi ~/Moonlight $ sudo moonlight -mapping xbox360.map stream 192.168.0.100
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Alsa shortm write, write 16 frames
Alsa shortm write, write 236 frames
Alsa error from writei: -11
Alsa shortm write, write 4 frames
Alsa error from writei: -11
Alsa shortm write, write 236 frames
Alsa shortm write, write 236 frames
Alsa error from writei: -11

Any clues?

schulzad, 2015-06-29:
I've had this issue also. I test this on the latest building the latest master and didn't notice the issue.

1. If you pull the latest and rebuild does it still happen?
2. Are you using the libopus.so that's available in the release?

Dirrelito071, 2015-06-29:
I never figured out how to make a build, instead I used the precompiled
version and the dependencies like so :

Add this line to your */etc/apt/sources.list*

deb http://archive.itimmer.nl/raspbian/moonlight wheezy main

Install the package using

apt-get update
apt-get install moonlight-embedded

Dirrelito071, 2015-06-29:
Yes, I also tried using an older Rasp version without the updates.
( IMAGE USED IS 2015-02-16-raspbian-wheezy)

I believe that the installation I used got med v 2.0.1 of moonlight-c

2015-06-29 15:42 GMT+02:00 Martin Kvistholm <mkvistholm@gmail.com>:

schulzad, 2015-06-29:
Looks like I edited my response just before you posted so first:

>Have you tried the solution in issue #126?

A previous release of the [Java version v1.2.2](https://github.com/irtimmer/moonlight-embedded/releases/tag/v1.2.2)

It was suggested to download [libopus.so](https://github.com/irtimmer/moonlight-embedded/releases/download/v1.2.2/libopus.so) and place it in the same location as the limelight binary. 

Libopus is used to decode the audio stream and this version was prebuilt with a specific features set, namely hard float being disabled if I remember right.

cgutman, 2015-06-29:
Can you try running rpi-update?

Dirrelito071, 2015-06-30:
I have tried running rpi-update as well.
I have tried with the libopus.so version that's along limelight.jar in v1.2.2

I will try to go over it again to see if I missed something.

schulzad, 2015-06-30:
Do you notice the video also stuttering?

Try running at 30fps:

`sudo moonlight -mapping xbox360.map -720 -30fps stream 192.168.0.100`

Dirrelito071, 2015-06-30:
Yes, found some sound issues still but not as much (I think).
Running 1080p 60 FPS did no increase the sound issues, just the video-lag.

schulzad, 2015-06-30:
How is your RPi routing to your streaming box?

Mine is:
Ethernet -> Access Point bridge(Wifi) -> Router -> Ethernet -> streamingbox

irtimmer, 2015-09-12:
Duplicate of #126

closed 2015-09-12 by irtimmer