moonlight-stream/moonlight-embedded #150: Compiling on Rpi2 Debian Jessie (OSMC)

tortilaman, 2015-06-25:
I'm trying to install moonlight on my Raspberry Pi 2 with OSMC installed, which is based off of Debian Jessie and not Wheezy or Raspbian. As I kind of expected but hoped wouldn't happen, I'm getting some unmet dependencies:

```
The following packages have unmet dependencies:
 moonlight-embedded : Depends: libudev0 (>= 146) but it is not installable
                      Depends: libraspberrypi0 but it is not installable
E: Unable to correct problems, you have held broken packages.
```
I'm assuming I'll have to compile this, but I'll still run into the problem that the dependencies are unmet with the repositories I have available. Does anybody have experience doing this or can provide assistance?

irtimmer, 2015-06-25:
For Debian Jessie you need to recompile against libudev1 and somehow install the userland GPU libraries (libraspberrypi0), but I have no idea of where to find them for Debian Jessie

tortilaman, 2015-06-25:
Thanks for the information. I'll probably just do this in a Debian Wheezy environment then.

swetoast, 2015-06-28:
http://paste.osmc.io/osikocixix.vbs

came this far before it broke

cgutman, 2015-06-28:
Looks like you haven't done: git submodule update --init

You might have to rerun cmake after that too.

swetoast, 2015-06-30:
Yepp worked tnx :D

closed 2015-06-30 by irtimmer