moonlight-stream/moonlight-embedded #152: Raspberry pi 1 1080p 30fps discarding it's RTP packages

yakka34, 2015-06-27:
Raspberry pi using wheezy with 4.0 kernel and moonlight repo keeps discarding packages.
I don't remember seeing this happen I build from sources using debian "testing" branch.
720p with 60fps causes no problem.

pi@raspberrypi ~ $ moonlight stream -1080 -30fps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Discarding RTP packet queued for too long
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet queued for too long
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Discarding RTP packet queued for too long
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Discarding RTP packet queued for too long
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Waiting for IDR frame
Resync complete
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
Resync complete
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
Resync complete
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet queued for too long
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Discarding RTP packet queued for too long
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Resync complete
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Resync complete
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
Resync complete
Loss stats thread terminating #1
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done
pi@raspberrypi ~ $

cgutman, 2015-06-28:
These messages indicate that the network is dropping packets. Are you using Ethernet to your computer and Pi?

yakka34, 2015-06-28:
Yes I'm. on the raspberry end is a cat 6 and on the streaming computer is cat 5 cable. Streaming is limited to cat 5 speed of course but shouldn't 10/100 be more than enough for 1080p 30fps.

maverickjesus, 2015-07-18:
What bitrate setting did you use?

yakka34, 2015-07-18:
I would guess that mine was 4250. Basically I tried many settings and 4500 was too high and 3500 too low. Reference steam in-house streaming settings for presets

closed 2015-08-08 by irtimmer