moonlight-stream/moonlight-embedded #153: not working on imx with custom kernel 3.14.14 (V4L2 MXC fixed)

rkorn86, 2015-06-27:
As already mentioned in https://github.com/irtimmer/moonlight-embedded/issues/58 im running my wandboard quad on a custom 3.14.14 kernel with these patches https://github.com/SolidRun/linux-imx6-3.14/pull/26 applied. I gave https://github.com/irtimmer/moonlight-embedded/commit/9bf1ad75e80786f6fffc6636e43e060b78bf4d42 a try, but without success.
Here is my log output:
```
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Waiting for IDR frame
Resync complete
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Received OOS audio data (expected 39309, but got 39312)
Received OOS audio data (expected 39313, but got 39317)
Audio packet queue overflow
Discarding RTP packet after queue overgrowth
Network dropped an entire frame
Resync complete
Received OOS audio data (expected 39318, but got 39356)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 39359, but got 39366)
Discarding RTP packet queued for too long
Received OOS audio data (expected 39367, but got 39369)
Discarding RTP packet queued for too long
Received OOS audio data (expected 39374, but got 39376)
Audio packet queue overflow
Discarding RTP packet after queue overgrowth
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 39397, but got 39433)
Received OOS audio data (expected 39434, but got 39436)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 39463, but got 39468)
Audio packet queue overflow
Received OOS audio data (expected 39469, but got 39500)
Discarding RTP packet after queue overgrowth
Video decode unit queue overflow
Resync complete
Received OOS audio data (expected 39532, but got 39535)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 39543, but got 39552)
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Audio packet queue overflow
Discarding RTP packet queued for too long
Received OOS audio data (expected 39553, but got 39596)
Discarding RTP packet after queue overgrowth
Network dropped an entire frame
Resync complete
Audio packet queue overflow
Received OOS audio data (expected 39597, but got 39628)
Audio packet queue overflow
Discarding RTP packet queued for too long
Received OOS audio data (expected 39644, but got 39679)
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Received OOS audio data (expected 39680, but got 39684)
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Received OOS audio data (expected 39685, but got 39688)
Waiting for IDR frame
Resync complete
Discarding RTP packet after queue overgrowth
Audio packet queue overflow
Waiting for IDR frame
Received OOS audio data (expected 39689, but got 39734)
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet queued for too long
Audio packet queue overflow
Audio packet queue overflow
Received OOS audio data (expected 39752, but got 39819)
Video decode unit queue overflow
Waiting for IDR frame
Resync complete
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet queued for too long
Received OOS audio data (expected 39876, but got 39881)
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Audio packet queue overflow
Video decode unit queue overflow
Waiting for IDR frame
Waiting for IDR frame
Resync complete
Received OOS audio data (expected 41910, but got 41944)
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Audio packet queue overflow
Received OOS audio data (expected 41954, but got 41994)
Discarding RTP packet after queue overgrowth
Network dropped beginning of a frame
Resync complete
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 42001, but got 42004)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 42005, but got 42008)
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Received OOS audio data (expected 42061, but got 42066)
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Audio packet queue overflow
Received OOS audio data (expected 42067, but got 42119)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 42139, but got 42141)
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Received OOS audio data (expected 42147, but got 42149)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 42151, but got 42153)
Received OOS audio data (expected 42173, but got 42176)
Discarding RTP packet queued for too long
Network dropped beginning of a frame
Discarding RTP packet queued for too long
Network dropped beginning of a frame
Waiting for IDR frame
Waiting for IDR frame
Resync complete
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Discarding RTP packet queued for too long
Audio packet queue overflow
Discarding RTP packet after queue overgrowth
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Received OOS audio data (expected 42185, but got 42241)
Received OOS audio data (expected 42252, but got 42257)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 42258, but got 42261)
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Discarding RTP packet after queue overgrowth
Discarding RTP packet after queue overgrowth
Audio packet queue overflow
Received OOS audio data (expected 42282, but got 42318)
Audio packet queue overflow
Received OOS audio data (expected 42341, but got 42372)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 42373, but got 42375)
Discarding RTP packet after queue overgrowth
Received OOS audio data (expected 42416, but got 42419)
Discarding RTP packet queued for too long
Video decode unit queue overflow
Discarding RTP packet queued for too long
Waiting for IDR frame
Resync complete
Received OOS audio data (expected 42444, but got 42448)
Audio packet queue overflow
Received OOS audio data (expected 42449, but got 42484)
Discarding RTP packet queued for too long
Received OOS audio data (expected 42513, but got 42516)
Discarding RTP packet queued for too long
Received OOS audio data (expected 42517, but got 42520)
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Waiting for IDR frame
Video decode unit queue overflow
Resync complete
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...
```

cgutman, 2015-06-28:
Things look pretty broken. Specifically "Video decode unit queue overflow" and "Audio packet queue overflow" messages are bad. It seems like the the audio and video renderers are deadlocked/slow.

What parameters are you passing when starting the stream? Do you get any video or audio output? Does a file called fake.h264 get written in the current directory?

rkorn86, 2015-06-28:
Sorry, I forgot to mention that. 
fake.h264 is not created. 
Command to run moonlight was `root@wandboard - 08:14:23 - /opt/moonlight-embedded/build # ./m
oonlight stream -app "FIFA 15" -30fps 192.168.0.13 >log 2>&1`

rkorn86, 2015-07-07:
@irtimmer do you have any idea?

irtimmer, 2015-07-07:
Yes you're trying a fix for the Cubox on a Wandboard (in no way this is gonna work)

rkorn86, 2015-07-08:
yeah i thought imx6 = imx6.

irtimmer, 2016-01-16:
Bug in kernel

closed 2016-01-16 by irtimmer