moonlight-stream/moonlight-embedded #155: Wrong keycodes in keyboard.h

wkewlin, 2015-06-29:
I found a bug in keyboard.h, the code for space is 0x20 and not 0x32
 ```
diff --git a/src/keyboard.h b/src/keyboard.h
index 66d67f8..2813254 100644
--- a/src/keyboard.h
+++ b/src/keyboard.h
@@ -75,7 +75,7 @@ static const short keyCodes[] = {
   0x10, //VK_SHIFT Right shift
   0, //VK_KPASTERISK
   0x11, //VK_ALT Left alt
-  0x32, //VK_SPACE
+  0x20, //VK_SPACE
   0x14, //VK_CAPS_LOCK
   0x70, //VK_F1
   0x71, //VK_F2
```
From randomly typing I think there are more issues with this translation table but I did not have the time to properly check and list them, maybe later.
What is the source of those codes apart from java app? Are they reverse engineered?

vulcandth, 2015-07-11:
I'm having this same issue with the keyboard.... Pressing the spacebar results in the number 2 key.

The above should fix this.

cgutman, 2015-07-11:
This change should be merged. It was somehow messed up in translation from Java. It was originally written as 32 then mistranslated to 0x32.

closed 2015-07-11 by irtimmer

ChildLearningClub, 2023-08-15:
Is it possible this crept back in? When using the keyboard through moonlight and sunshine with default settings the spacebar does not work as expected. If I first hit the shift key and then the spacebar it works, but not with the spacebar alone.