moonlight-stream/moonlight-embedded #156: In cec code key_action up and down reversed

wkewlin, 2015-06-29:
At least on my Panasonic TV cec sends duration == 0 for key press and duration != 0 for release. This is probably always true, seems at least reasonable.
So input.c need the following change:

```
  if (value != 0) {                                                                                                            
     short code = 0x80 << 8 | keyCodes[value];                                                                                  
-    LiSendKeyboardEvent(code, (key.duration > 0)?KEY_ACTION_DOWN:KEY_ACTION_UP, 0);                                            
+    LiSendKeyboardEvent(code, (key.duration > 0)?KEY_ACTION_UP:KEY_ACTION_DOWN, 0);                                            
   }     
```

schulzad, 2015-06-29:
I can test this on my Panasonic TV as well.

cgutman, 2015-06-30:
libcec-daemon source also indicates duration == 0 is down and duration != 0 is up.

https://github.com/bramp/libcec-daemon/blob/583653bd3668b2797ebf057d20dfa0ecbc329b5a/src/main.cpp#L324

@irtimmer Can you make the above change?

irtimmer, 2015-06-30:
Fixed with c10195f524c1f2856dfd5b9a043ee702978bd8b5

closed 2015-06-30 by irtimmer