moonlight-stream/moonlight-embedded #157: Arch repository corrupted

aPollO2k, 2015-06-30:
I tried to install via pacman on arch linux. I installed the latest arch linux version from here: http://archlinuxarm.org/platforms/armv6/raspberry-pi

I added the lines to /etc/pacman.conf and tried to install with an 404 error:

[root@alarmpi boot]# pacman -Sy moonlight-embedded
:: Synchronizing package databases...
 core                               166.8 KiB   758K/s 00:00 [################################] 100%
 extra                                2.2 MiB  1084K/s 00:02 [################################] 100%
 community                            2.8 MiB   830K/s 00:03 [################################] 100%
 alarm                               33.4 KiB  1114K/s 00:00 [################################] 100%
 aur                                 62.4 KiB  1041K/s 00:00 [################################] 100%
error: failed retrieving file 'moonlight.db' from archive.itimmer.nl : The requested URL returned error: 404
error: failed to update moonlight (unexpected error)
error: target not found: moonlight-embedded

After that i tried to force a resync of the pacman database with the same error. Before i tried these i updated my arch to the newest version over pacman. Maybe there is something wring with the repository? I use arch for the first time and i'm a pacman-noob maybe i'm wrong.

irtimmer, 2015-07-01:
It should now be fixed.

closed 2015-07-01 by irtimmer

Josh-Cropper, 2022-03-16:
This is happening again